aircrack-ng/rtl8812au #4: Failed to "make" rtl8814au driver

alisidani, 2017-08-25:
When i try to make from https://github.com/aircrack-ng/rtl8812au i get an error:
Make Rtl8814a :
/bin/sh: 1: syntax error: "(" unexpected
Makefile:1264: recipe for target 'modules' failed

kimocoder, 2017-08-25:
I just clone the repo and buildt rtl8814au on kernel v4.12 myself, and the build was successfull. Are you trying to build for kernel v4.11 or v4.12 btw?

kimocoder, 2017-08-25:
Aaah! I just firgured out your problem sir! I'll fix it right away.. Kernel support is still missing! A fix will be pushed to this repo in few hours.. Thank you

kimocoder, 2017-08-25:
The v4.3.21 branch have been updated with the kernel v4.11 import libs fix. It now builds fine, please update with "git pull" and "make RTL8814=1" and give feedback on how it now works :)

alisidani, 2017-08-25:
I did as you mentioned
Make rtl8814=1 success 
Make install: error install: cannot stat '8812au.ko' No such file or directory

kimocoder, 2017-08-25:
"make" first to build the 8812, then "make RTL8814=1", then "make install" to insert them both

alisidani, 2017-08-25:
Did it but still the same results..i can see networks but cannot join them: keep asking for password
I tried to remeve the password from the wireless: status keep on connecting then it changes to: not connected

kimocoder, 2017-08-25:
This error i can't reproduce at the moment, because I'm still on holiday. but we leave the issue report here for others too read/discuss then I'll get a look at it when im home with my adapter

alisidani, 2017-08-25:
Can you email me privatly your contact info to contact you directly and keep you updated on my issue?

kimocoder, 2017-08-25:
I did some hours ago, sending one more right away

kimocoder, 2017-08-27:
@alisidani check issue report #6 and try doing the same (updating the driver) & report if the problem still persists or if it was fixed (in the new version)

alisidani, 2017-08-27:
Thanks for your support @kimocoder
Finally i solved the issue after 1 week of trying to get the awus1900 to work

You helped me to build successfully the driver then i have the problem of the connecting password loop

The problem was from the Vmware who was originally configured to handle usb 2.0 but the awus1900 use usb 3.0 cable
So after changing the configuration of the vmware i am able now to connect to wireless networks successfully 

I hate vmware but its still better than virtual box

closed 2017-08-27 by alisidani

kimocoder, 2017-08-27:
Im happy to hear you figured it out sir :) happy hunting!