aircrack-ng/rtl8812au #6: Kali crashes after installing realtek-rtl88xxau-dkms

axldmg, 2017-08-25:
Using Kali 2017.1 updated to v4.12.0-kali1-amd64 kernel. Running as guest on VMWare Fusion 8.5.8, hosted on Mac OS X 10.12.6

To summarize: after installing the realtek-rtl-88xxau-dkms package, Kali ends up crashing (no exception), either:

- Instantly if the adapter is connected to an AP and, let's say a browser, is launched (or any other app generating traffic to the Internet): firefox don't even have the time to parse and display all the html/css code and the crash occurs before getting the full page displayed.

- Somehow after a several minutes if using the adapter otherwise: So, to gather more data, I chose this way of crashing Kali with Alfa AWUS036ACH and reproduced this incident by going into monitor mode and back to managed mode and connecting to my usual AP, disconnecting and returning to monitor mode, back and forth 2 or 3 times, before it did crash. (ip link set wlan0 down, iwconfig wlan0 mode monitor, ip link set wlan0 up, airodump-ng wlan0, ip link set wlan0 down, iwconfig wlan0 mode managed, ip link set wlan0 up, <connecting to AP>, <disconnecting from AP>, ip link set wlan0 down... and so forth...). Some other times I tried with using airman-ng check kill and/or stopping/starting/restarting network-manager at the relevant moments, with systemctl, but it does not change the outcome at all.

- Eventually within an hour, even if not plugging the Alfa adapter at all (actually, the only way not to end up crashing at some point has been for me te revert to the snapshot made right before installing the realtek-rtl88xxau-dkms package, and I have no issues of any kind thereafter).

The hereby documented crash occurred between 18:52:30 and 18:52:59 (I can't tell exactly, but all logs stop at 18:52:30). I rebooted at 21:51:00, and I am enclosing extracts from /var/log/:, (stamped from any recorded event occurring right prior to 18:52, to the time of reboot 21:51)
- daemon.log
- debug
- kern.log
- messages
- syslog
- user.log
As well as screen captures:
- rtl88xxau-dkms-install, showing proper install of latest version of the package
- boot screenshot after crash showing after the necessary hard reboot.
- screenshot with modules related to rtl8812u's info

As a sanity-check, I tried and got exactly the same behavior with a fresh install of Kali 2017.1.

I hope this helps. Obviously, this is academically very interesting, but definitely not urgent in any way (besides, AWUS036NHA is still doing a great job!).

[logs.zip](https://github.com/aircrack-ng/rtl8812au/files/1251928/logs.zip)

[image]
[image]
[image]

kimocoder, 2017-08-25:
Thanks for the good information on the issue. As we discussed on Kali bug system, I will personally be looking into the problem when I get home from holiday, but if others could try reproducing this 'crash' it would be just great. E.g, other systems then OSX, pure Kali setup without virtual machine and so on. Again, great report of the issue.

axldmg, 2017-08-25:
(Thanks @kimocoder, much appreciated). Enjoy the rest of your holidays man! ;)

Btw, by 'crash' I mean that it just literally freezes (like absolute zero kelvin freezing, totally unresponsive, just like it was a screenshot) :). And I have to force shutdown it using the menu of VMWare.
On my side, I will try to reproduce using VirtualBox sometimes during the weekend, and update this thread with the outcome...

yesimxev, 2017-08-25:
Doesn't freeze my pure Kali setup. But similar compability as before - at least airodump sometimes scans, but sometimes not. Still no captures at all, no injection. Also no airbase. Mdk3 and hostapd works fine though

kimocoder, 2017-08-27:
@axldmg & @yesimxev I just pushed a commit with DKMS & kernel v4.12 support for branch v5.1.5.
could you guys try updating the driver to see if the problem is fixed? Kali is using v4.3.21 at the moment, but the v5.1.5 is becoming ready.

* Howto?
Uninstall the "realtek-rtl88xxau-dkms" with apt-get
Use "git clone https://github.com/aircrack-ng/rtl8812au -b v5.1.5" for cloning the 'new' version
Use "make" for building 8812au driver, or "make RTL8814=1" for the 8814au driver. Remember when using "make install" that both drivers have to be buildt.

then check if your problem still persist & report back here

yesimxev, 2017-08-27:
I built both of them, then did make install RTL8814=1
Airodump scans straightaway (with a bit of lag sometimes), can monitor 2.4ghz with inject! 👍 
Airbase works! 👍 Can't set MTU to 1800 (not related to driver I guess?)
But 5ghz injection still not working? Just says BSSID not found whereas it's obviously existing. That might be not driver issue just aircrack tools
Thanks @kimocoder

kimocoder, 2017-08-27:
The 5ghz issue I actually got a fix for somewhere, it's a patch for aircrack-ng which I need to lookup again, then I may post it here later. There still is a ping buffer issue related to this driver stack, but I hope someone may resolve it with time. There's also released a v5.2.9 version (only for 8812au), but it needs monitor/injection support still. Thanks for your time sir

axldmg, 2017-08-27:
Hey :)
Well, I'm afraid it's not as straight forward as for @yesimxev on my side. :(

I get an stderr output when invoking make command (see attached screenshots).
PS: I have my main Kaliwork, which I have indeed personalized, it's the first screenshot, as well as a KaliDev, which is always reverted whatever I do with, except for updates, so it's like a fresh install up to date, and both are showing the same error message).
Third screenshot is the content of /lib/modules/4.12.0-kali1-amd64, since that it's the error raised...

[image]
[image]
[image]

axldmg, 2017-08-27:
As for the trial with VirtualBox, well, I just couldn't do it because VirutalBox support only USB 1.0 (no comment lol), unless installing an extension pack which is described, by VirtualBo,x as "Potentially fatal to your system"... which does not sound very attractive!.

axldmg, 2017-08-27:
btw, I know that, by default, [airodump-ng hop on 2.4GHz channels](http://bit.ly/2izoW46) and that you have to pass as an argument to sniff on 5.0GHz... I wonder if it could be the same with aireplay-ng?

yesimxev, 2017-08-27:
When I enter a 5ghz channel (eg. 36) on airodump, I can monitor 5ghz network. Only injection not working. For your build. Did u do full apt-get update, upgrade, dist-upgrade? I'm running the latest of everything and built with make and make RTL8814=1 . Then make install RTL8814=1  worked straightaway. Pure install though no VM

kimocoder, 2017-08-27:
There's something wrong with your modules, missing the "build" symlink and others.. have you updated everything btw? because the kernel got updated 3 times the past 46 hours as seen [here](http://git.kali.org/gitweb/?p=packages/linux.git;a=summary). you may also try to re-install the linux-headers package because your build is a bit f***** over there 🤕

axldmg, 2017-08-27:
Well, if I have a regular need for sniffing, it's not in the context of packet injection, so I barely know the aireplay-ng feature of the aircrack-ng suite (thus I'm not going to be of great use here :)).

Yeah, yeah, of course, absolutely up to date, packages and distro.

kimocoder, 2017-08-27:
@yesimxev as for the 5ghz issue take a look what I did [here](https://github.com/kimocoder/aircrack-ng/commits/master) you could clone the main aircrack-ng repo and cherry-pick those patches for fixing the 5ghz issue + adding channels or whatever you want :)

kimocoder, 2017-08-27:
@axldmg it should look like this..

[image]

axldmg, 2017-08-27:
@kimocoder LOL
Yeah, and the weird part is that my KaliDev is has exactly the same problem...

Please have a look at the screenshots below and tell me if that make sense to you...

[image]
[image]

In the meantime, I'm downloading a fresh iso from kali.org... I'll see how it ends up once installed
Also, I'll try to manually add the ln as per your screenshot

axldmg, 2017-08-27:
alright, headers being reinstalled now (of course, apt doesn't have this feature, only apt-get...) 
TBC... ;)

axldmg, 2017-08-27:
headers reinstalled, rebooted, updated, upgraded, distilled-upgraded (0 packages each time)
make: same error.
building symbolic links now (because the targets of build and source exist, and in the same up to date version as yours)...
TBC..

axldmg, 2017-08-27:
[That's the only way to put it](http://bit.ly/2ggkhTT) ;)
And, of course, I've been browsing like hell for the last 15 minutes, launching browser, mail app and IRC at the same time, and no more Mr Freeze!! 

Brilliant!!!!! Thank you so much for amazing fixing in record-breaking time @kimocoder 

POC:
[image]

axldmg, 2017-08-27:
I'm running out of time right now, but I'll have a thorough use of aircrack-ng during the week: I'll write down here next weekend any event worthwhile

kimocoder, 2017-08-27:
Hahaha, just loved your vid & thanks.. Made my day :D

Your happy, I'm happy! Glad we could help eachother figure this out sir. Have a fantastic day.

kimocoder, 2017-08-27:
issues with TLP / powersaving has now been resolved. Please use "git pull" to update the driver and build it again to resolve those issues :)

alisidani, 2017-08-27:
For awus1900 :not resolver
Can put it into monitor mode and also scan networks but cannot connect to any..keep asking for password in awus1900 and not connecting

axldmg, 2017-08-27:
@kimocoder update perfect! Monitor mode now lists APs fast as a blink of an eye and within an impressive range (even faster now than my old good 036NHA with Atheros AR9271, way wider range). Really nice, I'm positive I'll have a lot of fun with during the week (and, as I posted earlier, I'll come back to you next weekend with everything relevant)

@alisidani As per your description, I'm not even sure you correctly entered in monitor mode... Could you please elaborate and detail the commands you use? (it will help to help you)

kimocoder, 2017-08-27:
Great! This has been an educational day, got lots done. As for the AWUS1900, I got this myself and will take a look at it when im home in 7 days but besides that issue, it's all good :)

closed 2017-08-27 by kimocoder

kimocoder, 2017-08-27:
AWUS1900 issue fixed. Everything working great