aircrack-ng/rtl8812au #8: Error compilation branch v4.3.21 for 8812au in kali with kernel 4.12

kcdtv, 2017-08-31:
Hi there! 
I just saw that you opened this repsository. Well done! Thanks  :smiley_cat: 
I installed the master branch wihtout any problem (for 8112au chipset) and it was even looking to go better than with other recent kerenels. 
Than i wanted to install branch 4.3.21 but i got this errors whith `ioctl_cfg80211.o `
```
/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_connect’:
/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.c:807:6: error: passing argument 2 of ‘cfg80211_roamed’ from incompatible pointer type [-Werror=incompatible-pointer-types]
    , notify_channel
      ^~~~~~~~~~~~~~
In file included from /home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/include/osdep_service_linux.h:98:0,
                 from /home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/include/osdep_service.h:41,
                 from /home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/include/drv_types.h:32,
                 from /home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.c:22:
/usr/src/linux-headers-4.12.0-kali1-common/include/net/cfg80211.h:5435:6: note: expected ‘struct cfg80211_roam_info *’ but argument is of type ‘struct ieee80211_channel *’
 void cfg80211_roamed(struct net_device *dev, struct cfg80211_roam_info *info,
      ^~~~~~~~~~~~~~~
/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.c:809:6: warning: passing argument 3 of ‘cfg80211_roamed’ makes integer from pointer without a cast [-Wint-conversion]
    , cur_network->network.MacAddress
      ^~~~~~~~~~~
In file included from /home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/include/osdep_service_linux.h:98:0,
                 from /home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/include/osdep_service.h:41,
                 from /home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/include/drv_types.h:32,
                 from /home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.c:22:
/usr/src/linux-headers-4.12.0-kali1-common/include/net/cfg80211.h:5435:6: note: expected ‘gfp_t {aka unsigned int}’ but argument is of type ‘unsigned char *’
 void cfg80211_roamed(struct net_device *dev, struct cfg80211_roam_info *info,
      ^~~~~~~~~~~~~~~
/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.c:805:3: error: too many arguments to function ‘cfg80211_roamed’
   cfg80211_roamed(padapter->pnetdev
   ^~~~~~~~~~~~~~~
In file included from /home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/include/osdep_service_linux.h:98:0,
                 from /home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/include/osdep_service.h:41,
                 from /home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/include/drv_types.h:32,
                 from /home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.c:22:
/usr/src/linux-headers-4.12.0-kali1-common/include/net/cfg80211.h:5435:6: note: declared here
 void cfg80211_roamed(struct net_device *dev, struct cfg80211_roam_info *info,
      ^~~~~~~~~~~~~~~
/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_add_monitor_if’:
/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.c:3962:10: error: ‘struct net_device’ has no member named ‘destructor’; did you mean ‘priv_destructor’?
  mon_ndev->destructor = rtw_ndev_destructor;
          ^~
/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_preinit_wiphy’:
/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.c:6618:18: error: ‘WIPHY_FLAG_SUPPORTS_SCHED_SCAN’ undeclared (first use in this function)
  wiphy->flags |= WIPHY_FLAG_SUPPORTS_SCHED_SCAN;
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.c:6618:18: note: each undeclared identifier is reported only once for each function it appears in
/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.c: At top level:
/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.c:6651:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .change_virtual_intf = cfg80211_rtw_change_iface,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.c:6651:25: note: (near initialization for ‘rtw_cfg80211_ops.change_virtual_intf’)
/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.c:6676:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .add_virtual_intf = cfg80211_rtw_add_virtual_intf,
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.c:6676:22: note: (near initialization for ‘rtw_cfg80211_ops.add_virtual_intf’)
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.12.0-kali1-common/scripts/Makefile.build:307: fallo en las instrucciones para el objetivo '/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.o'
make[4]: *** [/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au/os_dep/linux/ioctl_cfg80211.o] Error 1
/usr/src/linux-headers-4.12.0-kali1-common/Makefile:1532: fallo en las instrucciones para el objetivo '_module_/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au'
make[3]: *** [_module_/home/kcdtv/GitHub/RTL88XXAU/driversbyaircrack/rtl8812au] Error 2
Makefile:152: fallo en las instrucciones para el objetivo 'sub-make'
make[2]: *** [sub-make] Error 2
Makefile:8: fallo en las instrucciones para el objetivo 'all'
make[1]: *** [all] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-4.12.0-kali1-amd64'
Makefile:1717: fallo en las instrucciones para el objetivo 'modules'
make: *** [modules] Error 2
```

kimocoder, 2017-08-31:
Hi there @kcdtv :) well, v4.3.21 is not maintained as well as it should be, but the work have been focused on the v5.1.5 branch as issues with TLP / powersaving and kernel support has been added there. If you want me to fix the v4.3.21 branch, I sure could do that too:)

realtek-rtl88xxau-dkms in Kali now contains v5.1.5 branch which fixes some issues reported by other users (freeze & crashes).

kimocoder, 2017-08-31:
Tried using the kernel v4.12 patch used in "master" and the v5.1.5 branch, but unfortunately it did not work and i don't really have the time to investigate this further right now. but using v5.1.5 branch would anyway the best, since the TLP issue (stuck at 12dBm) is fixed, more adapter support, kernel v4.12 support ++

btw, feel free to commit a fix for kernel support in the v4.3.21 branch.

kcdtv, 2017-09-01:
Hi! 

> btw, feel free to commit a fix for kernel support in the v4.3.21 branch.

I tried to fix it before reporting the issue but I unfortunatly really do not know C at all and I am useless for that.  I'll let you know if i find something. Cheers

kimocoder, 2017-09-10:
@kcdtv The v5.2.9 (for 8812AU only) has been added

kimocoder, 2017-09-16:
Kernel v4.12 Support has finally been added to v4.3.21 branch ! Happy hunting @kcdtv

closed 2017-09-16 by kimocoder

kcdtv, 2017-09-20:
Thank you so much! :smiley_cat: