aircrack-ng/rtl8812au #14: v5.1.5 wireless-ac

qwikwit, 2017-09-13:
I could not get v5.1.5 working with Wireless-AC. It would only connect at 5ghz N. Also the previous version 4.3.21 did have AC working. Is it possible to be able to use an earlier kernel so that v4.3.21 compiles

Things I have tried:
running apt-get update
then apt-get dist-upgrade.
reinstalled Kali 2017.1
reinstalled Kali rolling W37.

kimocoder, 2017-09-13:
Hi there!

If you use Kernel v4.11, the v4.3.21 will compile. Kernel v4.12 is not supported on that branch yet, because of some issues.

Are you using RTL8812AU or RTL8814AU ?

qwikwit, 2017-09-14:
Hi!

I am using rtl8814au. The device is an Alfa AC1900. It is my favorite wireless device. A test using airodump-ng i had better PWR,  RXQ, and beacons than any other 88xx device.

I would like to share some things that i learned.  Your v5.2.9 wont compile on Kali rolling. v5.1.5 waking up from sleep sometimes i had to unplug and plug in the device to my usb to get a better speed.

Astsam/rtl8812au v 4.3.21 works with 8812au, 8814au 8821au, 8811au. It supports monitor mode, injection, and tx-power. I did a ping test with the Alfa set to 30dbm overnight  to my AC1300 router. After 8 hours, 4 packets were dropped out of many thousands and the worst latency was 80ms. Wireless-AC provides the greatest range of the 5ghz band. I use channel 149 80mhz because it transmits at 30dbm vs 20-23dbm on channel 36. My router has no options for transmit power.

v5.2.9 is only for the 8812au and does not have monitor mode. Also don't run a wifi card without an antenna.

I found these which works for me on kernel v4.12: but tx-power is set by crda and you can't change tx-power with iwconfig. I assume 30dbm is used.
tpircher/rtl8814AU forked from diederikdehaas/rtl8814AU
nazar-pc/rtl8814AU forked from sbosshardt/rtl8814AU

kimocoder, 2017-09-14:
Thanks for the information. Will take a look at the v4.3.21 kernel v4.12 support, as for the v5.2.9 HAL should be added for 8814 (work in progress, see TODO). On v5.1.5 branch I recently resolved TLP issues. Remember that this is a fork from astsam, but astsam's is stuck behind on several fixes (including kernel v4.12 support for v5.1.5,  TLP issue fix + device support)

Will take a look at the sources you mention if there are patches that may be used.
I also got the AC1900, but haven't been home for a while to do testing.

Will do some work again very soon

kimocoder, 2017-09-16:
@qwikwit Kernel v4.11/v4.12 support has been added to branch v4.3.21. Happy hunting

qwikwit, 2017-09-17:
It Compiled with no errors. Works! I  tested tx-power, wireless AC, and monitor mode

kimocoder, 2017-09-19:
Closed because the kernel v4.12 support was successfully added, still working on the 30dBm issue (stuck on 12dBm) issue. Thanks

closed 2017-09-19 by kimocoder

lhernanz, 2017-09-26:
Hi @kimocoder,

This bug is still in the code, at least for the Alfa AC1900, i.e. AC is not working for the driver 5.1.5. If the guidance is that this driver version should not be used for AC, I would suggest that you would add this recommendation in the main page to avoid further confusion.

I have tried to find some information about what driver is better for what or what are the new features of the new drivers but I have not been successful so far.

Thanks for your work on this!

Regards