aircrack-ng/rtl8812au #16: v5.1.5 Alfa AC1900 not chaning to USB3 mode

lhernanz, 2017-09-26:
Hi,

as the title says, the Alfa stays in USB2 mode. I have tracked the problem to the code below. For whatever the reason (I have the impression that the EEPROMUsbSwitch variable is not properly initialized in the 8814 driver but I have not tested this thoroughly), EEPROMUsbSwitch is false and this stops the mode change. Commenting the two lines solves the problem.

Regards.

--- a/os_dep/linux/usb_intf.c
+++ b/os_dep/linux/usb_intf.c
@@ -649,15 +649,17 @@ static int usb_reprobe_switch_usb_mode(PADAPTER Adapter)
        u8 ret = _FALSE;
 
        /* efuse not allow driver to switch usb mode */
-       if (pHalData->EEPROMUsbSwitch == _FALSE)
-               goto exit;
+       //if (pHalData->EEPROMUsbSwitch == _FALSE)
+       //      goto exit;

kimocoder, 2017-09-26:
Thanks for the report, will take a look shortly :)

kimocoder, 2017-09-26:
The fix have been included, but could you do further testing on it? Don't have a 5ghz nearby for some weeks so testing would be highly appreciated.

lhernanz, 2017-09-26:
Thanks,
I will keep testing the changes, although AC is not working with this version of the driver (see me comment on the AC bug) and, therefore, I am thinking about testing a previous version of the driver.
Regards

kimocoder, 2017-09-26:
Well, then the changes does not need to be included in the v5.1.5 branch, but rather in the v4.3.21 maybe? Please be specific and do some test before the commit/changes are made and where the changes need to be made.

Im currently working on adding the 8814au HAL to the v5.2.9, so in short time I will start testing the works of it, but we're happy to include fixes on every branches that we have available.

lhernanz, 2017-09-26:
Sorry, I have not explained myself properly. 

The bug report and fix is for v5.1.5. I have tested the change and I can confirm that the card changes to USB3.0 as expected after applying the patch.

My point about the need to move to a different version is that in v5.1.5 AC is not working (at least for this card), so having USB3.0 is a moth point as you are not going to get wifi speeds above what USB 2.0 can support. Of course, the ideal thing would be to be able to fix the AC problem too :)

Regards

kimocoder, 2017-10-02:
@lhernanz I think I've found the AC issue.. adding the fix in v4.3.21 branch in few hours, while v5.1.5 needs some more days :)

kimocoder, 2017-10-02:
@lhernanz take a look at [this](https://github.com/weedy/rtl8812AU/commit/5f75242fce32cc1d86cb80a10ffdb7da9dac5185)

I suspect this has something to do with the AC, but the patch is not compatible without modifications,
feel free to take a look at it

lhernanz, 2017-10-02:
Hi,

Uhm, I am not entirely sure. In the v5.1.5 you can already see some similar code doing some similar (there are important differences between the drivers, though) things. See the function rtw_cfg80211_create_vht_cap in 5.1.5 vs the rtw_cfg80211_init_vht_capab in 4.3.21.

But clearly the code in 5.1.5 is not working for AC, so it could be that the code is incomplete in that version. It is kind of amazing that Realtek of Alfa are not providing updated drivers for their products that are fully functional.

kimocoder, 2017-10-02:
I will investigate further.. Have done lots of changes/additions on the v4.3.21 today + got some more upcoming tomorrow, but on the v5.1.5 the main goals should be getting both AC working + the txpower issue (stuck on 12dBm).

Some code in the v5.1.5 looks a bit dirty if you ask me.. like a chop shop with used parts

lhernanz, 2017-10-02:
Is there any real advantage of using v5.1.5 vs v4.3.21 as of today? As you say, code quality in v5.1.5 seems far from idea.

kimocoder, 2017-10-02:
Good question actually.. since txpower is stuck, im just staying on v4.3.21.
But work will be ongoing on all branches either way, just hoping to solve those issues soon

kimocoder, 2017-10-22:
@lhernanz this issue has been resolved :)

lhernanz, 2017-10-24:
Hi,
the new code is not working for me. I am now connecting at USB2.0 speeds again. The problem still is the `if (pHalData->EEPROMUsbSwitch == _FALSE)` comparison that you have uncommented in the last commit. If that comparison succeed, you get out of the method, therefore, you never get to execute the new code that you have added (that is working for me if I comment the mentioned statement as I was suggesting at the beginning of this thread).

My impression is that the variable `pHalData->EEPROMUsbSwitch` is not initialized in the 8814 HAL code as the only references that I can find to that variable are in the 8812 HAL code:

```C
hal/rtl8812a/usb/usb_halinit.c
2258:		pHalData->EEPROMUsbSwitch = _FALSE;
2261:		pHalData->EEPROMUsbSwitch = (PROMContent[EEPROM_USB_MODE_8812] & BIT1) >> 1;
2263:	RTW_INFO("Usb Switch: %d\n", pHalData->EEPROMUsbSwitch);
```

Regards

kimocoder, 2017-10-24:
Yes, personally haven't got time for testing at the moment.
@evilphish is working on a solution for it, opened the issue report again so the discussion may be seen by others too. Thanks

evilphish, 2017-10-24:
@lhernanz Hey there, my 8814 unit does not arrive until in 2 days or so, so I can not test this fix yet. Feel free to grab the sources from https://github.com/evilphish/rtl8812au (just committed a fix) and check whether it works now. If it works we can merge it back.

kimocoder, 2017-10-24:
Got a friend to test it, and the device (on 8814au) is now working inn SuperSpeed on bcdUSB 3.00.
The patch seem to work great! Will have those logs sent first thing in the morning but it finally seems to be resolved.

lhernanz, 2017-10-24:
Hi,
The @evilphish patch is working for me too!

I guess that we can close this one. We "only" need to make the AC mode to work and it will be almost perfect (have not checked if you have made any progress on that one lately). It is kind of crazy the mess that Realtek is creating and the low quality of the code.

Thanks for the help guys!

kimocoder, 2017-10-26:
Confirmed working for 8814, how about 8812? If that works, we'll closed it..

evilphish, 2017-10-26:
That's the thing. It is changing to USB3 mode but this results in a drop in reception.
You can actually see more networks with the 8812 if you plug it into an old USB2 rather than a USB3 port. This leads me to believe that something is not working right with the mode switching. Even though the device does do the switch and is then listed as a USB3 device. 

But I think we can close this and continue the discussion regarding the 8812 in the thread where this issue first manifested. https://github.com/aircrack-ng/rtl8812au/issues/29#issuecomment-338819141

kimocoder, 2017-10-26:
Closed. Continue om @evilphish issue report. Thanks

closed 2017-10-26 by kimocoder