aircrack-ng/rtl8812au #17: AWUS036ACH increasing TXPower doesn't increase speed or signal strength on ap

ShotTard, 2017-09-26:
OS: Kali

I'm running an AP with NetworkManager, and wanted to increase the signal strength.
But increasing TXPower from 15 to 30 then 32 has made no difference.

Am I missing something?

Thanks

kimocoder, 2017-09-26:
which branch are you using?

ShotTard, 2017-09-26:
v4.3.21

kimocoder, 2017-09-26:
the v4.3.21 branch should work fine, haven't got any other reports on that issue. what I know is that the v5.1.5 branch got a txpower issue, which I'm trying to figure out..

got 2 patches for v4.3.21 available for adding tomorrow, but that is a USB 3.0 fix & a LED addition patch.
but further testing/debug information would also be appreciated.

rawkahn, 2017-09-29:
Im having the same issue with 4.3.21 + AWUS036ACH. My onboard Marvel Avastar 350n set to 15dbm has a better signal. Txpower doesnt seem to do anything.

[image]

kimocoder, 2017-09-29:
Issue noted. Thanks for reporting

qwikwit, 2017-10-03:
When using network manager to create an AP I noticed that some devices not just realtek 8812/8814 do not respond to txpower changes. I'm experimenting with create_ap on github , you may want to give that a try.

On Managed mode or client mode, the signal level seems off compared to v5.1.5. The changes to txpower however can be seen on the host AP. This issue has been there since the astsam fork.

kimocoder, 2017-10-03:
Feel free to push changes to this repo if you got any changes that may fix those issues.
Been working on the v4.3.21 the past days, but all branches is maintained.

kimocoder, 2017-10-03:
[this](https://github.com/ulli-kroll/rtl8814au/issues/5#issuecomment-333788691) is also interesting reading on the AP issue

ghostindex, 2017-10-28:
I have this issue too with alfa AWUS036ACH bought from Taiwan.
When I got and configured the device using nm in debian stretch, I was once got an -5dBm signal strength on my phone. But after a while, the signal begin to fail. And it's never back.
By the way, I've plugged it on a windows 10. Maybe the official driver in windows updated the firmware? It's just my guess.
Now the issue is, does every one using this driver got this problem? I believe many people just forget to check the actual signal strength, because this card is really powerful even with default settings.

ghostindex, 2017-11-08:
Another thing. Is there any one who tried the unlocked driver in windows? If the txpower cannot be changed in windows, the problem might due to hardware limit, and unfortunately, maybe we should modify the hardware.
Anyway, I would like to give it a try.

evilphish, 2017-11-08:
I don't know to be honest but you can not change the txpower at all. Which also means you can not set it lower. Does not make a lot of sense to me as this is part of some energy saving mechanisms. Nonetheless, if you find something of interest please share :)

kimocoder, 2017-11-08:
Energy savings was turned off in Makefile, also regdomain settings have been checked/tested, but the driver supports 30dBm as "iw phy# info" shows.. this issue is propably the most important at the moment but could not find a solution for it on all these months. Help is highly appreciated. @evilphish has also spent some time on it, implementing the txpower control and stuff, but the txpower is still stuck in a position we want to increase.

ghostindex, 2017-11-10:
Bad news. I've tested it in windows and managed to set it on so-called 33dBm, but I only got 10dBm higher than the default 13dBm. I guess the hardware limit is at 24dBm.
And I found this:
https://forums.kali.org/showthread.php?28874-ALFA-AWUS036NHA-hacking-EEPROM-via-UART-JTAG&s=bd933ee3a7ff1c94d6a0703990e5962f
It seems the ALFA Inc has kept fooling us for a long time? My AWUS036ACH is from Taiwan, and maybe it's locked.

evilphish, 2017-11-12:
@ghostindex I do not seem to have this problem. My regulatory domain is not locked but I can freely change it. Also I do not just have a problem of the card not going over 20 dBm (or whatever it is set to) but it also makes no difference whether I set it to for example 3 dBm. So the power seems to always be the exact same value.

kimocoder, 2017-11-12:
After I started the v5.2.9 code porting, I've seen code that may be related to it. "pHalData" strings all over the places. Also seen lots of other interesting things, so getting this to work would fix lots of issues it seems.

Goga777, 2017-11-18:
I would like to clarify the question.

Is it normal to have error message as Operation is not permitted after command iwconfig interface txpower auto with my Edimax ew-7811dac based on 8811au with 5.1.5?

kimocoder, 2018-07-24:
This is probably related to #77 - closing this one for now

closed 2018-07-24 by kimocoder

azamet90, 2019-11-14:
i dont even have tx power :D 

```
wlan1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.43.39  netmask 255.255.255.0  broadcast 192.168.43.255
        inet6 fe80::9a28:e492:ac4:3f1f  prefixlen 64  scopeid 0x20<link>
        ether 0c:dd:24:2a:ff:76  txqueuelen 1000  (Ethernet)
        RX packets 2340  bytes 1517602 (1.4 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1953  bytes 367377 (358.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

kimocoder, 2019-11-14:
Yes you have, i haven't added output for it yet. "iw <adapter> info"

azamet90, 2019-11-14:
> Yes you have, i haven't added output for it yet. "iw info"

But still I cannot change my txpower thank you for fast reply first I did tried to change manually the txpower after I use the better bettercap and bettercap saying me unable to change txpower and like I said I tried manually too with set txpower fixed command but the txpower is not changing