aircrack-ng/rtl8812au #24: Issue with airodump-ng

Bluewild93, 2017-10-09:
I installed your driver, it works perfect for the first time when i put it in the monitor mode then sent DeAuth to my target devices. My target devices was disconnect from my wifi. But i couldn't get any handshakes. The second time, i couldn't get anything from the airodump-ng 
[image]

kimocoder, 2017-10-09:
just bring it down with:
ifconfig <adapter> down

then up again with:
ifconfig <adapter> up

in monitor mode, then it should work again..

Bluewild93, 2017-10-09:
do i need the newest version that you have now or just use it 4.3.21?

kimocoder, 2017-10-09:
v5.2.9 is for 8812au only, not the 8814au but it's not complete. But you could try the v5.1.5 :)

Bluewild93, 2017-10-09:
Thank you so much for your fast answer. 
I just wonder what course or program should people take to write a driver or program that you are doing now. Absolutely wonderful.

kimocoder, 2017-10-09:
credits still goes to "astsam", this is a fork from his repo with some extras added (like kernel support)

it's written in C

kimocoder, 2017-12-14:
The scanning issue has been resolved in latest "EFUSE fix" commits in v5.1.5 branch.
It has been tested and it seems to work fine noe. Closing.

closed 2017-12-14 by kimocoder