aircrack-ng/rtl8812au #29: AWSU036ACH (AC1200) stuck on USB2 and 12 dBm

evilphish, 2017-10-19:
Greetings,

my AC1200 adapter is stuck in USB2 mode and completely ignores and txpower setting.

I am running Linux Mint 18.2 with a 4.11.0-14 kernel. 
I have tried the 4.3.21, 5.1.5 and 5.2.9 version of the rtl8812au dkms module.
From the debug info I can not see any specific part that points to the module trying to switch the device over to USB3.

Appart from the txpower and the usb setting the adapter works. I can enter monitor mode and injection works as well as long as I stick to the 2.4 GHz band. However the range is severely limited with 12 dBm txpower and I get the feeling this has something to do with the adapter not being able to enter USB3 mode. 

I will attach the debug messages, if I can provide any further info I would be happy to do so.

`
RTW: module init start
RTW: 8812au v5.1.5_19247.20160830
RTW: build time: Oct 19 2017 02:02:09
RTW: usb_endpoint_descriptor(0):
RTW: bLength=7
RTW: bDescriptorType=5
RTW: bEndpointAddress=81
RTW: wMaxPacketSize=512
RTW: bInterval=0
RTW: RT_usb_endpoint_is_bulk_in = 1
RTW: usb_endpoint_descriptor(1):
RTW: bLength=7
RTW: bDescriptorType=5
RTW: bEndpointAddress=2
RTW: wMaxPacketSize=512
RTW: bInterval=0
RTW: RT_usb_endpoint_is_bulk_out = 2
RTW: usb_endpoint_descriptor(2):
RTW: bLength=7
RTW: bDescriptorType=5
RTW: bEndpointAddress=3
RTW: wMaxPacketSize=512
RTW: bInterval=0
RTW: RT_usb_endpoint_is_bulk_out = 3
RTW: usb_endpoint_descriptor(3):
RTW: bLength=7
RTW: bDescriptorType=5
RTW: bEndpointAddress=4
RTW: wMaxPacketSize=512
RTW: bInterval=0
RTW: RT_usb_endpoint_is_bulk_out = 4
RTW: usb_endpoint_descriptor(4):
RTW: bLength=7
RTW: bDescriptorType=5
RTW: bEndpointAddress=85
RTW: wMaxPacketSize=64
RTW: bInterval=1
RTW: RT_usb_endpoint_is_int_in = 5, Interval = 1
RTW: nr_endpoint=5, in_num=2, out_num=3

RTW: USB_SPEED_HIGH
RTW: CHIP TYPE: RTL8812
RTW: read_chip_version_8812a SYS_CFG(0xF0)=0x04411137
RTW: rtw_hal_config_rftype RF_Type is 2 TotalTxPath is 2
RTW: Chip Version Info: CHIP_8812_Normal_Chip_TSMC_C_CUT_2T2R_RomVer(0)
RTW: _ConfigChipOutEP_8812 OutEpQueueSel(0x07), OutEpNumber(3)
RTW: Boot from EFUSE, Autoload OK !
lxMAC: CTRL-EVENT-SCAN-FAILED ret=-19 retry=1
RTW: HW EFUSE
RTW: eFuse Content:
(lots of stuff)
RTW: EEPROM ID=0x8129
RTW: VID = 0x0BDA, PID = 0x8812
RTW: Customer ID: 0x00, SubCustomer ID: 0xCD
RTW: EEPROMRegulatory = 0x0
RTW: Board Type: 0x 0
RTW: hal_com_config_channel_plan chplan:0x7F
RTW: CrystalCap: 0x1c
RTW: ThermalMeter = 0x20
RTW: Hal_ReadRemoteWakeup_8812A...bSupportRemoteWakeup(0)
RTW: pHalData->PAType_2G is 0x33, pHalData->ExternalPA_2G = 1
RTW: pHalData->PAType_5G is 0x33, pHalData->ExternalPA_5G = 1
RTW: pHalData->LNAType_2G is 0x88, pHalData->ExternalLNA_2G = 1
RTW: pHalData->LNAType_5G is 0x88, pHalData->ExternalLNA_5G = 1
RTW: pHalData->TypeGPA = 0x0
RTW: pHalData->TypeAPA = 0x0
RTW: pHalData->TypeGLNA = 0x0
RTW: pHalData->TypeALNA = 0x0
RTW: RFE Type: 0x 3
RTW: Usb Switch: 1
RTW: PID= 0x8812, VID=  bda
RTW: PID= 0x8812, VID=  bda
RTW: Customer ID: 0x 0
RTW: hal_ReadUsbType_8812AU: antenna=2, wmode=3
RTW: hal_ReadUsbType_8812AU(): EFUSE_HIDDEN_812AU
RTW: rtw_hal_read_chip_info in 128 ms
RTW: init_channel_set((null)) ChannelPlan ID:0x7f, ch num:37
RTW: NR_RECVBUFF: 8
RTW: MAX_RECVBUF_SZ: 32768
RTW: NR_PREALLOC_RECV_SKB: 8
RTW: rtw_alloc_macid((null)) if1, hwaddr:ff:ff:ff:ff:ff:ff macid:1
RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
RTW: phy_ConfigBBWithPgParaFile(): No File PHY_REG_PG.txt, Load from HWImg Array!
RTW: default power by rate loaded
RTW: can't get autopm:
RTW: rtw_macaddr_cfg mac addr:00:c0:ca:95:e8:a9
RTW: bDriverStopped:True, bSurpriseRemoved:False, bup:0, hw_init_completed:0
RTW: rtw_wiphy_alloc(phy19)
RTW: rtw_wdev_alloc(padapter=ffff9cdb81fd5000)
RTW: rtw_wiphy_register(phy19)
RTW: Register RTW cfg80211 vendor cmd(0x67) interface
RTW: rtw_ndev_init(wlan0) if1 mac_addr=00:c0:ca:95:e8:a9
RTW: rtw_ndev_notifier_call(wlan0) state:16
RTW: cfg80211_rtw_get_txpower
RTW: rtw_ndev_notifier_call(wlan0) state:5
usbcore: registered new interface driver 8812au
RTW: module init ret=0
RTW: cfg80211_rtw_get_txpower
`

going for a iwconfig device txpower 20 results in:
RTW: cfg80211_rtw_set_txpower
and nothing happens.

cheers

kimocoder, 2017-10-19:
Hi there!

I'm aware of the issues you mention and are looking into ut.

First off,  the 12dBm issue.. the "rtw_get_txpower" is hardcoded, so the 12dBm is not a reality in itself.

Second, a patch was commited to fix the USB3 modeswitch not long ago, could you show a complete re-produce then I may got a solution for it.

Do you build from github or simply use the "realtek-rtl88xxau-dkms" inn Kali (which is not up to date yet?)

Thanks.

evilphish, 2017-10-19:
Hey kimo,

all versions I built from github yesterday evening so unless the usb3 fix was comitted overnight I would assume that I run the up-to-date version.

I can reproduce this every time. If you need more specific information I can get back to you this evening.

cheers
phish

kimocoder, 2017-10-19:
It's in the commits, but marked "needs more testing", but clearly does not seem to have any effects.
Will have a looking at it the next days/weeks, but any help would be highly appreciated.

evilphish, 2017-10-19:
Get back to me whenever you like. Happy to help if I can.

taleru, 2017-10-19:
Hi! Same for here, with 5.1.5 and 5.2.9, don't work on USB 3.0 for AWUS036ACH on Debian, Kernel 4.12

lsusb
Bus 004 Device 002: ID 058f:6366 ***** Multi Flash Reader
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 002: ID 0930:0220 Toshiba Corp. 
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

kimocoder, 2017-10-19:
Thanks for the information. This issue should be in focus from now om.

taleru, 2017-10-20:
Hello, I have changed the file os_dep/linux/usb_intf.c, adapting the version 4.3.2 to the 5.1.5 and respecting the hardware of the 5.1.5. Now the USB 3.0 works for me (AWSU036ACH ) on 5.1.5 . Below I leave you the file so they can test

﻿/******************************************************************************
 *
 * Copyright(c) 2007 - 2011 Realtek Corporation. All rights reserved.
 *
 * This program is free software; you can redistribute it and/or modify it
 * under the terms of version 2 of the GNU General Public License as
 * published by the Free Software Foundation.
 *
 * This program is distributed in the hope that it will be useful, but WITHOUT
 * ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or
 * FITNESS FOR A PARTICULAR PURPOSE. See the GNU General Public License for
 * more details.
 *
 * You should have received a copy of the GNU General Public License along with
 * this program; if not, write to the Free Software Foundation, Inc.,
 * 51 Franklin Street, Fifth Floor, Boston, MA 02110, USA
 *
 *
 ******************************************************************************/
#define _HCI_INTF_C_

#include <drv_types.h>
#include <hal_data.h>

#include <platform_ops.h>

#ifndef CONFIG_USB_HCI
#error "CONFIG_USB_HCI shall be on!\n"
#endif

#if defined (PLATFORM_LINUX) && defined (PLATFORM_WINDOWS)
#error "Shall be Linux or Windows, but not both!\n"
#endif

#ifdef CONFIG_80211N_HT
extern int rtw_ht_enable;
extern int rtw_bw_mode;
extern int rtw_ampdu_enable;//for enable tx_ampdu
#endif

#ifdef CONFIG_GLOBAL_UI_PID
int ui_pid[3] = {0, 0, 0};
#endif

extern int pm_netdev_open(struct net_device *pnetdev,u8 bnormal);
static int rtw_suspend(struct usb_interface *intf, pm_message_t message);
static int rtw_resume(struct usb_interface *intf);

static int rtw_drv_init(struct usb_interface *pusb_intf,const struct usb_device_id *pdid);
static void rtw_dev_remove(struct usb_interface *pusb_intf);

static void rtw_dev_shutdown(struct device *dev)
{
	struct usb_interface *usb_intf = container_of(dev, struct usb_interface, dev);
	struct dvobj_priv *dvobj = NULL;
	_adapter *adapter = NULL;
	int i;

	RTW_INFO("%s\n", __func__);

	if(usb_intf){
		dvobj = usb_get_intfdata(usb_intf);
		if (dvobj){
			rtw_set_surprise_removed(dvobj->padapters[IFACE_ID0]);
			ATOMIC_SET(&dvobj->continual_io_error, MAX_CONTINUAL_IO_ERR+1);
		}
	}
}
[... 1538 lines cut ...]
	usb_drv.drv_registered = _FALSE;

	usb_deregister(&usb_drv.usbdrv);

	platform_wifi_power_off();

	rtw_suspend_lock_uninit();
	rtw_drv_proc_deinit();
	rtw_ndev_notifier_unregister();

	RTW_PRINT("module exit success\n");

	rtw_mstat_dump(RTW_DBGDUMP);
}

module_init(rtw_drv_entry);
module_exit(rtw_drv_halt);

#ifdef CONFIG_INTEL_PROXIM
_adapter  *rtw_usb_get_sw_pointer(void)
{
	return rtw_sw_export;
}
EXPORT_SYMBOL(rtw_usb_get_sw_pointer);
#endif	//CONFIG_INTEL_PROXIM

kimocoder, 2017-10-20:
@evilphish this would need some testing..

kimocoder, 2017-10-20:
I've also found some interesting commits over [here](https://github.com/mk-fg/rtl8812au/commits/master) that may be used. All saved for this weekend to investigate.

kimocoder, 2017-10-20:
@taleru this file is actually unusable in Github, as the star " * " gets messed up, everything does.
could you upload the file somewhere so I could test it? just paste the link here

evilphish, 2017-10-20:
Actually, all that seems to be needed in order to get USB3 working is to copy the specific define block for the 8812au driver in the corresponding usb function. This fits well with the adapter having to get a certain byte sequence in order to reconnect as a superspeed USB3 device.

I have not tested this in 5.2.9 yet.

```diff
--- 515old/os_dep/linux/usb_intf.c	2017-10-20 23:00:13.605117029 +0200
+++ rtl8812au-kimo-5.1.5/os_dep/linux/usb_intf.c	2017-10-20 21:52:03.153703361 +0200
@@ -665,7 +665,25 @@
 	if (registry_par->switch_usb_mode == 0)
 		goto exit;
 
+#if defined(CONFIG_RTL8812A) || defined(CONFIG_RTL8821A)
+        if (IS_HIGH_SPEED_USB(Adapter)) {
+          if ((rtw_read8(Adapter, 0x74) & (BIT(2)|BIT(3))) != BIT(3)) {
+            rtw_write8(Adapter, 0x74, 0x8);
+            rtw_write8(Adapter, 0x70, 0x2);
+            rtw_write8(Adapter, 0x3e, 0x1);
+            rtw_write8(Adapter, 0x3d, 0x3);
+            /* usb disconnect */
+            rtw_write8(Adapter, 0x5, 0x80);
+            ret = _TRUE;
+          }
+
+        } else if (IS_SUPER_SPEED_USB(Adapter)) {
+          rtw_write8(Adapter, 0x70, rtw_read8(Adapter, 0x70) & (~BIT(1)));
+          rtw_write8(Adapter, 0x3e, rtw_read8(Adapter, 0x3e) & (~BIT(0)));
+        }
+#else
 	rtw_hal_set_hwreg(Adapter, HW_VAR_USB_MODE, &ret);
+#endif
 
 exit:
 	return ret;

```

evilphish, 2017-10-20:
Update, I can confirm the same patch works for 5.2.9

/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=rtl8812au, 5000M

kimocoder, 2017-10-20:
Feel free to push those commits, as credits will also would be signed to you.

kimocoder, 2017-10-20:
We may take the 'work' over [here](https://github.com/kimocoder/rtl8812au) for testing so that this one don't get messy (like I did in some reverts earlier)

taleru, 2017-10-20:
Can I up the file here, in some place?

taleru, 2017-10-20:
[usb_intf.c.txt](https://github.com/aircrack-ng/rtl8812au/files/1403404/usb_intf.c.txt)

kimocoder, 2017-10-20:
Yes, I can see it contains the same patch that @evilphish just provided there @taleru :)
It has been added to github.com/kimocoder/rtl8812au v5.1.5 branch at the moment, but does this fix both 8812 & 8814 or just the first one?

taleru, 2017-10-20:
@kimocoder and @evilphish, the scope of the antenna remains low. The antenna has less range than a wifi antenna conventional.

taleru, 2017-10-20:
I think that the both, but I haven't 8814 chip wifi for testing.

kimocoder, 2017-10-20:
Notice: MaxPower 500mA
lsusb -vvv Output:

Bus 003 Device 034: ID 0bda:8813 Realtek Semiconductor Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x8813 
  bcdDevice            0.00
  iManufacturer           1 Realtek
  iProduct                2 802.11ac NIC
  iSerial                 3 123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           53
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled

kimocoder, 2017-10-20:
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 2: Dev 34, If 0, Class=Vendor Specific Class, Driver=8814au, 480M
    |__ Port 4: Dev 29, If 0, Class=Communications, Driver=, 480M
    |__ Port 4: Dev 29, If 1, Class=Communications, Driver=cdc_acm, 480M
    |__ Port 4: Dev 29, If 2, Class=CDC Data, Driver=cdc_acm, 480M
    |__ Port 4: Dev 29, If 3, Class=Communications, Driver=cdc_acm, 480M
    |__ Port 4: Dev 29, If 4, Class=CDC Data, Driver=cdc_acm, 480M
    |__ Port 4: Dev 29, If 5, Class=Communications, Driver=cdc_wdm, 480M
    |__ Port 4: Dev 29, If 6, Class=Communications, Driver=cdc_ncm, 480M
    |__ Port 4: Dev 29, If 7, Class=CDC Data, Driver=cdc_ncm, 480M
    |__ Port 4: Dev 29, If 8, Class=Communications, Driver=cdc_wdm, 480M
    |__ Port 4: Dev 29, If 9, Class=Communications, Driver=cdc_acm, 480M
    |__ Port 4: Dev 29, If 10, Class=CDC Data, Driver=cdc_acm, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/3p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
        |__ Port 6: Dev 3, If 0, Class=Chip/SmartCard, Driver=, 12M
        |__ Port 8: Dev 16, If 0, Class=Hub, Driver=hub/7p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/3p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
        |__ Port 4: Dev 3, If 1, Class=Vendor Specific Class, Driver=btusb, 12M
        |__ Port 4: Dev 3, If 2, Class=Vendor Specific Class, Driver=btusb, 12M
        |__ Port 4: Dev 3, If 0, Class=Vendor Specific Class, Driver=btusb, 12M
        |__ Port 4: Dev 3, If 3, Class=Application Specific Interface, Driver=, 12M
        |__ Port 6: Dev 4, If 0, Class=Video, Driver=uvcvideo, 480M
        |__ Port 6: Dev 4, If 1, Class=Video, Driver=uvcvideo, 480M

evilphish, 2017-10-20:
For me it is max power 200 mA but it says that it is connected via usb3

Try adding a defined(CONFIG_RTL8814A) to the #if stanza in my patch. I guess at the moment the block is just not compiled for your module.

Bus 002 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x8812 RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
  bcdDevice            0.00
  iManufacturer           1 Realtek
  iProduct                2 802.11n NIC
  iSerial                 3 123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           83
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              200mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes           19
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Feedback
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
        bMaxBurst               0
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength           22
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat         255 micro seconds
Device Status:     0x0000
  (Bus Powered)

dfiloni, 2017-10-20:
@kimocoder : evilphish patch doesn't work on 8814 due to:
`#if defined(CONFIG_RTL8812A) || defined(CONFIG_RTL8821A)`
however this is weird because the same code is in:
https://github.com/aircrack-ng/rtl8812au/blob/v5.1.5/hal/rtl8812a/usb/usb_halinit.c#L2431
so the issue is probably with: registry_par->switch_usb_mode ( module param: rtw_switch_usb_mode ), as per hal/*/usb/usb_halinit.c it should be == 1 for USB3

kimocoder, 2017-10-20:
Yeah I know that.. I added changes to mye private fork [here](https://github.com/kimocoder/rtl8812au) and added the "|| defined (CONFIG_8814A)" behind 8812A & 8821A.

dfiloni, 2017-10-20:
@kimocoder could you try (the patch is not required): sudo modprobe 8814au rtw_switch_usb_mode=1
@evilphish: the same should work for you without he patch: sudo modprobe 8812au rtw_switch_usb_mode=1

evilphish, 2017-10-20:
Yes it works.

kimocoder, 2017-10-20:
tested. did see the hardcoded parametrer (and many other ones) in modinfo, but for 8814au is still remains inn bcdUSB 2.00 on 480M. but ut seems the 8812au is working at least.. and if it sets USB3 (superspeed) om plugin without using the param, that would be great. Finding the sheets for now, but great job om both of you guys! Really appreciated.

There drivers is a mix og old/new code and they absolutely introduced some issues ..

taleru, 2017-10-21:
Hi ! @kimocoder , the new file usb_intf.c don't work for me on USB 3.0 because you don't comment this lines too:

	/* registry not allow driver to switch usb mode */
	//      if (registry_par->switch_usb_mode == 0)
	//	goto exit;

With that and #if defined(CONFIG_RTL8812A) || defined(CONFIG_RTL8821A)... the USB 3.0 run ok on 5.1 and 5.2, at least for me on AWSU036ACH.

Can you comment /* registry not allow driver to switch usb mode */ ... again ? 

Thanks you for the driver!

kimocoder, 2017-10-21:
@evilphish have been working on this today, credits goes hos way so maybe he could investigate that one

evilphish, 2017-10-21:
@taleru the most recent commit should take care of that as we changed the default mode to switch to usb3. If you still have the older sources you need to use the rtw_switch_usb_mode=1 module parameter.

For me on a AWSU036ACH it works fine now. Did you just check out the sources?

taleru, 2017-10-21:
@evilphish It's ok, working now on USB 3.0 with rtw_switch_usb_mode=1 parameter. Thanks you!

evilphish, 2017-10-22:
If someone else has the AWSU036ACH dongle it would be awesome if you can check whether setting different txpower values actually change something in regards to the reception and not just the numeric values in the iwconfig readout.

It is working for kimo with his 8814 model but I can not see a difference in reception quality or signal strength with my 8812 model regardless of whether I set 5 dBm or 30 dBm.

kimocoder, 2017-10-22:
Having a look at other repos txpower control would be great for learning/understanding purposes, the one from e.g v4.3.21 branch is dirty for sure.

taleru, 2017-10-22:
Yes, thanks you @evilphish, really is better signal tx now in my AWSU036ACH, but not appreciated very much, just a little better.

evilphish, 2017-10-22:
@taleru how many dB of difference do you get between a txpower of 30 and a txpower of lets say 5?

taleru, 2017-10-22:
Three metres from the AP. 
With 30dBm of txpower >> -32dBm
With 5dBm of txpower >> -42dBm

difference = 10dBm

evilphish, 2017-10-22:
Hmm and how stable are the values? For me the signal strength fluctuates by +-10 dBm so a difference of 10 dBm would be noise in my oppinion. Is your -32 and -42 quite stable in itself?

taleru, 2017-10-22:
Yes, quite stable

taleru, 2017-10-22:
for 5dB ranges between -42dBm and -48dBm

taleru, 2017-10-22:
It is difficult to see signals of 20m or more distance with airodump. The txpower to 30dBm no seem 30dBm real.

aircrack-ng, 2017-10-22:
@taleru, just FYI, TX power plays no role whatsoever in receiving (hence why it's called "TX power" as in transmit power). For receiving, it's RX sensitivity

evilphish, 2017-10-22:
Yes, 30dBm should be 1 W of power. That is quite a lot. For me certain networks range between 49 dB and 55 dB and this does not change whether I am at a txpower of 5 or 30. 
Kimo hat a super high increase as I said 30 to 40 db, which is what I would imagine between a txpower of 5 and 30. 5 dB is only like 3 mW. So we are talking a difference in power of a factor of roughly 330 when changing from 5 dB to 30 dB. This is something that you should clearly notice :)

taleru, 2017-10-22:
Then, under rx, although an antenna, in theory having more power transmission, increases the reception.

taleru, 2017-10-22:
I don't know exactly what the problem is and I understand that potency varies. But it is also true that very short distance, the card works great but in the middle distance, 20 meters or more, it becomes very unstable.

taleru, 2017-10-22:
For example, many times with the AWSU036ACH I can only see my AP. With any other antenna conventional I can see my AP, and others. It is true that if it caught a signal in the AWSU036ACH and the other antenna conveencional, the AWSU036ACH shows the signal with the best quality. But why is it so difficult for the AWSU036ACH get to see other Aps ?

kimocoder, 2017-10-22:
HT Greenfield (RX Greenfield) capabilities has been added. 
Also found another VHT issue, will take a look on that one tomorrow, think I se a fix for it.

taleru, 2017-10-22:
There must be something wrong in the settings of the USB 3.0 . When I use the antenna AWSU036ACH with USB 2.0 I can see networks with 50dB, 70dB, 80dB, etc, In change, if I use USB 3.0 I can only see my router and sometimes, I don't see or if you want to my router with airodump.
That is the problem that I referred to before with the power.

evilphish, 2017-10-23:
@taleru I have just forced my AWSU036ACH into USB2 mode and I do not notice any difference at all. Neither in number if ssids discovered, nor in signal strength.

kimocoder, 2017-10-23:
Still a long way to go on these drivers for sure..

taleru, 2017-10-23:
@evilphish For me always, when I plug in USB 2.0 ports seee all nets, when I plug in USB 3.0 port just see one net (my router) or not one net. (In monitor and managed mode same). Try without forced, with real ports.

evilphish, 2017-10-23:
@taleru That did the trick. I get a similar effect. I see more low-power networks if using a hardware USB2 port. However the power levels for the networks that were visible before stay the same.
Interesting find.... So we still have to look more into USB. I get the feeling this USB2/USB3 switcheroo Realtek implemented in this chip is completely bonkers...

taleru, 2017-10-24:
@evilphish @kimocoder The same thing happens in version 4.3.21, with USB 3.0 does not work properly the antenna. I've only been able to find this about it, but I don't know if you can help.

https://github.com/abperiasamy/rtl8812AU_8821AU_linux/issues/229

taleru, 2017-10-26:
I search that, but don't fix for me.
The signal level is only cosmetic. If you want to mak4e it "look" better, then apply the following patch:
Index: rtl8188eu/os_dep/ioctl_linux.c

By  lwfinger 

--- rtl8188eu.orig/os_dep/ioctl_linux.c
+++ rtl8188eu/os_dep/ioctl_linux.c
@@ -7830,7 +7830,7 @@ static struct iw_statistics *rtw_get_wir
tmp_qual = padapter->recvpriv.signal_qual;
tmp_noise = padapter->recvpriv.noise;

              piwstats->qual.level = tmp_level;

              piwstats->qual.level = 80; //tmp_level;
            piwstats->qual.qual = tmp_qual;
            piwstats->qual.noise = tmp_noise;
    }

https://github.com/lwfinger/rtl8188eu/issues/35

rawkahn, 2017-10-28:
I have similar issues with my surface pro 2, running Kali dual-boot. Single USB 3.0 connection, unstable AP list, poor if any reception etc, yet my onboard wifi has zero problems. On another forum, some guy ditched Alfa's cable & runs his device on a regular micro-USB cable so its at least stable & functional in USB 2.0. Possibly a conflict between USB 2/3 within the driver? Talk about a headache.