aircrack-ng/rtl8812au #31: Access point setup fails if enabling ac mode

evilphish, 2017-10-19:
Using Debian GNU/Linux  Kernel 4.12.0-0.bpo.2-amd64 compiled version 4.3.21 from github.
Adapter AWSU036ACH (AC1200) using rtl8812au

Setting up an access point in 802.11n mode with hostapd works fine as far as I can see. Switching to ac mode however results in "Could not set channel for kernel driver".
I actually do not know whether this functionality is already part of this driver. I would be more than happy to offer additional information if requested or test patches.

**Log output from hostapd:**
random: Trying to read entropy from /dev/random
Configuration file: /etc/hostapd/hostapd.conf
ctrl_interface_group=0
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported vendor command: vendor_id=0x1018 subcmd=1
nl80211: Supported vendor command: vendor_id=0x1a11 subcmd=4106
nl80211: Supported vendor command: vendor_id=0x1a11 subcmd=4107
nl80211: Supported vendor event: vendor_id=0x1018 subcmd=0
nl80211: Supported vendor event: vendor_id=0x1018 subcmd=1
nl80211: interface wlan0 in phy phy1
nl80211: Set mode ifindex 15 iftype 3 (AP)
nl80211: Setup AP(wlan0) - device_ap_sme=1 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0x55791746b2d0 (device SME)
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55791746b2d0 match=04
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55791746b2d0 match=0501
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55791746b2d0 match=0503
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55791746b2d0 match=0504
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55791746b2d0 match=06
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55791746b2d0 match=08
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55791746b2d0 match=09
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55791746b2d0 match=0a
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55791746b2d0 match=11
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55791746b2d0 match=7f
nl80211: Enable Probe Request reporting nl_preq=0x55791746c960
nl80211: Register frame type=0x40 (WLAN_FC_STYPE_PROBE_REQ) nl_handle=0x55791746c960 match=
rfkill: initial event: idx=1 type=1 op=0 soft=0 hard=0
nl80211: Add own interface ifindex 23 (ifidx_reason 15)
nl80211: if_indices[16]: 23(15)
nl80211: Add own interface ifindex 15 (ifidx_reason -1)
nl80211: if_indices[16]: 23(15) 15(-1)
nl80211: Adding interface wlan0 into bridge br0
phy: phy1
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Previous country code DE, new country code DE 
nl80211: Regulatory information - country=DE (DFS-ETSI)
nl80211: 2400-2483 @ 40 MHz 20 mBm
nl80211: 5150-5250 @ 80 MHz 20 mBm (no outdoor)
nl80211: 5250-5350 @ 80 MHz 20 mBm (no outdoor) (DFS)
nl80211: 5470-5725 @ 160 MHz 26 mBm (DFS)
nl80211: 5725-5875 @ 80 MHz 13 mBm
nl80211: 57000-66000 @ 2160 MHz 40 mBm
nl80211: Added 802.11b mode based on 802.11g information
Allowed channel: mode=1 chan=1 freq=2412 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=2 freq=2417 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=3 freq=2422 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=4 freq=2427 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=5 freq=2432 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=6 freq=2437 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=7 freq=2442 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=8 freq=2447 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=9 freq=2452 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=10 freq=2457 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=11 freq=2462 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=12 freq=2467 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=13 freq=2472 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=14 freq=2484 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=34 freq=5170 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=36 freq=5180 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=38 freq=5190 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=40 freq=5200 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=42 freq=5210 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=44 freq=5220 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=46 freq=5230 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=48 freq=5240 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=144 freq=5720 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=149 freq=5745 MHz max_tx_power=13 dBm
Allowed channel: mode=2 chan=153 freq=5765 MHz max_tx_power=13 dBm
Allowed channel: mode=2 chan=157 freq=5785 MHz max_tx_power=13 dBm
Allowed channel: mode=2 chan=161 freq=5805 MHz max_tx_power=13 dBm
Allowed channel: mode=2 chan=165 freq=5825 MHz max_tx_power=13 dBm
Allowed channel: mode=2 chan=169 freq=5845 MHz max_tx_power=13 dBm
Allowed channel: mode=2 chan=0 freq=5920 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=23441 freq=5940 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=0 freq=5960 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=23361 freq=5980 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=0 freq=6000 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=23281 freq=6020 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=0 freq=6040 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=23201 freq=6060 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=0 freq=6080 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=1 freq=2412 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=2 freq=2417 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=3 freq=2422 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=4 freq=2427 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=5 freq=2432 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=6 freq=2437 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=7 freq=2442 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=8 freq=2447 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=9 freq=2452 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=10 freq=2457 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=11 freq=2462 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=12 freq=2467 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=13 freq=2472 MHz max_tx_power=20 dBm
Allowed channel: mode=0 chan=14 freq=2484 MHz max_tx_power=20 dBm
hw vht capab: 0x10b0, conf vht capab: 0x0
DFS support is enabled
Completing interface initialization
Mode: IEEE 802.11a  Channel: 36  Frequency: 5180 MHz
DFS 0 channels required radar detection
**Could not set channel for kernel driver**
Interface initialization failed
wlan0: interface state COUNTRY_UPDATE->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
hostapd_interface_deinit_free(0x55791746a950)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0x55791746a950)
wlan0: interface state DISABLED->DISABLED
hostapd_bss_deinit: deinit bss wlan0
wlan0: Deauthenticate all stations
nl80211: sta_remove -> DEL_STATION wlan0 ff:ff:ff:ff:ff:ff --> -22 (Invalid argument)
wlan0: AP-DISABLED 
hostapd_cleanup(hapd=0x55791746bc50 (wlan0))
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started
hostapd_interface_deinit_free: driver=0x557915b9c960 drv_priv=0x55791746c7d0 -> hapd_deinit
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Disable Probe Request reporting nl_preq=0x8888ddf19fce41e9
nl80211: Remove monitor interface: refcount=0
nl80211: Remove beacon (ifindex=15)
netlink: Operstate: ifindex=15 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 15 iftype 2 (STATION)
nl80211: Teardown AP(wlan0) - device_ap_sme=1 use_monitor=0
nl80211: Unsubscribe mgmt frames handle 0x8888ddf19fce3a59 (AP teardown (dev SME))
hostapd_interface_free(0x55791746a950)
hostapd_interface_free: free hapd 0x55791746bc50
hostapd_cleanup_iface(0x55791746a950)
hostapd_cleanup_iface_partial(0x55791746a950)
hostapd_cleanup_iface: free iface=0x55791746a950

**hostapd.conf**
interface=wlan0
bridge=br0
driver=nl80211
ssid=SOMETHING
country_code=DE
ieee80211d=1
ieee80211h=1
hw_mode=a
channel=36
wmm_enabled=1
ieee80211n=1
require_ht=1
ieee80211ac=1
require_vht=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
wpa=2
wpa_passphrase=something
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP

Uncommenting ieee80211ac=1 results in the access point being created just fine, albeit in 802.11n mode only.

evilphish, 2017-10-19:
Some further information and I am not sure whether this is a problem of this driver or hostapd but I am leaning towards the former:

It seems the setup of the hostap fails because it *thinks* that the wireless adapter does not provide certain capabilities even though it does. An example:

iw list reports
VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9

This tells us that we have 2 supported streams which means we can use 
[RX-STBC-12]
as a configuration flag for the vht capabilities config in hostapd. The problem is that while iw list can obviously report the correct capabilities of the card, when hostapd asks the 8812 driver for its capabilities the driver reports that the card does not have this capability. If I take the source of hostapd and remove the check whether a card actually has the configured capabilities the access point starts up fine in ac mode. However, this is quite a dirty hack and apparently with other drivers hostapd has no difficulties when asking for the capabilities of a card. The same problem also occurs with ht capabilities, nost just vht.

kimocoder, 2017-10-19:
Thanks! again, great information! I've seen a commit in another repo with some fixes for VHT, compared it to this code and lots seemed missing here. The weekend is set to investigate further, I'll post the commit from the repo I've when I'm back at my computer then you'll see the issue I REF too.

The HT caps I didn't know about though.

These drivers is a mix of old/new code and gots some issues that has to be resolved.
I'll get back to you this weekend with some VHT info. That's a starter.

kimocoder, 2017-10-19:
[here](https://github.com/diederikdehaas/rtl8812AU/pull/14) is some interesting readings about the issue.

evilphish, 2017-10-19:
Sure go ahead. Also when checking my card with iw list, the vht capabilities are reported as
VHT Capabilities (0x000010b0):
			Max MPDU length: 3895
			Supported Channel Width: neither 160 nor 80+80
			RX LDPC
			short GI (80 MHz)
			TX STBC
			SU Beamformee

*technically* there is nothing here that points to RX STBC 12, it just mentions TX STBC. Of course in my last post I showed an excerpt of the following info block in iw list which goes:
VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 866 Mbps

and this clearly shows that there are 2 streams available. I don't know whether hostapd maybe expects the information regarding the streams as a flag in the VHT capabilities block. This also seems to be somewhat of a general issue as you can find loads of example or tutorial posts which show iw list outputs with more than 1 stream and the corresponding example hostapd config only using RX-STBC-1 which is supposed to disallow usage of more than one stream. 

I am at a total loss whether the RX-STBC-12 is supposed to be included in the VHT capabilities block but I guess it would make sense to have *all* capabilities as flags in one place and not spread the information accross different blocks of information.

Either way if you need further stuff or want to do some testing on the weekend I am game.

Oh and regarding your interesting link, it seems like they introduced functions to basically prime the card in order to activate the capabilities? I think I read somewhere that some cards do not advertise certain capabilities unless you activate them which is pretty problematic if a piece of software only enables capabilities that are advertised :) It's like a wireless mexican standoff ^^

kimocoder, 2017-10-19:
[image]

Actually, look at this one. Try changing these, it's in the Makefile.

evilphish, 2017-10-19:
compiles with a warning 
WARNING: "mpt_ProSetPMacTx" [/root/rtl8812au-kimo-4.3.21/8812au.ko] undefined!

which leads to an error when trying to modprobe 8812au
modprobe: ERROR: could not insert '8812au': Unknown symbol in module, or unknown parameter (see dmesg)

and dmesg complains:
8812au: Unknown symbol mpt_ProSetPMacTx (err 0)

kimocoder, 2017-10-19:
Ok. it works om the v5.1.5 branch a user confirms, but he got no knowledge on testing.
v4.3.21 branch hmmm.. unsure on the future support for it, as the main work is noe focused on v5.1.5.

I'll uploads the info he sends me in short time

Simply can't wait to get back home.

kimocoder, 2017-10-19:
This is the 8814au after enabling "CONFIG_MP_VHT_HW_TX_MODE" in Makefile.

[iw_phy_new.txt](https://github.com/aircrack-ng/rtl8812au/files/1400146/iw_phy_new.txt)

evilphish, 2017-10-19:
Interesting, vht caps say nothing about the rx-stbc besides listing 4 streams in the next block. Which would mean one should be able to use RX-STBC-1234 as a flag in hostapd and I would bet that it will complain about missing capabilities in that case... 

I will have to hit the sack for today anyway :) Maybe, in combination with the other issue I opened, if we can get the usb3 stuff fixed in 5.1.5 I would be more than happy to switch sources and continue the hunt on 5.1.5 if you are interested. I also do not see a big gain from pouring too much work into 4.3.21 but alas, it is the most feature complete as of now I guess.

kimocoder, 2017-10-19:
Great sir!

We'll get back to it. Got some intel on the USB3 stuff (EEPROM USBModeSwitch) too.
Have a good one in the meantime & thanks.

evilphish, 2017-10-20:
Now that the USB3 part apparently is solved, I have bad news for this issue. Even with 5.1.5 or 5.2.9 the module is not loadable:
8812au: Unknown symbol mpt_ProSetPMacTx (err 0)

This might be related to you using the 8814 driver and not the 8812. 
I am currently trying to find out where this gets defined but grep is of no big help. I will keep looking but maybe you have more luck :)

dfiloni, 2017-10-20:
> I am currently trying to find out where this gets defined but grep is of no big help. I will keep looking but maybe you have more luck :)
You can find it here: https://github.com/aircrack-ng/rtl8812au/blob/v5.1.5/hal/hal_mp.c#L2185

kimocoder, 2017-10-20:
Remember, I also did [this](https://github.com/aircrack-ng/rtl8812au/commit/4282ab418914d508179a199a5bde0e0a0139fed2), but if it was SMART or not, maybe you could decide that part. Should it be left commented out, or simply changed back?

dfiloni, 2017-10-20:
In order to build CONFIG_MP_VHT_HW_TX_MODE for 8812 I think you have to change:
> hal/hal_mp.c:#if defined(CONFIG_RTL8814A) || defined(CONFIG_RTL8821B) || defined(CONFIG_RTL8822B) || defined(CONFIG_RTL8821C)
os_dep/linux/ioctl_mp.c:#if defined(CONFIG_RTL8814A) || defined(CONFIG_RTL8821B) || defined(CONFIG_RTL8822B) || defined(CONFIG_RTL8821C)

to:
`#ifdef CONFIG_MP_VHT_HW_TX_MODE`

I don't have a 8812 to test this.

@kimocoder : about your question, I think you should try to fix hal_ReadUsbModeSwitch_8814AU [1], hal_ReadUsbModeSwitch_8812AU [2] is properly implemented, you'll have to use EEPROM_USB_MODE_8814A instead of EEPROM_USB_MODE_8812, I think this could work.
[1] https://github.com/aircrack-ng/rtl8812au/blob/v5.1.5/hal/rtl8814a/usb/usb_halinit.c#L2007
[2] https://github.com/aircrack-ng/rtl8812au/blob/v5.1.5/hal/rtl8812a/usb/usb_halinit.c#L2249

evilphish, 2017-10-20:
Thanks for the hint @dfiloni. 
This is not the only thing that needs changing but there are a couple more. I managed to get all and let it compile. It ended up being a bit messy though since I had to edit a couple of ifdefs in order to include 8814A stuff for using it for the 8812 module. Don't know whether this should be capsulated or just commented and left as is. 

either way, hostapd still complains...

> Driver does not support configured HT capability [TX-STBC]

I shall look further, this is what I have so far:

```diff
diff --git a/Makefile b/Makefile
index b689e06..67d99cf 100755
--- a/Makefile
+++ b/Makefile
@@ -85,7 +85,7 @@ CONFIG_AP_WOWLAN = n
 ######### Notify SDIO Host Keep Power During Syspend ##########
 CONFIG_RTW_SDIO_PM_KEEP_POWER = y
 ###################### MP HW TX MODE FOR VHT #######################
-CONFIG_MP_VHT_HW_TX_MODE = n
+CONFIG_MP_VHT_HW_TX_MODE = y
 ###################### Platform Related #######################
 CONFIG_PLATFORM_I386_PC = y
 CONFIG_PLATFORM_ANDROID_X86 = n
diff --git a/hal/hal_mp.c b/hal/hal_mp.c
index bf6b4c2..37bd4b1 100644
--- a/hal/hal_mp.c
+++ b/hal/hal_mp.c
@@ -35,7 +35,7 @@
        #ifdef CONFIG_RTL8192E
                #include <rtl8192e_hal.h>
        #endif
-       #ifdef CONFIG_RTL8814A
+       #if defined(CONFIG_RTL8814A) || defined(CONFIG_RTL8812A)
                #include <rtl8814a_hal.h>
        #endif
        #if defined(CONFIG_RTL8812A) || defined(CONFIG_RTL8821A)
@@ -841,7 +841,7 @@ VOID mpt_SetRFPath_8814A(PADAPTER   pAdapter)
        mpt_ToggleIG_8814A(pAdapter);
 }
 #endif /* CONFIG_RTL8814A */
-#if defined(CONFIG_RTL8814A) || defined(CONFIG_RTL8822B) || defined(CONFIG_RTL8821C)
+#if defined(CONFIG_RTL8814A) || defined(CONFIG_RTL8822B) || defined(CONFIG_RTL8821C) || defined(CONFIG_RTL8812A)
 VOID
 mpt_SetSingleTone_8814A(
        IN      PADAPTER        pAdapter,
@@ -2038,7 +2038,7 @@ u32 hal_mpt_query_phytxok(PADAPTER        pAdapter)
 
 }
 
-#if defined(CONFIG_RTL8814A) || defined(CONFIG_RTL8821B) || defined(CONFIG_RTL8822B) || defined(CONFIG_RTL8821C)
+#if defined(CONFIG_MP_VHT_HW_TX_MODE)
 /* for HW TX mode */
 static VOID mpt_StopCckContTx(
        PADAPTER        pAdapter
diff --git a/os_dep/linux/ioctl_mp.c b/os_dep/linux/ioctl_mp.c
index e2d34f5..ef1dee8 100644
--- a/os_dep/linux/ioctl_mp.c
+++ b/os_dep/linux/ioctl_mp.c
@@ -1947,7 +1947,7 @@ int rtw_mp_hwtx(struct net_device *dev,
        struct mp_priv *pmp_priv = &padapter->mppriv;
        PMPT_CONTEXT            pMptCtx = &(padapter->mppriv.MptCtx);
 
-#if defined(CONFIG_RTL8814A) || defined(CONFIG_RTL8821B) || defined(CONFIG_RTL8822B) || defined(CONFIG_RTL8821C)
+#if defined(CONFIG_MP_VHT_HW_TX_MODE)
        u8              input[wrqu->data.length];
 
        if (copy_from_user(input, wrqu->data.pointer, wrqu->data.length))
```

evilphish, 2017-10-20:
@kimocoder regarding your change with the eeprom, I have uncommented it and it still works at my end.

dfiloni, 2017-10-20:
@evilphish : please try enabling: CONFIG_80211N_HT

evilphish, 2017-10-21:
added an
EXTRA_CFLAGS += -DCONFIG_80211N_HT
but it doesn't change anything. Still no STBC

evilphish, 2017-11-12:
closed due to it being solved. txpower issue remains in its own thread https://github.com/aircrack-ng/rtl8812au/issues/29

closed 2017-11-12 by evilphish