aircrack-ng/rtl8812au #35: RTL8812AU does not support Deauthenticate and other necessay commands

samiux, 2017-10-22:
I compiled with the latest commit ac5840e of v5.1.5 for my RTL8812AU dongle.  Since I do not have USB3.0 port on my laptop, I cannot sure if the driver is working on USB3.0 or not.  However, I am sure that txpower can be set.  Meanwhile, the frequency can be switched automatically when it is targeted to the AP.

When I am going to get the handshake, there is no deauthenticate traffic.  Therefore, I check with the "iw list wlan1" and notice that there is no "deauthenticate" in the "Supported commands" section and there is a few supported commands only.

Any idea?

kimocoder, 2017-10-22:
Yes we're aware of the issue. The TODO will be updated today with new focus on different stuff so it will get listes too. Anyway, thanks for reporting.

samiux, 2017-10-22:
I noticed that when txpower is not set manually, the "Supported commands" is same to txpower is set.  So, I think that the driver is lacking of some features.

kimocoder, 2017-10-22:
@evilphish these are related to "supported commands" it seems

evilphish, 2017-10-22:
Looks fine for me. At least I have deauthenticate, etc... 

                 * new_interface
		 * set_interface
		 * new_key
		 * start_ap
		 * new_station
		 * new_mpath
		 * set_mesh_config
		 * set_bss
		 * authenticate
		 * associate
		 * deauthenticate
		 * disassociate
		 * join_ibss
		 * join_mesh
		 * remain_on_channel
		 * set_tx_bitrate_mask
		 * frame
		 * frame_wait_cancel
		 * set_wiphy_netns
		 * set_channel
		 * set_wds_peer
		 * tdls_mgmt
		 * tdls_oper
		 * start_sched_scan
		 * probe_client
		 * set_noack_map
		 * register_beacons
		 * start_p2p_device
		 * set_mcast_rate
		 * connect
		 * disconnect
		 * channel_switch
		 * Unknown command (104)
		 * Unknown command (105)
		 * Unknown command (121)

kimocoder, 2017-10-22:
This is my 8814au at v5.1.5

`Wiphy phy32
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1963
			RX LDPC
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 450 Mbps
		HT TX/RX MCS rate indexes supported: 0-23
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
			* 2442 MHz [7] (20.0 dBm)
			* 2447 MHz [8] (20.0 dBm)
			* 2452 MHz [9] (20.0 dBm)
			* 2457 MHz [10] (20.0 dBm)
			* 2462 MHz [11] (20.0 dBm)
			* 2467 MHz [12] (20.0 dBm)
			* 2472 MHz [13] (20.0 dBm)
			* 2484 MHz [14] (20.0 dBm)
	Band 2:
		Capabilities: 0x1863
			RX LDPC
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 450 Mbps
		HT TX/RX MCS rate indexes supported: 0-23
		VHT Capabilities (0x03c050b2):
			Max MPDU length: 11454
			Supported Channel Width: neither 160 nor 80+80
			RX LDPC
			short GI (80 MHz)
			TX STBC
			SU Beamformee
			+HTC-VHT
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 1300 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT TX highest supported: 1300 Mbps
		Bitrates (non-HT):
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 5170 MHz [34] (30.0 dBm)
			* 5180 MHz [36] (30.0 dBm)
			* 5190 MHz [38] (30.0 dBm)
			* 5200 MHz [40] (30.0 dBm)
			* 5210 MHz [42] (30.0 dBm)
			* 5220 MHz [44] (30.0 dBm)
			* 5230 MHz [46] (30.0 dBm)
			* 5240 MHz [48] (30.0 dBm)
			* 5260 MHz [52] (30.0 dBm) (no IR, radar detection)
			* 5280 MHz [56] (30.0 dBm) (no IR, radar detection)
			* 5290 MHz [58] (30.0 dBm) (no IR, radar detection)
			* 5300 MHz [60] (30.0 dBm) (no IR, radar detection)
			* 5310 MHz [62] (30.0 dBm) (no IR, radar detection)
			* 5320 MHz [64] (30.0 dBm) (no IR, radar detection)
			* 5500 MHz [100] (30.0 dBm) (no IR, radar detection)
			* 5520 MHz [104] (30.0 dBm) (no IR, radar detection)
			* 5540 MHz [108] (30.0 dBm) (no IR, radar detection)
			* 5560 MHz [112] (30.0 dBm) (no IR, radar detection)
			* 5580 MHz [116] (30.0 dBm) (no IR, radar detection)
			* 5600 MHz [120] (30.0 dBm) (no IR, radar detection)
			* 5620 MHz [124] (30.0 dBm) (no IR, radar detection)
			* 5640 MHz [128] (30.0 dBm) (no IR, radar detection)
			* 5660 MHz [132] (30.0 dBm) (no IR, radar detection)
			* 5680 MHz [136] (30.0 dBm) (no IR, radar detection)
			* 5700 MHz [140] (30.0 dBm) (no IR, radar detection)
			* 5720 MHz [144] (30.0 dBm) (radar detection)
			* 5745 MHz [149] (30.0 dBm) (radar detection)
			* 5765 MHz [153] (30.0 dBm) (radar detection)
			* 5785 MHz [157] (30.0 dBm) (radar detection)
			* 5805 MHz [161] (30.0 dBm)
			* 5825 MHz [165] (30.0 dBm) (radar detection)
			* 5845 MHz [169] (30.0 dBm)
			* 5920 MHz [184] (30.0 dBm)
			* 5940 MHz [188] (30.0 dBm)
			* 5960 MHz [192] (30.0 dBm)
			* 5980 MHz [196] (30.0 dBm)
			* 6000 MHz [200] (30.0 dBm)
			* 6020 MHz [204] (30.0 dBm)
			* 6040 MHz [208] (30.0 dBm)
			* 6060 MHz [212] (30.0 dBm)
			* 6080 MHz [216] (30.0 dBm)
	Supported commands:
		 * new_interface
		 * set_interface
		 * new_key
		 * start_ap
		 * new_station
		 * set_bss
		 * join_ibss
		 * set_pmksa
		 * del_pmksa
		 * flush_pmksa
		 * remain_on_channel
		 * frame
		 * set_channel
		 * connect
		 * disconnect
	Supported TX frame types:
		 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0xd0
		 * managed: 0x40 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	WoWLAN support:
		 * wake up on anything (device continues operating normally)
	software interface modes (can always be added):
		 * monitor
	interface combinations are not supported
	Device supports scan flush.`

kimocoder, 2017-10-22:
while on the "iwlwifi" just to compare, these are supported:

	Supported commands:
		 * new_interface
		 * set_interface
		 * new_key
		 * start_ap
		 * new_station
		 * new_mpath
		 * set_mesh_config
		 * set_bss
		 * authenticate
		 * associate
		 * deauthenticate
		 * disassociate
		 * join_ibss
		 * join_mesh
		 * set_tx_bitrate_mask
		 * frame
		 * frame_wait_cancel
		 * set_wiphy_netns
		 * set_channel
		 * set_wds_peer
		 * probe_client
		 * set_noack_map
		 * register_beacons
		 * start_p2p_device
		 * set_mcast_rate
		 * connect
		 * disconnect
		 * set_qos_map
		 * Unknown command (121)

samiux, 2017-10-22:
There is no support for rtl8812au at official Linux kernel ( https://github.com/torvalds/linux/blob/master/drivers/net/wireless/realtek/rtl8xxxu/rtl8xxxu.h ).  There is only rtl8812.

kimocoder, 2017-10-27:
"Associate", "Disassocate" & "Deauthenticate" support is missing. These should be added somehow to get some functions working, like "deauth" attacks

evilphish, 2017-10-27:
Deauthenticate attacks are working fine for me btw. Just kicked my laptop out of the wifi a few days ago. Maybe it's just a cosmetic thing?

kimocoder, 2018-03-08:
Yes. this driver supports deauthentication and other necessary commands for penetration testing.
as @evilphish mentions, this is in fact a cosmetic thing. will leave it open for others to see and for those who wants to fix it.. feel free