aircrack-ng/rtl8812au #40: Channel lock doesnt' work

yesimxev, 2017-10-24:
Channel keeps changing whenever I try to do anything which would need a specific channel. Tried with iwconfig, iw, using channel number and frequency too

kimocoder, 2017-10-24:
It's noted. Think er got a solution for it, but we'll need some more time to fix it. This report will be left here & updated when resolved.

evilphish, 2017-10-26:
I can not reproduce this on my AWUS036ACH. It locks channels perfectly fine, for example in airodump-ng. More information would be nice.

yesimxev, 2017-10-26:
Sorry forgot to mention. Using RTL8814AU (AWUS1900). Latest Kali

samiux, 2017-12-01:
I am using TP-Link Archer T4UHP (rtl8812au).  I tried Kali Linux driver (version 5.1.5-20170828) and aircrack-ng version (the latest github).  I cannot do the deauth for 2.4 and 5 GHz targets even they are next to the APs.  Both 2.4 and 5 GHz APs are set to auto channel.  However, the drivers (including Kali and aircrack-ng version) cannot detect the channel of the target APs correctly.  

The static channel is also tried with the same problem.  The channel of the APs are always changing no matter they are in static or auto channel.  As a result, the deauth cannot be done properly.

Mister-X-, 2017-12-01:
@samiux, this look more like a user issue. 

Being too close to AP is as bad as being too far, you'll get packet loss. On top of that, some client react different to deauth (directed vs broadcast). Regarding the channel lock, you most likely left the network managers on (you should kill them).

Not being able to detect channel of target AP is vague, you'll have to mention 3 things:
- What you're doing (commands, etc)
- What you're expecting
- What is actually happening (including error messages).

On top of it, please don't duplicate issue on multiple tickets (#40).

samiux, 2017-12-02:
@Mister-X-, 

After killing "wpa_supplicant", it works for 2.4 and 5 GHz to my neighbours.  However, my AP (5 GHz) which is about 2 meters from the dongle cannot detect the channel correctly.

My 5GHz AP channel is set to static 48.  I used "--band abg" to scan.  However, it gives me -1 or other value which is other than 48.

Any idea?

kcdtv, 2017-12-02:
Try with this patches (aircrack-ng itself not the driver): [https://github.com/aircrack-ng/aircrack-ng/pull/122](url)
 I could not detect correctly the channel of my PA neither, the modification from jpmv27 fixed it. 
You can clone it form jpmv27 rep' `git clone https://github.com/jpmv27/aircrack-ng `

samiux, 2017-12-02:
@kcdtv,

Using jpmv27's repository with "airmon-ng check kill", 2.4 and 5GHz channel is detected correctly and handshake is captured.  

TP-Link Archer T4UHP (rtl8812au) works property with jpmv27's repository but not for the TP-Link Archer T9UH (rtl8814au).  It seems rtl8814au driver is not yet completed.

Thanks for the information.

kimocoder, 2017-12-02:
You should always use 'airmon-ng check kill' to avoid the interference from other stuff (wpa supplicant, network-manager +).

As @kcdtv mentioned, the patch from @jpmv27 is confirmed working great.

As for the 8814au, I can also confirm that handshake capturing is a struggle at this time.

kimocoder, 2017-12-02:
This issue report is closed as it was resolved with "airmon-ng check kill" command.

closed 2017-12-02 by kimocoder