aircrack-ng/rtl8812au #41: Interface not showing (ifconfig)

Laurent11, 2017-10-25:
I was using DWA-182 rev C1 (rtl8812au chipset) earlier this week and it was working fine with those drivers. I reinstalled my Kali and I was able to successfully install the drivers but i cannot see it in ifconfig. I hope you will be able to help me.

lsusb : 
```
root@kali:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 2001:3315 D-Link Corp.  **(My adapter)**
Bus 001 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 001 Device 005: ID 0e0f:0008 VMware, Inc. 
Bus 001 Device 004: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 001 Device 002: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

dmesg : 
```
[  344.362376] usb 1-2.1: new high-speed USB device number 8 using xhci_hcd
[  344.466782] usb 1-2.1: New USB device found, idVendor=2001, idProduct=3315
[  344.466786] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  344.466788] usb 1-2.1: Product: D-Link Wireless Adapter
[  344.466789] usb 1-2.1: Manufacturer: Realtek
[  344.466791] usb 1-2.1: SerialNumber: 123456
```

ifconfig : 
```
root@kali:~# ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet IP_HERE  netmask 255.255.255.0  broadcast IP_HERE
        inet6 IP_HERE  prefixlen 64  scopeid 0x20<link>
        ether MAC_HERE  txqueuelen 1000  (Ethernet)
        RX packets 67  bytes 8217 (8.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 194  bytes 15968 (15.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 32  bytes 1752 (1.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 32  bytes 1752 (1.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```

Laurent11, 2017-10-25:
Nevermind, I have found; I have had to click "Connect D-Link Wireless Adapter" in Network Settings. Feeling kind of stupid right now...

closed 2017-10-25 by Laurent11

Dr-Insanity, 2024-09-28:
Happens to the best, no worries. Can't know everything right away from the top of your head, right?