aircrack-ng/rtl8812au #42: Can we get link information added to iw link?

lhernanz, 2017-10-25:
Hi,

as you know, AC mode seems not to be working for some versions of the driver. In some cases, that you are not having an AC connection is pretty evident based on the link speed. But I have been unable to really figure out how to see the negotiated rate with the AP. In some other drivers, this information can be retrieved by the iw link command as shown below:

```
~ » iw dev wlan0 link
Connected to xx:xx (on wlan0)
	SSID: xxx
	freq: 5300
	RX: 239354305 bytes (268745 packets)
	TX: 38976845 bytes (145393 packets)
	signal: -60 dBm
	tx bitrate: 173.3 MBit/s VHT-MCS 8 short GI VHT-NSS 2

	bss flags:	short-slot-time
	dtim period:	1
	beacon int:	100
```

Unfortunately, for this driver, the information available is far less detailed:

```
~ » iw dev wlx00c0ca9585fa link
Connected to xx:xx (on wlx00c0ca9585fa)
	SSID: xxx
	freq: 5300
	signal: -45 dBm
	tx bitrate: 216.7 MBit/s
```

Can we add the negotiated rate to the output of the iw link command? I assume that the command is retrieving the information from the kernel (and, therefore, the driver).

This will help us to do some testing in relation to AC support in the driver.

Thanks

Regards

kimocoder, 2017-10-25:
Thanks for reporting, we'll leave it here for further investigation.

evilphish, 2017-10-26:
Check this https://github.com/evilphish/rtl8812au and see whether it goes into the right direction. There should be a lot more info there but I am not done yet. Some debug prints are commented out and I left comments in the code  about what is still unclear. 
It works as it is and I have commented out all the half-assed stuff that is not yet reliable so should be good to go for now, just not 100% complete.

lhernanz, 2017-11-08:
Hi,

sorry for the delay. I have been away some days. The last version of your branch (@evilphish) does not compile in a 4.10 kernel, so I cannot test. 

Regards

```
  CC [M]  evilphish/os_dep/linux/ioctl_cfg80211.o
evilphish/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_add_monitor_if’:
evilphish/os_dep/linux/ioctl_cfg80211.c:3735:10: error: ‘struct net_device’ has no member named ‘priv_destructor’
  mon_ndev->priv_destructor = rtw_ndev_destructor;
          ^
```

evilphish, 2017-11-08:
@lhernanz Just pushed a fix, please try again.

kimocoder, 2017-11-08:
Merged :)

kimocoder, 2017-11-08:
This issue has been resolved, both iw link states information & compile error on kernel v4.10. thanks to participants for getting it fixed. Closed

closed 2017-11-08 by kimocoder

evilphish, 2017-11-08:
Just for future reference: the information output still needs more work. I am still in the process of implementing more output as comparable to e.g. the intel cards. Some output works but I would not call it "feature complete" just yet ;)

lhernanz, 2017-11-08:
I can confirm that it compiles now and that there is additional information showing up (bss flags, dtim period and beacon int). Unfortunately, there is nothing yet about the negotiated rate. Is that something that you are still working on?

evilphish, 2017-11-08:
Yes, that is actually the more difficult part as these rates depend on pretty much everything. bw, channel, mode, etc so I am at the moment trying to figure out how to check all of this without generating redundant code as there are matrices in some of the header files which contain information regarding this.