aircrack-ng/rtl8812au #44: USB wireless Adapter NOT detected after install the driver

koreaidi0t, 2017-10-29:
I use TP-link Archer T4U AC1300. Before installing driver, the adapter was detected(I mean that I can see the Deviceid of it after run the command "lsusb")but, after installing the driver, the id quickly disappeared.
But the driver worked few days ago, but now it didn't work. When the driver worked before, I just tried to inject frame, capture packets, and change the txpower to 30. It seemed that everything was OK except changing txpower.
I tried to install some different version of driver, but failed...

I'm sorry for my poor English, and thank you for reading.

lion24, 2017-11-17:
Hi! 
Try to unload the driver and load it again. 
rmmod 8812au 
insmod 8812au

or via dkms:
dkms remove -m ${MODULE_NAME}/${MODULE_VERSION} --all

kimocoder, 2018-03-08:
The "lsusb" should show the device (with VID/PID) even when the driver isn't installed. Check your OS setup. Also, try to avoid DKMS at this time, some users are complaining about it not working.

Compile & build manually instead. Check README.

closed 2018-03-08 by kimocoder