aircrack-ng/rtl8812au #46: help with installation kali linux awus1900 RTL8814AU

gearcapitan, 2017-11-03:
I have tried to do every step for the installation of this driver, but apparently for more than try to make the RTL8814 = 1, it does not work, just install the RTL8812AU, help how do I install the correct driver ?, also the apt command -get install realtek-rtl88xxau-dkms, I also did not work, if the wifi icon appears, but when looking for wifi networks it shows absolutely nothing

https://ibb.co/gurxaw

kimocoder, 2017-11-03:
make RTL8814=1 && make install RTL8814=1

I don't visit links btw, rather put images / outputs here.

gearcapitan, 2017-11-03:
[image]
[image]
[image]
[image]
[image]

kimocoder, 2017-11-03:
You dont need to use DKMS install when it's manually buildt.. it's installed as you can see.. what's the problem?

gearcapitan, 2017-11-03:
the device does not appear, as if it were not connected

gearcapitan, 2017-11-03:
[image]

kimocoder, 2017-11-03:
well, I got the same adapter and it is working fine here. try unloading & loading through virtual machine again.

gearcapitan, 2017-11-03:
[image]
I already did what you told me, also try to restart but even so, it still does not appear

gearcapitan, 2017-11-03:
I did the steps as you said, but in another virtual machine and now I run into this problem, it does not load the wifi networks 
[image]

kimocoder, 2017-11-03:
Have you enabled USB3 support in virtual machine then? Might want to use the extension pack for vmware

gearcapitan, 2017-11-03:
if it is enabled
[image]

gearcapitan, 2017-11-04:
any idea?

kimocoder, 2017-11-04:
Not really. Im using it in Kali (not VM) and it's working, so the problem could be VM .. leaving this issue open for others to discuss it. Anyway, output of dmesg after adapter plugin could be usefull too

gearcapitan, 2017-11-05:
agree to then open another topic on this problem and hopefully it will be solved soon

evilphish, 2017-11-05:
For other reasons I just downloaded a fresh copy of kali, created a virtual machine with Virtualbox,  cloned the github repo and built the 8814 module. The module loads up just fine and the dongle fires up. 
Also I am normally using the wifi dongle (8814 as well as 8812) in Linux Mint 18.3 and in Debian GNU/Linux 9. With both distributions the driver builds and loads just fine. 

@gearcapitan you are either doing something wrong while building / installing the module or something is not right with your kali install. In the case of the latter, you might try doing a fresh kali install to make sure you are on the same page.

gearcapitan, 2017-11-05:
@evilphish Could you please kindly explain to me what you did? because the installation of virtualbox that appears in the other post, the installation I made of kali linux in virtualbox was the last iso in the page of kali, then I did an apt-upgrade and install everything before starting to install the driver
#47

Mister-X-, 2017-11-05:
```apt-get upgrade``` will not upgrade the kernel, you need to do an ```apt-get dist-upgrade``` then reboot.

evilphish, 2017-11-05:
@gearcapitan 
1. Downloaded http://cdimage.kali.org/kali-2017.2/kali-linux-2017.2-amd64.iso
2. Set up VM with USB3 controller, harddrive, etc
3. Install Kali
4. Upgrade Kernel to 4.13, either by installing (manually upgrading) linux-image-amd64 or as mister states do a dist-upgrade
5. install git
6. git clone the repo
7. make RTL8814=1
8. Either make install or for a quick test: cp 8814au.ko /lib/modules/<kernelversion> followed by depmod -a
9. connect my AWUS1900 to the usb port
10. right click the usb icon in the Virtualbox window and attach the wifi dongle
11. iwconfig wlan0 mode monitor
12. ifconfig wlan0 up
13. run wifite or whatever you like

Mister-X-, 2017-11-05:
There is actually no need to install VM from ISO, there are prebuilt VM. So, just download the pre-built one and skip 1-3.

gearcapitan, 2017-11-05:
[image]
[image]
[image]
[image]
????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????

evilphish, 2017-11-08:
Did you unplug and replug your device after make install? Afaik installing the module does not necessarily load it immediately. Also more information would be appreciated. As in output of iwconfig, ifconfig, lsmod | grep 8814 and the last couple of lines of dmesg right after plugging in your wlan adapter.

gearcapitan, 2017-11-09:
https://github.com/tpxSecurity/alfa_AWUS1900_adapter 

check this, is a tool actomatic installation driver, working for me

kimocoder, 2017-11-09:
If your problem is to get it into monitor mode, then I could explain.. this driver does not support virtual interface (needed for airmon-ng), therefor it has to be det manually with

ifconfig wlan# (eg wlan1) down
iwconfig wlan# mode monitor
ifconfig wlan# up

That is what the mon0 script in the link does..

nflanders, 2017-12-14:
I cannot for the life of me get my AWUS1900 to work on any build. 2017.2 following your steps @evilphish - nor using the automatic installation driver @gearcapitan 

Is this adaptor just not meant to work in kali!? Have spent the last 6 hours installing new kali virtual boxes, following all of the tutorials, etc. 

Will post my steps soon

Mister-X-, 2017-12-14:
@nflanders Posting in the forum is more appropriate for that issue. This is a bug tracker, to track bugs

nflanders, 2017-12-14:
@Mister-X- Got it. Thanks