aircrack-ng/rtl8812au #49: i can't compile inder debian and 4.11 kernel

Goga777, 2017-11-11:
here is compilation error on my notebook with debian and 4.11 kernel
can you fix it ?

CC [M]  /var/lib/dkms/rtl8812au/5.1.5/build/os_dep/linux/ioctl_cfg80211.o
/var/lib/dkms/rtl8812au/5.1.5/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_add_monitor_if’:
/var/lib/dkms/rtl8812au/5.1.5/build/os_dep/linux/ioctl_cfg80211.c:3735:10: error: ‘struct net_device’ has no member named ‘priv_destructor’; did you mean ‘destructor’?
  mon_ndev->priv_destructor = rtw_ndev_destructor;
          ^~
/usr/src/linux-headers-4.11.0-trunk-common/scripts/Makefile.build:299: ошибка выполнения рецепта для цели «/var/lib/dkms/rtl8812au/5.1.5/build/os_dep/linux/ioctl_cfg80211.o»
make[4]: *** [/var/lib/dkms/rtl8812au/5.1.5/build/os_dep/linux/ioctl_cfg80211.o] Ошибка 1
/usr/src/linux-headers-4.11.0-trunk-common/Makefile:1509: ошибка выполнения рецепта для цели «_module_/var/lib/dkms/rtl8812au/5.1.5/build»
make[3]: *** [_module_/var/lib/dkms/rtl8812au/5.1.5/build] Ошибка 2
Makefile:152: ошибка выполнения рецепта для цели «sub-make»
make[2]: *** [sub-make] Ошибка 2
Makefile:8: ошибка выполнения рецепта для цели «all»
make[1]: *** [all] Ошибка 2
make[1]: выход из каталога «/usr/src/linux-headers-4.11.0-trunk-amd64»
Makefile:1869: ошибка выполнения рецепта для цели «modules»
make: *** [modules] Ошибка 2

Goga777, 2017-11-11:
I updated the kernel for 4.13 and I could compile 5.1.5 driver

evilphish, 2017-11-12:
Thanks for the report. Even though you solved the problem by switching kernel versions, I have not released a fix in https://github.com/evilphish/rtl8812au

If you are interested you could try again with your old 4.11 version.

kimocoder, 2017-11-12:
The fix will be merged here in 1 min.

Goga777, 2017-11-12:
thank you a lot for the fix - I could compile the driver on 4.11 amd64 kernel.

PS
I am just wondering - what is the best place for discussion about rtl8812au/8821au drivers? Here is or on forum at https://forum.aircrack-ng.org/index.php/topic,1086.msg5680.html#msg5680

kimocoder, 2017-11-12:
Yes that forum would be all good since it's related :)
Great you got it working

Goga777, 2017-11-12:
Nevertheless, I will continue to test with 4.13 kernel

kimocoder, 2017-11-18:
This issue has been resolved. Closing

closed 2017-11-18 by kimocoder