aircrack-ng/rtl8812au #51: No Controll Packets received

UlfWetzker, 2017-11-14:
Hi,
I am using a TPLink Archer T9UH AC1900 card (RTL8814AU) with a Debian running a 4.13.4-2 kernel and v5.1.5 for this rtl8812au driver fork.
After setting the card into monitor mode it receives Management and Data packets and I also saw some MIMO packets with an MCS up to 11.

I noticed that there are no ACKs show in wireshark so I filtered for FC type 0 packets in the pcap file.
It is strange since my card is not receiving any Control packets in monitor mode.

Is this also the case with other RTL8814AU based cards?

kimocoder, 2017-11-14:
We're aware of different problems with the drivers, as the FCS packets missing from RadioTap header and so several others. Currently looking at porting fixes from the v5.2.9 branch. These drivers are kinda experimental as they are mix of old/new code and more work has to be done to them. Don't know if 8814 got the same issue you mention, but they should be at the same stage besides we got 8814 USB3 (SuperSpeed) working

clonekrieger, 2018-05-19:
Hi,
are there any updates on this one yet? I have a 8814au Edimax AC1750 USB Stick and cannot see Management-Frames apart from beacons and Probe-Requests/Responses. Additionally, no frames with broken FCS can be seen.

Are there any cards where Mgmnt-Frames and broken FCS frames are visible? I cannot check, if this is a 8814au-problem because I don't have a 8812au card.

kimocoder, 2018-07-24:
This issue should be resolved, if not open the issue again. Thanks!

closed 2018-07-24 by kimocoder

antifermion, 2018-12-02:
I still this issue too with my AWUS1900 (RTL8814AU).
I testet with all branches besides 5.2.9 which I could not get to work on 4.18.0-kali2-amd64.