aircrack-ng/rtl8812au #56: 5GHz not working at all

samiux, 2017-11-28:
I am using TP-Link Archer T4UHP which is rtl8812au chip.  I installed the latest (v5.1.5, commit aa590f3) driver.

When doing deauthentication to 2.4GHz target, the speed is very slow and just a few or no acknowlege from AP or client most of time even I am very close to the AP and client.  However, it captures the handshake finally.

Furthermore, when doing deauthentication to 5GHz target, the speed is very very slow and there is no acknowlege from AP and client.  Handshake cannot be captured.

By the way, I am using USB 2.0 on my laptop and Kali Linux 2017.3 with the latest updates.

samiux, 2017-11-29:
It seems that 2.4GHz target deautherntication is not always success.  I tried for many times to the same target but failed.

evilphish, 2017-11-29:
I have no problems deauthenticating 2.4 GHz targets with either my 8812 or my 8814. 5 GHz also works but as far as I understand it 5 GHz deauth support in aircrack in general is still a little bit flaky.

samiux, 2017-11-30:
However, my Intel 5100 AGN wifi card is operating more smoother and faster than this driver.  I think it is the problem of the driver.

evilphish, 2017-11-30:
It is not disputed that the fragments of code Realtek calls a driver constitute an unholy mess and we are trying to make the best of it but still, deauthing 2.4 GHz works absolutely flawlessly for me using an AWUS036ACH (8812au) and an AWUS1900 (8814au), even better than on my intel card because of the extended range of the Alfa adapters. 5 GHz can take 10 seconds longer to deaut but works most of the time. 
However, there is still much work to be done on the rtl driver suite so it could very well happen that a couple fixes down the road will see an improvement of the 5 GHz deauth capabilities.

samiux, 2017-12-01:
I am using TP-Link Archer T4UHP (rtl8812au). I tried Kali Linux driver (version 5.1.5-20170828) and aircrack-ng version (the latest github). I cannot do the deauth for 2.4 and 5 GHz targets even they are next to the APs. Both 2.4 and 5 GHz APs are set to auto channel. However, the drivers (including Kali and aircrack-ng version) cannot detect the channel of the target APs correctly.

The static channel is also tried with the same problem. The channel of the APs are always changing no matter they are in static or auto channel. As a result, the deauth cannot be done properly.

samiux, 2017-12-03:
This thread should be closed as it is solved by installing jpmv27's repository (please refer to #40).

Thanks.

closed 2017-12-06 by kimocoder