aircrack-ng/rtl8812au #58: Does not support IEEE 802.11w (Protected Management Frames)

samiux, 2017-12-06:
I am using Kali Linux 2017.3 with latest update.  My rtl8812au cannot connect to the router that is set to "required" for Protected Management Frames (IEEE 802.11w).

I have checked the source code and it is enabled when compile.

kimocoder, 2017-12-06:
Jupp. working on that issue along with some other minor stuff. I'll leave a note when it's pushed & ready.

kimocoder, 2017-12-10:
Try updating with "git pull", build/install the updated driver and check if the IEEE 802.11w support works now. Thanks

samiux, 2017-12-11:
Since the MAC address of my rtl8812au dongle always changing, I cannot test it on my ARP protected network.

samiux, 2017-12-14:
The Kali Linux 2017.3 with latest update caused all wireless network interfaces with random MAC address on every plug in and reboot.

I tested this feature with Ubuntu Gnome 16.04.3.  It cannot connect to the Protected Management Frames enabled (Required) router (ASUS RT-RC88U).

kimocoder, 2017-12-14:
Ok. I will also do some test on this in some weeks, currently busy with the 8814au efuse. Thanks for reporting back.

kimocoder, 2017-12-17:
Nuked the 802.11w + adaptivity support for now.
Kernel v4.15 + 8814au EFUSE prioritized and 802.11w code is missing at this point anyway. Closed

kimocoder, 2018-05-03:
@samiux I think I've got a fix for this issue. any chance you may to some test after i pushed it?

samiux, 2018-05-03:
@kimocoder 

I can test the driver at any time. :)

kimocoder, 2018-05-04:
Ok. Gimme few days then

kimocoder, 2018-05-04:
The driver actually have a switch for changing automaticly, it has been turned off not long ago sine the driver should not do this at all.

In the README there's a note on how to switch of the Network-Manager random MAC switch 
But there is some problems in the 802.11w codes anyway, pushing a patch over the weekend and we'll see if the support Will be in place.

kcdtv, 2018-05-04:
> The driver actually have a switch for changing automaticly, it has been turned off not long ago sine the driver should not do this at all.

I did not notice that as I am OK with this feature and have it enabled anyway. Saying so you are absolutely right: The driver shouldn't do that automatically and you were right in taking away this behavior. God luck with the patches and thanks again for al your work :smiley_cat:

kimocoder, 2018-05-06:
@samiux pushed a patch for PMF / 802.11w. try cloning the repo and test the 802.11w when your available.

flip the switch of "CONFIG_80211W" to "y" and it should be good to go. 

the patch may be seen [here](https://github.com/aircrack-ng/rtl8812au/commit/db3e2fa1738214037dc8dc2ccf3cc0a930ce1788)

samiux, 2018-05-06:
@kimocoder 

Compile successful but the AP cannot be connected when the "required" is enabled.

samiux, 2018-05-07:
@kimocoder 

It cannot be connected via NetworkManager as the AP is dimmed.

It cannot be connected via wpa_supplicant.conf with the following error messages when the AP is set to IEEE802.11w=required :

wpa_supplicant[8092]: wlan1: WPA: Failed to configure IGTK to the driver
wpa_supplicant(8092): wlan1: RSN: Failed to configure IGTK

kimocoder, 2018-05-07:
Hmmmm.... will look at it soon again

kimocoder, 2018-07-24:
@samiux could you test 802.11W on the v5.2.20 branch?

samiux, 2018-07-25:
@kimocoder 

Branch v5.2.20 has been tested.  The SSID is dimmed out and cannot be connected.

kimocoder, 2018-07-25:
Ok. Thanks alot for testing sir!

kimocoder, 2018-12-23:
IEEE802.11w support is dropped as it's not needed anyway

closed 2018-12-23 by kimocoder