aircrack-ng/rtl8812au #59: Ask for explaination

samiux, 2017-12-06:
I do not know the following setting mean.  However, I have compile without error when set it to "y".

###################### MP HW TX MODE FOR VHT #######################
CONFIG_MP_VHT_HW_TX_MODE = n

Can anyone explain what that mean?

samiux, 2017-12-07:
Believe or not.  I am using Kali Linux with 4.14 kernel.  After some experiments, I find that the following setting at Makefile is the most good for performance with the patch of @jpmv27 on the latest commit aa590f3.  I set the txpower to 3000.  However, some clients cannot be deauth.

iw wlan0 set txpower fixed 3000

CONFIG_AUTOCFG_CP = y
CONFIG_APPEND_VENDOR_IE_ENABLE = y
CONFIG_MP_VHT_HW_TX_MODE = y

For your information, please.

kimocoder, 2017-12-07:
I will do a walkthrough of several settings in Makefile soon to see if somethings needs to be enabled, disabled or simply unused ones that may be removed.

kimocoder, 2017-12-08:
Yes I got that myself. "statement will never be executed" due to -switch-unreachable "u32 reg0xC50 = 0;" and thats weird, cause I checked [this](http://ubuntu.pl/forum/viewtopic.php?t=182513) patch with out Makefile and it's enabled allready + working fine for 8814au again. this is related to the antenna(s).

samiux, 2017-12-08:
rtl8814au works when it is enable but not rtl8812au.

kimocoder, 2017-12-10:
Fixed the antenna switch in 8812au, but there is still a warning coming out.. use "git pull" to update.
What's new/fixed, see https://github.com/aircrack-ng/rtl8812au/commit/37d751630fff4353bd4dd310e6989d3bd784000a

closed 2018-12-23 by kimocoder