aircrack-ng/rtl8812au #61: VHT 80Mhz wrong representation with iw

jggithub2017, 2017-12-10:
Setting the control channel 40 (5200MHz) and a bandwith of 80Mhz with the center channel 42 (5210MHz)
`# iw dev wlan1 set freq 5200 80 5210`
results in (observe the 30MHz difference to the control channel)
```
root@kali:/usr/src/rtl8812au# iw dev wlan1 info
Interface wlan1
	ifindex 10
	wdev 0x700000001
	addr 00:c0:ca:96:23:d0
	type monitor
	wiphy 7
	channel 40 (5200 MHz), width: 80 MHz, center1: 5170 MHz
	txpower 18.00 dBm 
```
or
```
root@kali:/usr/src/rtl8812au# iw dev wlan1 set freq 5300 80 5290
root@kali:/usr/src/rtl8812au# iw dev wlan1 info
Interface wlan1
	ifindex 12
	wdev 0x900000001
	addr 00:c0:ca:96:23:d0
	type monitor
	wiphy 9
	channel 60 (5300 MHz), width: 80 MHz, center1: 5270 MHz
	txpower 18.00 dBm
```
this probably due to os_dep/linux/ioctl_cfg80211.c line 4422 (static int cfg80211_rtw_get_channel):
```
	case CHANNEL_WIDTH_80:
				RTW_INFO("%s width 80\n", __func__);
				width = NL80211_CHAN_WIDTH_80;
				if(offset==HAL_PRIME_CHNL_OFFSET_LOWER){
					center_freq = control_freq +30;
				}else{
					center_freq = control_freq -30;
				}

```

in core/rtw_wlan_util.c would be rtw_get_center_ch for finding the center channel for every control channel.
```
u8 rtw_get_center_ch(u8 channel, u8 chnl_bw, u8 chnl_offset)
{
        u8 center_ch = channel;

        if (chnl_bw == CHANNEL_WIDTH_80) {
                if (channel == 36 || channel == 40 || channel == 44 || channel == 48)
                        center_ch = 42;
                else if (channel == 52 || channel == 56 || channel == 60 || channel == 64)
                        center_ch = 58;
                else if (channel == 100 || channel == 104 || channel == 108 || channel == 112)
                        center_ch = 106;
                else if (channel == 116 || channel == 120 || channel == 124 || channel == 128)
                        center_ch = 122;
                else if (channel == 132 || channel == 136 || channel == 140 || channel == 144)
                        center_ch = 138;
                else if (channel == 149 || channel == 153 || channel == 157 || channel == 161)
                        center_ch = 155;
                else if (channel == 165 || channel == 169 || channel == 173 || channel == 177)
                        center_ch = 171;
                else if (channel <= 14)
                        center_ch = 7;
        } else if (chnl_bw == CHANNEL_WIDTH_40) {
                if (chnl_offset == HAL_PRIME_CHNL_OFFSET_LOWER)
                        center_ch = channel + 2;
                else
                        center_ch = channel - 2;
        } else if (chnl_bw == CHANNEL_WIDTH_20)
                center_ch = channel;
        else
                rtw_warn_on(1);

        return center_ch;
}
```

With the iwlwifi Intel wlan it works as it should.

kimocoder, 2017-12-10:
Thanks for reporting, will have a look at it soon.

jggithub2017, 2017-12-10:
With
```
                        case CHANNEL_WIDTH_80:
                                RTW_INFO("%s width 80\n", __func__);
                                width = NL80211_CHAN_WIDTH_80;
                                /*
                                if(offset==HAL_PRIME_CHNL_OFFSET_LOWER){
                                        RTW_INFO("OFFSET LOWER");
                                        center_freq = control_freq +30;
                                }else{
                                        RTW_INFO("OFFSET HIGHER");
                                        center_freq = control_freq -30;
                                }*/
                                offset = rtw_get_offset_by_ch(channel);
                                center_ch1 = rtw_get_center_ch(channel, CHANNEL_WIDTH_80, offset);
                                center_freq = ieee80211_channel_to_frequency(center_ch1, band);
                                break
```

I get

```
# iw wlan1 info
Interface wlan1
	ifindex 21
	wdev 0x1200000001
	addr 00:c0:ca:96:23:d0
	type monitor
	wiphy 18
	channel 40 (5200 MHz), width: 80 MHz, center1: 5210 MHz
	txpower 18.00 dBm
```
and the debug message
`[ 7631.421149] RTW: cfg80211_rtw_get_channel : channel 40 width 3 freq1 5210 freq2 0 center_freq 5200 offset 0`

Don't know if this is helpful.