aircrack-ng/rtl8812au #63: Add set_bitrate_mask

jggithub2017, 2017-12-10:
Setbitrate with iw is not supported.
`# iw wlanX set bitrates legacy-5 ht-mcs-5 vht-mcs-5 2:9`

Some inspiration: [initial ath10k implementation](https://github.com/openwrt/bcm63xx-next/commit/51ab1a0a09a894442e8b091d57222c5425b6d700) and [current ath10k implementation](https://github.com/torvalds/linux/blob/master/drivers/net/wireless/ath/ath10k/mac.c#L6994
)

kimocoder, 2017-12-10:
Thanks for the information. Will have a look in few days.

ValZapod, 2019-02-15:
@kimocoder Is it how iw commands are implemented in drivers? Looks complicated enough.

kimocoder, 2019-02-15:
Yep. I just tested CONFIG_CONCURRENT_MODE now.. crashes with the "vif" function :+1: 
I'll dive into it

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)

bakwc, 2021-08-23:
Any workaround? Is there any way to set bitrate?