aircrack-ng/rtl8812au #67: Use fixed mac address when one is not specified

kimocoder, 2017-12-13:
Leaving a commit here for later savings.
The code will use a fixed mac address when one is not specified.

Would people be interested in this? pros vs cons?

https://github.com/goncalossilva/rtl8812AU/commit/eaa89bf90cf30a1e6ed4e828417d52f4e9b97a9d

Thanks.

samiux, 2017-12-14:
In my opinion, when pentesting, the random address is good.  However, when using the dongle for normal usage in an ARP spoofing protected network or MAC filtering AP, it is bad.

evilphish, 2017-12-14:
I don't think a driver should take care of changing the mac address randomly. There are tools for that like macchanger. We shouldn't implement stuff that is not expected to be part of a driver. This will ultimately confuse people.

Mister-X-, 2017-12-14:
Drivers should never change MAC and should always use the MAC of the adapter. Tools don't need to change the MAC at all, it is in the packets (and packets can be crafted).

If MAC changing is necessary, it just need to be ensured that the drivers plays nice when the MAC is changed via MACChanger.

kimocoder, 2017-12-14:
I totally agree that MAC should be fixed to the set (standard) one, but by default the driver actually is set to use random MAC as seen om the screenshot.. so it looks like we'd agree om turning the 'stupid' feature off in the next commits pushed then? Thanks for some feedback on the matter!

[image]

samiux, 2017-12-15:
I reported the MAC address randomization to Kali Linux and they said that MAC address randomization is a new feature of Network Manager.  It is not a bug.

I did some Google search and found out that there is a way to disable this feature.

The current driver is working great on Ubuntu Gnome 16.04.3 HWE.  I think that we better do not alter the driver for this matter.

samiux, 2017-12-15:
For Ubuntu 17.04 (or later) and Kali Linux 2017.3 (or later) as well as NetworkManager 1.2.0 or later add the following lines

[device]
wifi.scan-rand-mac-address=no

at the end of file /etc/NetworkManager/NetworkManager.conf and restart NetworkManager with the command:

sudo service NetworkManager restart

kimocoder, 2017-12-16:
This is current output of macchanger, and it switches MAC when unplugged/plugged.
I say we set it fixed (at least in the driver) as @Mister-X- and @evilphish mentions, the driver should not take care of MAC switching. I'll push the commit with fixed MAC. thanks for input/feedback.

`root@kali:~/Downloads/rtl8812au# macchanger -s wlan1
Current MAC:   46:7a:15:cd:70:88 (unknown)
Permanent MAC: 00:c0:ca:95:4c:c6 (ALFA, INC.)`

kimocoder, 2017-12-16:
The commit along with few others will be pushed soon. 
Random MAC may/should be set in NetworkManager, or with "macchanger" instead of the driver. 
Closed.

closed 2017-12-16 by kimocoder

rdinse, 2019-06-26:
It seems MAC address randomization was only removed in version v5.1.5. After that randomization was seemingly re-introduced, or am I missing something?

https://github.com/aircrack-ng/rtl8812au/blob/v4.3.21/core/rtw_ieee80211.c#L1490

https://github.com/aircrack-ng/rtl8812au/blob/v5.1.5/core/rtw_ieee80211.c#L1382

https://github.com/aircrack-ng/rtl8812au/blob/v5.2.9/core/rtw_ieee80211.c#L1382