aircrack-ng/rtl8812au #69: Build fails on 4.15-rc3

ethan961, 2017-12-17:
Build fails as follows under 4.15-rc3 on Gentoo.

```
ethan@ethan-desktop ~/build/rtl8812au $ KVER=4.15.0-rc3 make RTL8814=1
make -j 12 ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.15.0-rc3/build M=/home/ethan/build/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-4.15-rc3'
  CC [M]  /home/ethan/build/rtl8812au/core/rtw_cmd.o
  CC [M]  /home/ethan/build/rtl8812au/core/rtw_security.o
  CC [M]  /home/ethan/build/rtl8812au/core/rtw_debug.o
  CC [M]  /home/ethan/build/rtl8812au/core/rtw_io.o
  CC [M]  /home/ethan/build/rtl8812au/core/rtw_ioctl_query.o
  CC [M]  /home/ethan/build/rtl8812au/core/rtw_ioctl_set.o
  CC [M]  /home/ethan/build/rtl8812au/core/rtw_ieee80211.o
  CC [M]  /home/ethan/build/rtl8812au/core/rtw_mlme.o
  CC [M]  /home/ethan/build/rtl8812au/core/rtw_mlme_ext.o
  CC [M]  /home/ethan/build/rtl8812au/core/rtw_mi.o
  CC [M]  /home/ethan/build/rtl8812au/core/rtw_wlan_util.o
  CC [M]  /home/ethan/build/rtl8812au/core/rtw_vht.o
  CC [M]  /home/ethan/build/rtl8812au/core/rtw_pwrctrl.o
  CC [M]  /home/ethan/build/rtl8812au/core/rtw_rf.o
  CC [M]  /home/ethan/build/rtl8812au/core/rtw_recv.o
  CC [M]  /home/ethan/build/rtl8812au/core/rtw_sta_mgt.o
  CC [M]  /home/ethan/build/rtl8812au/core/rtw_ap.o
In file included from ./include/linux/kernel.h:10:0,
                 from ./include/linux/list.h:9,
                 from ./include/linux/module.h:9,
                 from /home/ethan/build/rtl8812au/include/basic_types.h:81,
                 from /home/ethan/build/rtl8812au/include/drv_types.h:31,
                 from /home/ethan/build/rtl8812au/core/rtw_pwrctrl.c:22:
/home/ethan/build/rtl8812au/core/rtw_pwrctrl.c: In function ‘pwr_state_check_handler’:
./include/linux/kernel.h:930:51: error: ‘_adapter {aka struct _ADAPTER}’ has no member named ‘pwrctrlpriv’
  BUILD_BUG_ON_MSG(!__same_type(*(ptr), ((type *)0)->member) && \
                                                   ^
./include/linux/compiler.h:301:19: note: in definition of macro ‘__compiletime_assert’
   bool __cond = !(condition);    \
                   ^~~~~~~~~
./include/linux/compiler.h:324:2: note: in expansion of macro ‘_compiletime_assert’
  _compiletime_assert(condition, msg, __compiletime_assert_, __LINE__)
  ^~~~~~~~~~~~~~~~~~~
./include/linux/build_bug.h:47:37: note: in expansion of macro ‘compiletime_assert’
 #define BUILD_BUG_ON_MSG(cond, msg) compiletime_assert(!(cond), msg)
                                     ^~~~~~~~~~~~~~~~~~
./include/linux/kernel.h:930:2: note: in expansion of macro ‘BUILD_BUG_ON_MSG’
  BUILD_BUG_ON_MSG(!__same_type(*(ptr), ((type *)0)->member) && \
  ^~~~~~~~~~~~~~~~
./include/linux/kernel.h:930:20: note: in expansion of macro ‘__same_type’
  BUILD_BUG_ON_MSG(!__same_type(*(ptr), ((type *)0)->member) && \
                    ^~~~~~~~~~~
./include/linux/timer.h:144:2: note: in expansion of macro ‘container_of’
  container_of(callback_timer, typeof(*var), timer_fieldname)
  ^~~~~~~~~~~~
/home/ethan/build/rtl8812au/core/rtw_pwrctrl.c:403:23: note: in expansion of macro ‘from_timer’
  _adapter *padapter = from_timer(padapter, t, pwrctrlpriv.pwr_state_check_timer);
                       ^~~~~~~~~~
In file included from ./include/linux/compiler_types.h:58:0,
                 from ./include/uapi/linux/stddef.h:2,
                 from ./include/linux/stddef.h:5,
                 from ./include/uapi/linux/posix_types.h:5,
                 from ./include/uapi/linux/types.h:14,
                 from ./include/linux/types.h:6,
                 from /home/ethan/build/rtl8812au/include/basic_types.h:80,
                 from /home/ethan/build/rtl8812au/include/drv_types.h:31,
                 from /home/ethan/build/rtl8812au/core/rtw_pwrctrl.c:22:
./include/linux/compiler-gcc.h:166:2: error: ‘_adapter {aka struct _ADAPTER}’ has no member named ‘pwrctrlpriv’
  __builtin_offsetof(a, b)
  ^
./include/linux/stddef.h:17:32: note: in expansion of macro ‘__compiler_offsetof’
 #define offsetof(TYPE, MEMBER) __compiler_offsetof(TYPE, MEMBER)
                                ^~~~~~~~~~~~~~~~~~~
./include/linux/kernel.h:933:21: note: in expansion of macro ‘offsetof’
  ((type *)(__mptr - offsetof(type, member))); })
                     ^~~~~~~~
./include/linux/timer.h:144:2: note: in expansion of macro ‘container_of’
  container_of(callback_timer, typeof(*var), timer_fieldname)
  ^~~~~~~~~~~~
/home/ethan/build/rtl8812au/core/rtw_pwrctrl.c:403:23: note: in expansion of macro ‘from_timer’
  _adapter *padapter = from_timer(padapter, t, pwrctrlpriv.pwr_state_check_timer);
                       ^~~~~~~~~~
make[2]: *** [scripts/Makefile.build:310: /home/ethan/build/rtl8812au/core/rtw_pwrctrl.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1502: _module_/home/ethan/build/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/linux-4.15-rc3'
make: *** [Makefile:1871: modules] Error 2
```

Attempting build again a couple times without cleaning also reveals this from parallel jobs:
```
make[2]: *** Waiting for unfinished jobs....
/home/ethan/build/rtl8812au/os_dep/linux/mlme_linux.c:64:6: error: conflicting types for ‘rtw_join_timeout_handler’
 void rtw_join_timeout_handler(void *FunctionContext)
      ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/ethan/build/rtl8812au/include/drv_types.h:97:0,
                 from /home/ethan/build/rtl8812au/os_dep/linux/mlme_linux.c:23:
/home/ethan/build/rtl8812au/include/rtw_mlme.h:775:6: note: previous declaration of ‘rtw_join_timeout_handler’ was here
 void rtw_join_timeout_handler(struct timer_list *t);
      ^~~~~~~~~~~~~~~~~~~~~~~~
/home/ethan/build/rtl8812au/os_dep/linux/mlme_linux.c:70:6: error: conflicting types for ‘_rtw_scan_timeout_handler’
 void _rtw_scan_timeout_handler(void *FunctionContext)
      ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/ethan/build/rtl8812au/include/drv_types.h:97:0,
                 from /home/ethan/build/rtl8812au/os_dep/linux/mlme_linux.c:23:
/home/ethan/build/rtl8812au/include/rtw_mlme.h:776:6: note: previous declaration of ‘_rtw_scan_timeout_handler’ was here
 void _rtw_scan_timeout_handler(struct timer_list *t);
      ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/ethan/build/rtl8812au/os_dep/linux/mlme_linux.c: In function ‘rtw_init_mlme_timer’:
/home/ethan/build/rtl8812au/os_dep/linux/mlme_linux.c:106:2: error: implicit declaration of function ‘_init_timer’; did you mean ‘__init_timer’? [-Werror=implicit-function-declaration]
  _init_timer(&(pmlmepriv->assoc_timer), padapter->pnetdev, rtw_join_timeout_handler, padapter);
  ^~~~~~~~~~~
  __init_timer
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:310: /home/ethan/build/rtl8812au/os_dep/linux/mlme_linux.o] Error 1
/home/ethan/build/rtl8812au/os_dep/linux/os_intfs.c: In function ‘devobj_init’:
/home/ethan/build/rtl8812au/os_dep/linux/os_intfs.c:1737:2: error: implicit declaration of function ‘_init_timer’; did you mean ‘__init_timer’? [-Werror=implicit-function-declaration]
  _init_timer(&(pdvobj->dynamic_chk_timer), NULL, _dynamic_check_timer_handlder, pdvobj);
  ^~~~~~~~~~~
  __init_timer
```

If there's anything else you need, let me know.

kimocoder, 2017-12-17:
Oooh yes I'm aware of this.. the kernel v4.15 support patch introduced bugs in several places (wowlan and others), the patch was released originally released by "lwfinger" but he withdrew the patch and released this https://github.com/lwfinger/rtl8812au/commit/3b4e11e9ecddd3f2fb399f3c53ea5bf782ff2cfc instead. which compiles on his "gnab" fork, but since I did the stupidest thing not to test the patch before pushing it to this repo (from my private) I have to do a cleanup and do it over (since he's fork is based on v4.2.2 while we're on v5.1.5 and code differs)

kimocoder, 2017-12-17:
Committed a cleanup patch with preparation for a new support patch (deactivated 80211w + adaptivity), but I don't got more time at this point to add the 4.15 support.

closed 2018-01-05 by kimocoder

kimocoder, 2018-01-05:
Kernel v4.15 support is now added. Thanks to @CGarces