aircrack-ng/rtl8812au #71: airodump-ng read failed, USB disconnect

jggithub2017, 2017-12-18:
`airodump-ng wlx00c0ca9623d0` crashes with:
```
read failed: Network is down
Interface wlx00c0ca9623d0: 
ioctl(SIOCGIFINDEX) failed: No such device
Can't reopen wlx00c0ca9623d0
```
This happens from time to time. Attached is the output of `journactl --system -l -a` with the crash and one output aborted by Ctrl+C. The relevant part seems to be:
```
Dec 18 16:28:12 debian kernel: RTW: reg 0x454, usb read 1 fail, status:-71 value=0x3ce0000, vendorreq_times:1
Dec 18 16:28:12 debian kernel: RTW: reg 0x454, usb read 1 fail, status:-71 value=0x3ce0000, vendorreq_times:2
Dec 18 16:28:12 debian kernel: RTW: reg 0x454, usb read 1 fail, status:-71 value=0x3ce0000, vendorreq_times:3
Dec 18 16:28:12 debian kernel: RTW: reg 0x454, usb read 1 fail, status:-71 value=0x3ce0000, vendorreq_times:4
Dec 18 16:28:12 debian kernel: RTW: reg 0x454, usb read 1 fail, status:-71 value=0x3ce0000, vendorreq_times:5
Dec 18 16:28:12 debian kernel: RTW: [dvobj:ffff9ceb97fe9000][ERROR] continual_io_error:5 > 4
Dec 18 16:28:12 debian kernel: usb 2-1: USB disconnect, device number 31
```
[aborted-by-user.log.txt](https://github.com/aircrack-ng/rtl8812au/files/1568551/aborted-by-user.log.txt)
[crash.log.txt](https://github.com/aircrack-ng/rtl8812au/files/1568552/crash.log.txt)

My /etc/NetworkManager/NetworkManager.conf looks like:
```
[main]
plugins=ifupdown,keyfile

[ifupdown]
managed=false

[keyfile]
unmanaged-devices=mac:00:c0:ca:96:23:d0

[device]
wifi.scan-rand-mac-address=no
```

kimocoder, 2017-12-18:
Which adapter + chipset are you using?
May do some testa in few hours, but last I checked airodump-ng worked fine here (on 8814au)

jggithub2017, 2017-12-18:
RTL8812AU with ALFA Networks AWUS036AC and  AWUS036ACH
Happens with both. Sometimes it happens directly after startup, then 5min after running airodump-ng, then 15min laters.
But I have the problem too, that the adapters go regularly into a bizarre mode, not monitoring but a mode where I only see around 1/10th of the usual APs (this was described somewhere in the issues before mine). 
And running multiple adapters with a USB Hub doesn't work either. So may be it has something to do with my HP notebook. Will test the whole thing on a Asus later too.

jggithub2017, 2017-12-19:
The USB disconnect is probably a notebook problem. Checked the error message (-71) stuff and found out that I already did some research on this for USB flash drives months ago concerning the HP notebook. The Asus doesn't seem to have this problem. But the 1/10th APs is on the Asus too. But as the main problem is hardware related I close this issue.

closed 2017-12-19 by jggithub2017

kimocoder, 2017-12-19:
Ok. there is a 'missing AP's (5ghz fix) available' [here](https://github.com/aircrack-ng/rtl8812au/issues/40) you may want to check it out. It's for the aircrack-ng suite itself, not the driver. Look at the comment by  @kcdtv

jggithub2017, 2017-12-19:
@kimocoder thank you for the link and comments. I'll try the 5 GHz stuff this evening.

For the history books. The USB disconnect stuff can be fixed with updating the firmware packages from stretch-backports - at least for the HP EliteBook 820 G4. A BIOS update couldn't harm either.
 
Add following line to /etc/apt/source.list:
`deb http://ftp.debian.org/debian stretch-backports main contrib non-free`
Then
```
# apt update
# apt upgrade
# apt -t stretch-backports install firmware-iwlwifi firmware-linux firmware-linux-free firmware-linux-nonfree firmware-realtek firmware-misc-nonfree
```