aircrack-ng/rtl8812au #72: Help with Installing ALFA AWUS036AC Wifi Adapter

KNSSCyberSec, 2017-12-19:
Trying to run `./dkms-install.sh` and I am receiving errors that the directory does not exist. See attached screen shot. 

[image]

I would appreciate your help. 

Thx you,

kimocoder, 2017-12-19:
Try using "make && make install" method instead then.

KNSSCyberSec, 2017-12-19:
I was able to finish the driver installation but for some reason on Kali Linux running on VMWare Fusion on my MacBook Pro the WiFi adapter doesn't seem to be recognizing. 

[image]
[image]

Mister-X-, 2017-12-19:
What does lsusb says? Could you also attach dmesg. Please copy/paste text in this case.

KNSSCyberSec, 2017-12-19:
I was advised to run 'mobprobe 8812au` and this error came up 

"root@kali:~# modprobe 8812au
modprobe: ERROR: could not insert '8812au': Unknown symbol in module, or unknown parameter (see dmesg)
root@kali:~# lsusb
Bus 001 Device 006: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 004: ID 0e0f:0008 VMware, Inc. 
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

How do retrieve the dmesg information for you (sorry new to Linux)

jggithub2017, 2017-12-20:
Run the command `dmesg`
`# dmesg`

You can pipe the output to a text file for attaching it here.
`# dmesg > dmesg-output.txt`

jggithub2017, 2017-12-20:
I think the problem is related to where you stored (Download folder) the source.
Use:
`# apt install realtek-rtl88xxau-dkms`
then the driver gets compiled automatically. Reboot your system and the next time you plug in the stick, it should work.

Or download the source manually and compile it by hand (may be you have to install git, build-essential and the kernel headers for your current kernel first):
```
# git clone https://github.com/aircrack-ng/rtl8812au.git
# cd rtl8812au
# make
# make install
```

KNSSCyberSec, 2017-12-21:
Thanks, 

Here is the dmesg output as requested. 

[dmesg-output.txt](https://github.com/aircrack-ng/rtl8812au/files/1579262/dmesg-output.txt)

KNSSCyberSec, 2017-12-21:
I received this message 

"I think the problem is related to where you stored (Download folder) the source.
Use:
# apt install realtek-rtl88xxau-dkms
then the driver gets compiled automatically. Reboot your system and the next time you plug in the stick, it should work.

Or download the source manually and compile it by hand (may be you have to install git, build-essential and the kernel headers for your current kernel first):

# git clone https://github.com/aircrack-ng/rtl8812au.git
# cd rtl8812au
# make
# make install"

but it appears it was deleted. 

1) I just tried the 
# apt install realtek-rtl88xxau-dmks
Which appears I already have it installed see print screen here 

[image]

After that I rebooted the VMWare session of Kali Linux and tried to plugin the ALFA Adaptor 
[image]

[image]

[image]

The ALFA adapter is not powering up see here

[image]

jggithub2017, 2017-12-21:
I wrote the installation stuff before I checked that you've got the driver compiled. So I deleted it afterwards.

I don't see anything in the dmesg. Did you make it just after a restart?
Could you do the following:
1. Start your virtual machine.
2. Connect the Wifi adapter to the virtual machine (as you did in the pictures above).
3. Run the `lsusb` command and check if the Wifi adapter shows up (like above).
4. Then do a `dmesg > dmesg1.txt`
5. Try `modprobe -v 8812au` . The -v stands for verbose and provide the error message.
6. Do a `dmesg > dmesg2.txt`
7. Upload the dmesgs here.

You could live following your system journal (dmesg) when you open a second terminal and type `journalctl -a -l -f` there. So you could see whats happening.

KNSSCyberSec, 2017-12-21:
Here you go 

[image]

[dmesg1.txt](https://github.com/aircrack-ng/rtl8812au/files/1580076/dmesg1.txt)

I tried to run 'modprobe -v 8812au' but it came up with an error 

[image]

What would you like me to do next?

kimocoder, 2017-12-21:
Are you using an USB hub? Which kernel are you using?

KNSSCyberSec, 2017-12-21:
No I am not using a USB Hub, see image above.  I am on a Macbook Pro with the new OS, using VMWare Fusion and the session I am using is Kali Linux. 

I hope that answers your questions.

jggithub2017, 2017-12-21:
Could you please run before you try the stuff in the message above?
`# dpkg-reconfigure realtek-rtl88xxau-dmks`

KNSSCyberSec, 2017-12-21:
sorry, do you want me to run that command first and then run all the previous commands again?

kimocoder, 2017-12-21:
Is it a _vfs_read error i see om the bottom there? That's related to kernel v4.14 support (which the driver got btw)

KNSSCyberSec, 2017-12-21:
is this what you would like me to do?

1. Start your virtual machine.
2. Connect the Wifi adapter to the virtual machine (as you did in the pictures above).
3. Run the lsusb command and check if the Wifi adapter shows up (like above).
4. run this first an
'dpkg-reconfigure realtek-rtl88xxau-dmks'
5. Then do a 'dmesg > dmesg1.txt'
6. Try 'modprobe -v 8812au' . The -v stands for verbose and provide the error message.
7. Do a 'dmesg > dmesg2.txt'
8. Upload the dmesgs here.

jggithub2017, 2017-12-21:
May be the 4.14 support is not in the kali packages yet.
 
@KNNSCyberSec: yes

KNSSCyberSec, 2017-12-21:
Sorry what is it about the driver? I have tried several different drivers @kimocoder

KNSSCyberSec, 2017-12-21:
Here you go @jggithub2017 

[image]

[dmesg1.txt](https://github.com/aircrack-ng/rtl8812au/files/1580172/dmesg1.txt)

jggithub2017, 2017-12-21:
The error you've got in your dmesg is related to an older version of the driver. The current version in the git repository take care of this error.

That's how you'll get the newest driver version with 4.14 support.
# apt remove realtek-rtl88xxau-dkms
restart your VM.
Then:
# git clone https://github.com/aircrack-ng/rtl8812au.git
# cd rtl8812au
# make
# make install

kimocoder, 2017-12-21:
It is included in the latest Kali package (If a proper apt-get update & upgrade is done), but building from repo should include more fixes. Just pushed a new one 1 minute ago.

Try using @jggithub2017 method..

KNSSCyberSec, 2017-12-21:
@jggithub2017 when I download the new driver, where would you like me to keep the files. I remember you mention there might be an issue having it in the download folder. 

Also when I go to the drive I only see an option to download. is there a way to get the "clone" if so how due I do that?

Thank you,

jggithub2017, 2017-12-21:
The place doesn't mattter with the git variant. For now, just do it at the current home folder (`cd ~`) git clone will create a local folder, from where then you can get always the newest version of the driver (I explain that at the end of this thread).

Install git with `# apt install git`

KNSSCyberSec, 2017-12-21:
ok I ran the commands below 

`git clone https://github.com/aircrack-ng/rtl8812au.git`

`cd rtl8812au`

`make`

`make install`

Here is where I am currently 

[image]

What would you like me to do next?

jggithub2017, 2017-12-21:
excellent
just for cosmetic install bc
`# apt install bc`

Did you remove the dkms driver? (apt remove stuff?)

Now i propose that you restart your vm and try to connect your adapter. If it is not running, try 
`# modprobe 8812au`

kimocoder, 2017-12-21:
well.. at least "apt-get install bc" and THEN "make && make install" before "modprobe 8812au"

KNSSCyberSec, 2017-12-21:
yes I removed the dkms driver first.  @jggithub2017

kimocoder, 2017-12-21:
As seen on the dmesg, loading out-of-tree module taints the kernel.. it works great on a pure 'setup' but virtual machines got an issue with out-of-tree modules as mentioned several places @ Google.

Have experienced related issues on OSX before, where ACPI needs to be disabled.
At the bottom is an howto on turning ACPI off

what is a tainted kernel? look at the explain over [here](https://unix.stackexchange.com/questions/118116/linux-what-is-a-tainted-kernel)

a solution may be to turn ACPI off in boot parameter, as in rebooting to the blue kernel selection menu, select kernel v4.14 and press "E" to edit the parameters and then add "ACPI=off" at the end.

kimocoder, 2017-12-21:
run "iwconfig" and post the output here

KNSSCyberSec, 2017-12-21:
You bring up a good point @kimocoder my next question was going to be how to get this WiFi adapter to work on my OSX next. 

Here is the iwconfig 

[image]

jggithub2017, 2017-12-21:
cool! That's it. (Do you know how to use airodump-ng, kismet and co for wireless sniffing?)
Now, you use the driver directly from the repository of aircrack-ng on github. @kimocoder is actively working on this driver - so there'll be new features, bug fixes, etc. on regular basis. 
And for every new kernel version you need to recompile the module too. Here a small step by step guide:

```
# cd rtl8812au
# git pull
# make clean
# make
# make install
```

kimocoder, 2017-12-21:
Great. there you go.

Remember, virtual interface support isn't working at the moment, so the adapter would need to be set in monitor mode manually with:

# ifconfig wlan0 down
# iwconfig wlan0 mode monitor
# iwconfig wlan0 txpower 30

Thanks to all of you for taking apart in a solution.

jggithub2017, 2017-12-21:
@KNSSCyberSec set the card in monitor mode and then:
`# airodump-ng -b abg wlan0`
You see know all APs around you, stations (laptops, smartphones) connected to the APs or sending probe requests. With the key "a" you can select what airodump-ng should show (APs only, stations only, etc)
Ctrl+C to terminate airodump-ng

KNSSCyberSec, 2017-12-21:
Thank you so much @jggithub2017 and @kimocoder 

No I do not know how to use airodump-ng, kismet and co for wireless sniffing. I am actually just learning Penetration testing and needed to get this adapter to work for the labs. 

I wrote down the steps to update the for different kernal versions. 

These are the steps to set the monitoring mode?

ifconfig wlan0 down

iwconfig wlan0 mode monitor

iwconfig wlan0 txpower 30

jggithub2017, 2017-12-21:
```
# ifconfig wlan0 down
# iwconfig wlan0 mode monitor
# ifconfig wlan0 up
```
for the monitor mode.
@kimocoder has a tutorial on all we discussed here on the [main page of the repository](https://github.com/aircrack-ng/rtl8812au/)

KNSSCyberSec, 2017-12-21:
Thanks @jggithub2017  about the monitoring mode. 

oh where does @kimocoder have that tutorial? looking through the Read.me file it appears different from 

`git clone https://github.com/aircrack-ng/rtl8812au.git`
`cd rtl8812au`
`make`
`make install`

`apt install bc`

You and @kimocoder where amazing helping me resolve this.

jggithub2017, 2017-12-21:
It's not in the same order, but for example the git clone is under the Notes section.

Happy Pentesting!

KNSSCyberSec, 2017-12-21:
ahh got it. Thank you @jggithub2017 now it's time to learn how :-)

kimocoder, 2018-01-03:
Issues fixed. Closing

closed 2018-01-03 by kimocoder