aircrack-ng/rtl8812au #75: Can't find 5GHz network

YXX-to-YY, 2017-12-27:
In my kali（4.14 kernel） vmware，I install v5.1.5 driver for 8812au ，it can find 2.4G network ，but can't find 5G, however in win10 it can find the 5G network! How can I deal with this problem?

kimocoder, 2017-12-27:
Look what @kcdtv mentions in the report over [here](https://github.com/aircrack-ng/rtl8812au/issues/40) try patching the "aircrack-ng" suite itself, or clone it from jpmv27 and compile/build it from that source.

YXX-to-YY, 2017-12-27:
@kimocoder  thank you. airodump-ng has found the 5G networks. and I want to know has any patches for the driver ? because iwlist or networkmanager still can't find 5G and sometimes they can't find any networks even 2.4G. This question has been bothering me for a long time.

kimocoder, 2017-12-27:
The driver hasn't got any and I got any possibilities testing this at this moment.
Leaving this report open for lster. Thanks for reporting.

kcdtv, 2017-12-27:
> because iwlist or networkmanager still can't find 5G and sometimes they can't find any networks even 2.4G

This is not driver related 
In case you did not notice network manager is... a piece of shit. This fedora/system.d/networkmanager crew have no clue about what they are doing "wirelessly" speaking. Network manager literally crashes in some OS when you plug an USB or when you switch network. The fact that it is the only "graphical solution" for network management  in our GNU-Linux world is hell of a problem. 
iwlist is deprecated (you are supposed to use iw instead) and it was never meant to be a powerful scanner, it is just to check briefly what is around and in close range. And if it doesn't work sometimes it is due to... network manager (conflict)
For example network manager doesn't let you use wpa_cli and you have to totally turn it off and launch wpa_suppplicant in deamon mode first... Some years before that was not the case. I do not want to "offtopic" too much but it is to point out that the dirver is the not the reason for this bugs.  
I would personally close this ticket

kimocoder, 2017-12-27:
Closed after request & explaintation from @kcdtv

closed 2017-12-27 by kimocoder

YXX-to-YY, 2017-12-28:
I have learned. Thank you for your help very much!