aircrack-ng/rtl8812au #76: can not be compiled under the older kernel

Fredy72, 2017-12-29:
Can be repeated, for example:
Ubuntu 16.04 LTS x86_64
1. wget https://cdn.kernel.org/pub/linux/kernel/v3.x/linux-3.16.51.tar.xz
2. tar -xJvpf linux-3.16.51.tar.xz
3. cd linux-3.16.51
4. make defconfig
5. make allmodconfig
6. make all
7. cd ..
8. wget https://github.com/astsam/rtl8812au/archive/v5.1.5.zip
9. 7z x v5.1.5.zip
10. cd rtl8812au-5.1.5
11. make KVER=3.16.51  KSRC=../linux-3.16.51
    astam is ok, but alternate
12. rm v5.1.5.zip; rm -rf rtl8812au-5.1.5
13. https://github.com/aircrack-ng/rtl8812au/archive/v5.1.5.zip 
    or wget https://github.com/kimocoder/rtl8812au/archive/v5.1.5.zip
14. 7z x v5.1.5.zip
15. cd rtl8812au-5.1.5
16. make KVER=3.16.51  KSRC=../linux-3.16.51
    aircrack-ng or kimocoder compile bad, this error, same under arm and kernel 3.10 what I know

rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_get_station’:
rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:1754:31: error: ‘STATION_INFO_BSS_PARAM_SHORT_PREAMBLE’ undeclared (first use in this function)
     sinfo->bss_param.flags |= STATION_INFO_BSS_PARAM_SHORT_PREAMBLE;
                               ^
rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:1754:31: note: each undeclared identifier is reported only once for each function it appears in
rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:1757:31: error: ‘STATION_INFO_BSS_PARAM_SHORT_SLOT_TIME’ undeclared (first use in this function)
     sinfo->bss_param.flags |= STATION_INFO_BSS_PARAM_SHORT_SLOT_TIME;
                               ^
rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:1761:31: error: ‘STATION_INFO_BSS_PARAM_CTS_PROT’ undeclared (first use in this function)
     sinfo->bss_param.flags |= STATION_INFO_BSS_PARAM_CTS_PROT;
                               ^
rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:1764:29: error: ‘STATION_INFO_BSS_PARAM_DTIM_PERIOD’ undeclared (first use in this function)
   sinfo->bss_param.flags |= STATION_INFO_BSS_PARAM_DTIM_PERIOD;
                             ^
  CC [M]  rtl8812au-5.1.5/os_dep/linux/wifi_regd.o
rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c: At top level:

kimocoder, 2018-03-08:
Will have a look at older kernel support this weekend. Thanks for reporting, I'll dive into it.

CGarces, 2018-03-08:
@kimocoder 
Consider use travis ci integration with github to test old kernels
You can use this file
https://github.com/CGarces/rtl8192eu-linux-driver/blob/realtek-4.4.x/.travis.yml

kimocoder, 2018-03-08:
I don't know how to use the Travis CI for testing old kernels, haven't used it at all so I don't know how that works. But I'm on the issue, doing it the old way this time, but maybe you could leave a comment on how the Travis CI here works?

kimocoder, 2018-03-08:
I found the commit that broke older kernel support. It's located [here](https://github.com/kimocoder/rtl8812au/commit/cb1454b73814c6593b179da921ea65135d7a2bb6)

Will take a look at it, how it works.. maybe bringing the hardcoded regdomain stuff back to it's place again.

CGarces, 2018-03-08:
Hi @kimocoder, is very easy

- pick Travis CI from https://github.com/marketplace, is free.
- Enter on Travis CI web page and enable the repository, should fail because there is no .travis.yml file. 
- Get the .travis.yml file that is already done for rtl8192 and push it on a PR. I think that Travis CI by default test the PR so you should see the compilation logs.
- Tweak in the PR the .travis.yml file if you need a different kernel number, Travis CI will build ot after each commit.
- Accept the PR and merge.

Usually I have a "update_travis" branch on my repo to made builds and not put garbabe on the master branch

kimocoder, 2018-03-08:
Since @aircrack-ng is the owner of this repo, this has to be done with from him/them.
But thanks for the information regarding Travis anyway!

kimocoder, 2018-03-08:
@Fredy72 just pushed a commit that should fix building on older kernels + arm. 
Could you test it and report back? or someone else?

Fredy72, 2018-03-08:
after commit (that should fix building on older kernels + arm) it reports the same errors, as in the first post

kimocoder, 2018-03-08:
@Fredy72 you did a "git pull" and compiled again right?

kimocoder, 2018-03-08:
Found a solution..

Try commenting out lines 1754 -> 1768 in "os_dep/linux/ioctl_cfg80211.c" like shown below.
Try compiling, building and DO TEST it and report back if the drivers actually works (not just compiles / builds) and report back.

It should look like this:

`                //if (!psta->no_short_preamble_set)
                  //sinfo->bss_param.flags |= STATION_INFO_BSS_PARAM_SHORT_PREAMBLE;

                //if (!psta->no_short_slot_time_set)
                  //sinfo->bss_param.flags |= STATION_INFO_BSS_PARAM_SHORT_SLOT_TIME;

                /* no idea how to check this yet */
                //if (0)
                  //sinfo->bss_param.flags |= STATION_INFO_BSS_PARAM_CTS_PROT;

                /* is this actually the dtim_period? */
                //sinfo->bss_param.flags |= STATION_INFO_BSS_PARAM_DTIM_PERIOD;
                //sinfo->bss_param.dtim_period = pwrctl->dtim;

                //sinfo->bss_param.beacon_interval = get_beacon_interval(&cur_network->network);`

kimocoder, 2018-03-09:
kernel check (switch) has been added. It should be working now.

closed 2018-03-09 by kimocoder

Fredy72, 2018-03-10:
The compilation went well on kernel 3.16 x86 and 3.10 on arm. Good work.

kimocoder, 2018-03-10:
Just dont use the dirty solution i posted above, but rather "git pull" and build/install again.

yuvals-xperitos, 2021-04-13:
Hey, this fix doesn't seem to be incorporated into v5.6.4.2 or v5.7.0. Is there a patch anywhere that describes the correct kernel version tests?

yuvals-xperitos, 2021-04-13:
Ah! Found it. It is on v5.2.20. Is there specific rationale behind not keeping such fixes on other branches? Or just a matter of manpower?

commit 06d23f22 (2018-03-09): Fix build on older kernels

* Added a kernel check for older kernels, fix compiling. fixes #76