aircrack-ng/rtl8812au #77: USB3 power issue

kimocoder, 2018-01-03:
**A user is reporting that the "AWUS036ACH" is underpowered running at 200mA instead of 500mA**

"I have noticed that with the Alfa AWUS036ACH device plugged in to a USB 3.0 port, it only is able to receive 200mA of power. When plugged into a USB 2.0 port it is able to receive 500mA. I'm not sure why this is, but the result of the loss in power is that the device has a very limited range for transmit and receive using these drivers. This information was obtained via the command lsusb -v."

"The AWUS036H being referenced in that post uses the RTL8187 driver, and runs at a full 500mA. The range on that adapter is much greater than the range on the AWUS036ACH, which uses the RTL8812AU driver. None of the users involved in this discovery are developers, so I'm adding this note here in hopes someone who knows what might fix this issue will see it."

**Other sources:**
https://forums.kali.org/showthread.php?38574-Old-AWUS036H-vs-new-AWUS036ACH-performance-comparison
https://github.com/abperiasamy/rtl8812AU_8821AU_linux/issues/12

evilphish, 2018-01-03:
This is a great find! It completely explains why I get such an abysmal range and why changing the txpower does pretty much nothing. At least now we can be reasonably sure that the problem is not with the way the txpower setting is comunicated in the driver but it probably is related to the adapter not getting enough power in order to have any effect.

The only question would be how accurate the lsusb -v is in that regard. It would be interesting to measure the actual current running through the cable with a current clamp, just to confirm and be absolutely sure. Unfortunately I don't have one myself, but I could check whether we have one in the lab and check the AWUS036ACH as well as the AWUS1900, maybe tomorrow.

kimocoder, 2018-01-03:
@i64X use the issue report here instead, my private fork is for experimental usage. It all ends up here anyways. But thanks for the report (wiki report) anyways, now we're aware of the issue as you may see here. Feel free to share thoughts here instead. :smiley_cat:

evilphish, 2018-01-04:
Ok, bad news. If I use a USB2 cable to force the device into USB2 mode, lsusb -v also tells me that it uses 500 mA whereas if I use it directly and it switches to super-speed USB3 mode it tells me it uses 200 mA für the AWUS036ACH and 214mA for the AWUS1900. **But** this is actually not true.
This is the shot of the AWUS1900 in USB2 mode:
[image]
and this is the AWUS036ACH in USB2 mode:
[image]

As you can see, this is by far not 500 mA. Also the current does not change while scanning, etc. In addition, I tried changing txpower and it has **no effect** on the current. Which also means the txpower setting does not affect both devices at all. :/

Trace-2, 2018-01-04:
Hi evilphish ... I am the other user that was discussing this on the kali forum ... I ordered a short usb3 male/female extension cable last night that I'm going to hack and measure the current from. Post is slow here this time of year but I'll hopefully get it in the next couple of days.

I've been testing many of the different drivers from version 4.3.14 onwards in debian stretch (just for range) and the best range I get is with 4.3.20 ... every driver after that has severely restricted range and shows MaxPower as 200ma on the usb3 ports .... version 4.3.20 which has the best range shows MaxPower as 500ma on the usb3 ports ... so I will compare actual true current readings between the 5.1.5 driver and the 4.3.20 driver on the same hardware.

Thanks to all you devs involved for looking into this ... :)

Trace-2, 2018-01-04:
Just thought ... How many mw is the 36ACH?

If it's 1000mw (1w) ... then 1w @5v = 200ma.

evilphish, 2018-01-04:
That sounds great @Trace-2 , looking forward to your measurements. I am a little busy with thesis writing at the moment so can not spend too much time on it ;-)

Regarding the power it all depends on whether the 1W designation for the 36ACH (which I have also read somewhere) relates to the transmit power or to the overall power consumption of the device. We can safely assume that the onboard stuff also takes some current and that it does not put all of the current drawn from the usb port into boosting the signal strength.
Also the 170 mA for the 36ACH are a lot less than 200 mA and this includes the current required to operate the dongle electronics. The actual power available for the signal strength should therefore be less than 170 mA. Also interesting to see is that the AWUS1900 draws 260 mA.  I think we really need USB3 measurements to arrive at a halfway decent conclusion. I also do not know whether the transmitter antenna driver operates with the full 5 V or whether it requires a lower voltage which would mean that we need more current for the same transmit power.

It would also help if you could try changing the txpower value and see whether that changes the current drawn. After all, a txpower of 30 dB is a factor of 1000 higher than a txpower of 1 dB. So **if** txpower setting works with any of your other drivers, the difference in current drawn should be **very** noticeable, not just a couple mA or so.

Trace-2, 2018-01-04:
Just read in the spec. on this page: https://tienda.siliceo.es/en/alfa-network/256-alpha-awus036ach-wifi-usb-30-dual-band-anti-interference.html

That the 36ACH is a 1600mw device.

Yes I agree practical measurements are the only logical way forward.

When the usb3 cable arrives I'll see what current and voltage measurements I can take and also try the different drivers in Debian stretch and Kali 2017.3

If the adapter is easy to get into I'll maybe check out the internal component spec. too.

i64X, 2018-01-04:
I'm in a bit over my head here because I'm not really versed in EE so much, but in response to trying the drivers that Kali suggests on their webpage, that's what I tried first, and the 36AHC didn't show up at all when I plugged it into the Thinkpad. Searching based on that lead me to other users that were having the same issue, which then lead me here. Obviously these drivers fixed the problem, minus the range issue that we're talking about.

My current install is Kali 2017-3 on a bare-metal install on a Thinkpad T450S, if that matters to you Trace for when you experiment as well.

i64X, 2018-01-04:
Just a thought too... I haven't tried it yet, but I've seen other people who say that in Windows they're able to use the device without installing drivers (it must pick them up from Windows Update and install them automatically), but the range issue isn't an issue. I'd be curious to see power measurements when you get that USB3 cable while the device is plugged in to a Windows machine, if in fact it works properly under Windows. I'd test that quick right now but I don't have access to a Windows machine.

Trace-2, 2018-01-04:
Yes Win10 is already on the test list ... I have tried it in windows ... the driver for it is in the default set of drivers as I have never connected to the internet with the windows install.

I'll be putting the adapter through the ringer and disassembling it too ... :)

Trace-2, 2018-01-04:
The antenna Power Amp I.C's are:

 SE2576L (2.4Ghz): http://www.skyworksinc.com/uploads/documents/202428A.pdf
 
 SE5023L (5Ghz): http://www.skyworksinc.com/uploads/documents/SE5023L_202521B.pdf

evilphish, 2018-01-04:
So for 2.4 GHz it draws 650 mA total supply current at a txpower of 29 dBm. This means that if we measure 170 mA we could estimate that it is currently locked to a txpower of 10*log(10^2,9*170/650) = 23 dBm. Probably a little less as the base operating current is higher in percentage for 170 mA than for 650 mA. So maybe 20.5 to 21 dBm? This is actually awfully close to the maximum allowed transmit power by regulatory domain for Germany for the 2.4 Ghz band which is limited to 100 mW or 20 dBm. Could be just a coincidence though. 

For 5 GHz  the power amp draws 600 mA at 26 dBm equating to 10⋅log(10^2,6⋅170/600;10) = 20.5 dBm. The limit for 5GHz is 200 mW or 23 dBm, so this is probably not the deciding factor.

Of course it could be a combination of the two if both are active at the same time and I have no idea whether they only draw the maximum power during an established connection or also in monitor mode. But if the 2.4GHz spec says that this amp alone can draw 650 mA we can for sure say that the 170 mA that I measured tell us that it is not running at full power. 

Will be interesting to see what you can dig up with your USB3 cable.

Trace-2, 2018-01-05:
Are you in Germany? The 036ACH I think just defaults to the global domain which is 20dB for 2.4Ghz

I am in the UK and have recompiled the crda and regulatory.bin for global and GB to 30dB in my copy of debian stretch for temporary testing purposes .... it didn't make any difference after reinstalling the 4.3.20 driver that I get the best range from in managed mode though.

I've also questioned realtek support for a breakdown of the supposed 1600mw power consumption ... but I've no idea what they are like for replying.

When the cable arrives I'll test the current draw in: 

Win10 (new install with latest win driver from Realtek)

Kali (latest weekly install with drivers 5.1.5 & 4.3.20)

Debian Stretch (kernel 4.9 with driver 4.3.20 and recompiled regulatory.bin)

In different modes, changing TX power value, etc. and we'll see how they compare in range and power consumption ... :)

evilphish, 2018-01-05:
@Trace-2, yes I am in Germany. But as I said I doubt that any regulatory stuff is in effect. The option is disabled in the driver and switching to other domains also doesn't change anything about the reception for me. 

I'm very interested in what you find out, thank you for taking the time to dive into that. Wish I could be of more help but as I said, kinda busy with my thesis atm :(

Trace-2, 2018-01-05:
It's no problem ... if I can help increase the range of the adapter that would be a big plus for me. 

My main objective is to get the best signal possible to a workstation in an outbuilding we have here through two walls and a wooden fence ... at the moment I can get a good signal to part of the building but ideally I'd like to position the workstation in a further away location. 

I've got a feeling that the adapter is not performing as well as it should do and would like to do something about it ... driver development is outside my skill set but I don't mind doing a bit of  electronic testing ... :)

Trace-2, 2018-01-05:
Haven't disassembled mine yet but here's a pic I've found online ... note the intriguing switch (SW6) top right ... I'll be trying to trace that back when I get inside it ... :)

[image]

jggithub2017, 2018-01-05:
Power settings are set via the firmware or driver.
bMaxPower is in 2 mA steps.
0x64 == 100 => 200mA
0xfa == 250 => 500mA

The 500mA is probably due to the USB 2 initialization and then the 200mA when the driver changes to USB 3.

```
root@debian:/usr/src/rtl8812au# dmesg | grep -A 10 -B 10 bMax
[ 5724.257634] RTW: 8812au v5.1.5_19247.20160830
[ 5724.257663] RTW: 
               8712_usb_configuration_descriptor:
[ 5724.257663] RTW: bLength=9
[ 5724.257664] RTW: bDescriptorType=2
[ 5724.257664] RTW: wTotalLength=35
[ 5724.257665] RTW: bNumInterfaces=1
[ 5724.257665] RTW: bConfigurationValue=1
[ 5724.257665] RTW: iConfiguration=0
[ 5724.257666] RTW: bmAttributes=80
[ 5724.257666] RTW: bMaxPower=fa
[ 5724.257667] RTW: 
               usb_endpoint_descriptor(0):
[ 5724.257668] RTW: bLength=7
[ 5724.257668] RTW: bDescriptorType=5
[ 5724.257669] RTW: bEndpointAddress=81
[ 5724.257669] RTW: wMaxPacketSize=512
[ 5724.257670] RTW: bInterval=0
[ 5724.257670] RTW: RT_usb_endpoint_is_bulk_in = 1
[ 5724.257671] RTW: 
               usb_endpoint_descriptor(1):
--
[ 5725.285762] usb 2-1: SerialNumber: 123456
[ 5725.286605] RTW: 
               8712_usb_configuration_descriptor:
[ 5725.286608] RTW: bLength=9
[ 5725.286610] RTW: bDescriptorType=2
[ 5725.286611] RTW: wTotalLength=53
[ 5725.286613] RTW: bNumInterfaces=1
[ 5725.286614] RTW: bConfigurationValue=1
[ 5725.286616] RTW: iConfiguration=0
[ 5725.286617] RTW: bmAttributes=80
[ 5725.286618] RTW: bMaxPower=64
[ 5725.286620] RTW: 
               usb_endpoint_descriptor(0):
[ 5725.286622] RTW: bLength=7
[ 5725.286623] RTW: bDescriptorType=5
[ 5725.286625] RTW: bEndpointAddress=81
[ 5725.286626] RTW: wMaxPacketSize=1024
[ 5725.286627] RTW: bInterval=0
[ 5725.286629] RTW: RT_usb_endpoint_is_bulk_in = 1
[ 5725.286631] RTW: 
               usb_endpoint_descriptor(1):
--
[ 5751.481781] usb 2-1: SerialNumber: 123456
[ 5751.482603] RTW: 
               8712_usb_configuration_descriptor:
[ 5751.482605] RTW: bLength=9
[ 5751.482607] RTW: bDescriptorType=2
[ 5751.482609] RTW: wTotalLength=53
[ 5751.482610] RTW: bNumInterfaces=1
[ 5751.482611] RTW: bConfigurationValue=1
[ 5751.482613] RTW: iConfiguration=0
[ 5751.482614] RTW: bmAttributes=80
[ 5751.482616] RTW: bMaxPower=64
[ 5751.482618] RTW: 
               usb_endpoint_descriptor(0):
[ 5751.482619] RTW: bLength=7
[ 5751.482621] RTW: bDescriptorType=5
[ 5751.482622] RTW: bEndpointAddress=81
[ 5751.482624] RTW: wMaxPacketSize=1024
[ 5751.482625] RTW: bInterval=0
[ 5751.482626] RTW: RT_usb_endpoint_is_bulk_in = 1
[ 5751.482628] RTW: 
               usb_endpoint_descriptor(1):
```

jggithub2017, 2018-01-06:
The coresponding USB traces (usbmon)
USB 2.0
```
ffff8f127f53b300 93028227 S Ci:1:005:0 s 80 06 0200 0000 0009 9 <
ffff8f127f53b300 93028296 C Ci:1:005:0 0 9 = 09023500 01010080 fa
ffff8f127f53b300 93028308 S Ci:1:005:0 s 80 06 0200 0000 0035 53 <
ffff8f127f53b300 93028434 C Ci:1:005:0 0 53 = 09023500 01010080 fa090400 0005ffff ff000705 81020002 00070502 02000200
```
for USB 3.0
```
ffff8f126b8c4300 104056258 S Ci:2:002:0 s 80 06 0200 0000 0009 9 <
ffff8f126b8c4300 104056281 C Ci:2:002:0 0 9 = 09025300 01010080 64
ffff8f126b8c4300 104056296 S Ci:2:002:0 s 80 06 0200 0000 0053 83 <
ffff8f126b8c4300 104056322 C Ci:2:002:0 0 83 = 09025300 01010080 64090400 0005ffff ff000705 81020004 00063004 00000007
```

Trace-2, 2018-01-06:
It would appear some are getting fantastic range with this driver.

**Re4son** has commented back on the above linked kali forum thread that he gets 150m range before he even starts to get a drop in signal when associated with an AC router.

He said he had attached an ammeter and the current draw was under 200ma till the signal started dropping  at 150m then spiked to 500ma when he went further.

Wonder if it's the AC router that's making the difference ... with the same driver and an N router I can hardly associate at 5m ... :(

Trace-2, 2018-01-08:
Ok ... did some comparative current testing yesterday as detailed above.

First off the AWUS036ACH is 12m from my N router and the signal has to go through a brick wall, stud plasterboard wall and a wooden fence and all tests are using a USB3 cable with an accurate multimeter in series on the vbus power wire in the cable.

An interesting qwirk:  In kali with the 5.1.5 driver and USB3 cable plugged into USB3 port I only get my own AP getting picked up and normally can't associate ... but when I put the multimeter in series to measure current I also could then pick up a lot more AP's all on only channels 1 or 11 and could sometimes associate with my AP.

**Kali 2017.3 (kernel 4.14) live usb with persistence driver 5.1.5:**

On both USB3 and USB2 ports results were pretty much the same with the adapter drawing between 175 and 200ma at "idle" up to 266ma on transmit (although on the USB3 port the current went slightly higher to 311ma on download and upload speed tests.

Link Quality and Signal level for both were around 52/70 and -58dBm.

Monitor mode and injection worked.

Changing TX-power value from 18 to 30 dBm made no change to the current drawn.

**Debian Stretch (Kernel 4.9) live usb with persistence driver 4.3.20** (which I get the best range from in managed mode).

On both USB3 and USB2 ports the "idle" results were the same as kali with the adapter drawing around 175ma but on speed tests and transmit the adapter would sometimes go over 500ma.   Adapter always shows as USB2 device bMaxPower=500ma on both USB3 and 2 ports.

The noticeable difference with this driver was the Link Quality and Signal level at 70/70 and around -10 dBm.

Monitor mode and Injection don't work with this driver

TX-power was fixed at 12dBm and couldn't be changed.

**Windows 10 with latest win driver from Realtek support**

Adapter draws between 170 and 230ma (with good Signal levels like the 4.3.20 Linux driver).

After trying many drivers what I've noticed here is that the Link Quality, Signal level and range dropped in the drivers after 4.3.20 when several patches were added. Although I think now this is probably hardware related as some like Re4son mentioned above are getting great range from the 5.1.5 driver.

jggithub2017, 2018-01-08:
> An interesting qwirk: In kali with the 5.1.5 driver and USB3 cable plugged into USB3 port I only get my own AP getting picked up and normally can't associate ... but when I put the multimeter in [...]

I observe exactly the same from time to time too. Then I have to replug the adapter or reinitialize monitor mode or both once or more to get all APs around. But till now haven't investigated it.
By the way USB3 can create interferences with the 2.4 GHz band (observed with a USB3 hub) and you'll see only 5 GHz APs.

Trace-2, 2018-01-08:
@jggithub2017 ... It didn't matter how many times I re-plugged the adapter etc. without the meter in series I only ever got my own AP and could hardly ever associate with it.

Every time I put the meter in series I would get all the local AP's on channels 1 and 11 too ... all 2.4Ghz nobody uses 5Ghz locally here.

Bad range, link and signal quality on all drivers after  4.3.20 ... :(

th3sha10wbr04rs, 2018-01-09:
Hi Friends, i am using AWUS036ACH; i cant to connect any WiFi. some times this adapter not showing any access point [OS:Kali Linux 2017.3 | Virtual Box]

sheshes, 2018-01-10:
Guys I asked the manufacturer to give me as much info regarding the transmission power and power consumption. This are for Alfa AWUS036ACH. Hope they help.

[AWUS036ACH RF Report.pdf](https://github.com/aircrack-ng/rtl8812au/files/1617830/AWUS036ACH.RF.Report.pdf)

[036ACH_PowerConsumption.pdf](https://github.com/aircrack-ng/rtl8812au/files/1617832/036ACH_PowerConsumption.pdf)

kimocoder, 2018-01-10:
Great information in this issue report. Would like to add one more, related to the USB standard (USB 3.0 also listed) the information is taken from [here](https://en.wikipedia.org/wiki/USB) and shows power outages, transmission rate & etc supported. See link for more information.

[image]

Trace-2, 2018-01-10:
Loads of info coming in now .... So is there anything can be done with the latest drivers to improve the very poor Link Quality, Signal Level and Range that a lot of people are experiencing?

Are many of you guys getting great range like Re4son mentioned above ? I've hardly seen anybody say they get really good range with the drivers after 4.3.20.

sheshes, 2018-01-10:
Using kernels <=4.13 not yet. All these issues with the power consumption and txpower are causing the range problems. There’s still a lot to be covered but it looks like a very promising project. It’s a matter of time all these great devs find the proper solution, and when they will it surely going to be an excellent choice for kali

Trace-2, 2018-01-10:
Hi sheshes ... the driver compiles for me on kernel 4.13 & 4.14 no problem if the packages are updated/upgraded and the required depends: Linux-headers, build-essentials, BC etc. are installed first.

There still seems to be at least one person claiming great performance from these latest drivers ... just wondering if anyone else is also getting great performance.

sheshes, 2018-01-10:
Yes i do understand the question. The compilation is done ok for both kernels. The actual functionality of the driver is not yet there in both cases. What Re4son did i presume is updated kali to the latest frm version 2017.1 without doing a dist-upgrade and aupgrading the kernel. I have tested it on kernel 4.10 and i can say that the device functions almost perfectly with some glitches here and there

Trace-2, 2018-01-13:
Here's a couple of shots from Wavemon showing the difference in Link Quality & Signal Level between the 4.3.20 driver with kernel 4.9 and 5.1.5 driver with kernel 4.14 on the same hardware:

Top is 4.3.20 ... bottom is 5.1.5
[image]
[image]

Trace-2, 2018-01-25:
Hi all ... 

Has anybody had a chance to have another look at the range problem with the latest OS and driver or do you think this range issue only affects a small minority of us with certain hardware?

kimocoder, 2018-01-30:
I simply don't have a clue on this one. but it should be the highest prioritized issue to solve, as injection issue now seems resolved.

Trace-2, 2018-01-31:
I'm just going to try the new 5.2.20 driver.

Do you get good range Kimocoder?

kimocoder, 2018-01-31:
I'm using 8814, not 8812 but if you see some changes, please report back and I'll probe through the code for porting.

Trace-2, 2018-01-31:
Just tried the 8812  5.2.20 driver with kali 2017.3 (kernel 4.14) as I already had that installed and there was **no** noticeable improvement in range ... still terrible on this hardware (Optiplex 7010 with i5 CPU) ... :(

Kaliforniah4kor, 2018-02-03:
Can confirm that I pick up a bunch more connections when my 036ach is plugged into a usb 2 port. Not only that but getting handshakes when in usb 2 port runs a whole lot smoother and is fast. In tests I have done I can confirm 100% that the 036ach when plugged into usb 2 port  is on par with my 036nha in both signals it picks up and smoothness in getting handshakes. On sun 3 port it's the shits.

 Once the usb 3 issue is addressed and power can be used to the full manufacturer specs the 036ach will replace my trusty go to 036nha for all my printing needs.

alimp5, 2018-02-15:
Hi all,
at first, I would thank @Trace-2 @kimocoder @evilphish for testing and help to others for fixing 8812AU's issues.
my laptop model is: ASUS N56 - Core i7
i have three wlan interfaces (1 built-in + 2 usb dongles).

built-in: wlan0
usb-dongles: wlan1 + wlan2
wlan1: Alfa 036ACH
wlan2: Alfa 052NH-v2

> phy0	wlan0		ath9k		Qualcomm Atheros QCA9565 / AR9565 Wireless Network Adapter (rev 01)
> phy8	wlan1		8812au		Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
> phy9	wlan2		rt2800usb	Ralink Technology, Corp. RT3572

```
root@kali:~# uname -a
Linux kali 4.14.0-kali3-amd64 #1 SMP Debian 4.14.13-1kali1 (2018-01-25) x86_64 GNU/Linux
```

```
root@kali:~# apt-get install realtek-rtl88xxau-dkms 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
realtek-rtl88xxau-dkms is already the newest version (5.1.5~20180125-0kali1).
```

```
root@kali:~# iwconfig 
kvnet     no wireless extensions.

wlan0     IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          
eth0      no wireless extensions.

wlan2     IEEE 802.11  Mode:Monitor  Frequency:2.417 GHz  Tx-Power=20 dBm   
          Retry short  long limit:2   RTS thr:off   Fragment thr:off
          Power Management:off
          
lo        no wireless extensions.

wlan1     IEEE 802.11  Mode:Monitor  Frequency:2.437 GHz  Tx-Power=18 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
```

a big difference between two Alfa cards:
https://imgur.com/a/fz3Uc

dear @Trace-2, I changed SW6 on 036ACH board, 4 times and checked again the Airodump. but i didn't see any difference. did you find !?
https://user-images.githubusercontent.com/35101372/34620216-990ba796-f23c-11e7-8d4b-05401c1e7f1d.JPG

If you have any ideas or developed new codes and then need to test it, please  say me. because i have a 8812AU card.

Tnx a lot :X

Trace-2, 2018-02-16:
@alimp5 

It's not actually fixed for me ... monitor mode and packet injection work but I still get terrible short range with all the latest drivers.

I didn't get a chance to investigate SW6 ... that's interesting that you tried and no difference ... I was going to trace back the circuit and see what the switch did but I haven't had any time for anything like that for ages now.

kimocoder, 2018-02-16:
[image]

Found this.. "CONFIG_USE_EXTERNAL_POWER" doesn't exist in Makefile. Implementing it after prøving through more code. Check file "rtl8812a_phycfg.c" for some stuff about limitations.

alimp5, 2018-02-18:
@kimocoder @Trace-2 
Hello dears, so you think the root cause of this limitation is for high-TX-power of  Alfa 036ACH !?
In Ebay site, TX power of this Alfa card is: 1600mw. it's really !!!??

dear Kim, I should rebuild wireless driver again !?.  do you have full driver for 8812AU card !?.
I could find just below files in Kali 2018 version:
```
root@kali:~# find / -name "rtl8812a_phycfg*"
/usr/src/rtl8812au-5.1.5/hal/rtl8812a/rtl8812a_phycfg.c
/usr/src/realtek-rtl88xxau-5.1.5~20180125/hal/rtl8812a/rtl8812a_phycfg.c
/usr/src/rtl8812au-4.3.14/hal/rtl8812a/rtl8812a_phycfg.c
```
if you have better customized driver or a full driver for 8812AU and we want just check above functions, please say me which functions should be revised and need code modification.
sorry for bad English language. :)

https://www.ebay.com/itm/Adaptador-WiFi-USB-Alfa-Network-AWUS036ACH-1600mW-Realtek-8812AU-2-4GHz-/173016867553

kimocoder, 2018-02-18:
"git clone https://github.com/aircrack-ng/rtl8812au -b v5.1.5"

Jinovas, 2018-02-19:
@kimocoder 

I've been having some problems with my laptop with a couple issues regarding these drivers(however I believe I just need to just wipe it and start with a fresh install lol).

However, i did regardless install the latest kali vm image on my gigabyte brix for several reasons including this usb power issue. So I'll run some tests today for the 8812 and let you know what I find.

kimocoder, 2018-02-19:
Yeah I bet it's not the issue, have been looking further but.. really annoying as the range (power) limitation is the "final" big issue. Not at least an important one.

Jinovas, 2018-02-19:
Well here's what I've got working pretty smoothly. I used latest Kali 2018.1 VM release, Alfa AWUS036ACH with the "aircrack-ng/rtl8812au" driver version 5.1.5. 

Accidentally closed terminal so running again now to log it

Jinovas, 2018-02-19:
[image]

Trace-2, 2018-02-20:
I will test the new cleaned up 8812au here as soon as I've got free time and report back ... :)

Jinovas, 2018-02-24:
after hours of testing on 2 different machines, I too am not getting much performance out of this chip. Even with power change. I remember this working great in the past with a previous kernel but as of now, not so much :(

nerfologist, 2018-02-28:
I'm also trying out the AWUS036ACH on the v5.1.5 driver (current master 3e78e9), VMWare Fusion, running on a MacBook Retina late 2013 (2 USB3 ports).

I don't have hard evidence to back this up, but the adapter range seems to have improved a bit in comparison with the previous versions. It now picks up at least an AP outside the range of the built-in Apple NIC and I'm able to associate with at least one AP (albeit very close).

I will compare this performance and signal levels with an AWUS036NHA tomorrow if it can be any help. In the meanwhile, thanks all for your debugging effort 👍

kimocoder, 2018-03-01:
kernel v4.15 released in Kali repos. found by "apt-get" and supported by driver. testing power issue should also include the newer kernel as there are quite some changes to it.

Trace-2, 2018-03-01:
**@ Nerfologist** ... a comparison with the NHA would be interesting. Still doesn't sound like your getting anywhere near the outstanding (150m+) range that some are getting.

I've been really short on time recently but I'm hoping to get time this weekend to try the latest two drivers with the new (4.15) kernel ... :)

Trace-2, 2018-03-01:
**@kimocoder**  ... a member on the kali forum has claimed to have compared and found some discrepancy between the 4.3.20, 4.3.21 and 5.1.5 drivers at the bottom of this post:

https://forums.kali.org/showthread.php?38574-Old-AWUS036H-vs-new-AWUS036ACH-performance-comparison

Here's the code comparison:
```

        **MASTER 4.3.20**

        static int usb_reprobe_to_usb3(PADAPTER Adapter)
        {
        	struct registry_priv  *registry_par = &Adapter->registrypriv;
        	int ret = _FALSE;
        	
        	if (registry_par->switch_usb3 == _TRUE) {

        		if (IS_HIGH_SPEED_USB(Adapter)) {
        			if ((rtw_read8(Adapter, 0x74) & (BIT(2)|BIT(3))) != BIT(3)) {
        				rtw_write8(Adapter, 0x74, 0x8);
        				rtw_write8(Adapter, 0x70, 0x2);
        				rtw_write8(Adapter, 0x3e, 0x1);
        				rtw_write8(Adapter, 0x3d, 0x3);
        				/* usb disconnect */
        				rtw_write8(Adapter, 0x5, 0x80);
        				ret = _TRUE;
        			}
        		} else if (IS_SUPER_SPEED_USB(Adapter))	{
        			rtw_write8(Adapter, 0x70, rtw_read8(Adapter, 0x70) & (~BIT(1)));
        			rtw_write8(Adapter, 0x3e, rtw_read8(Adapter, 0x3e) & (~BIT(0)));
        		}

        	}
        	return ret;
        }

        **4.3.20**

        static int usb_reprobe_to_usb3(PADAPTER Adapter)
        {
        	struct registry_priv  *registry_par = &Adapter->registrypriv;
        	u8 ret = _FALSE;

        #if defined(CONFIG_RTL8812A) || defined(CONFIG_RTL8821A)
        	if (IS_HIGH_SPEED_USB(Adapter)) {
        		if ((rtw_read8(Adapter, 0x74) & (BIT(2)|BIT(3))) != BIT(3)) {
        			rtw_write8(Adapter, 0x74, 0x8);
        			rtw_write8(Adapter, 0x70, 0x2);
        			rtw_write8(Adapter, 0x3e, 0x1);
        			rtw_write8(Adapter, 0x3d, 0x3);
        			/* usb disconnect */
        			rtw_write8(Adapter, 0x5, 0x80);
        			ret = _TRUE;
        		}
        		
        	} else if (IS_SUPER_SPEED_USB(Adapter))	{
        		rtw_write8(Adapter, 0x70, rtw_read8(Adapter, 0x70) & (~BIT(1)));
        		rtw_write8(Adapter, 0x3e, rtw_read8(Adapter, 0x3e) & (~BIT(0)));
        	}
        #else
        	rtw_hal_set_hwreg(Adapter, HW_VAR_USB_MODE, &ret);
        #endif
        	return ret;
        }

        **5.1.5**

        static int usb_reprobe_switch_usb_mode(PADAPTER Adapter)
        {
        	struct registry_priv *registry_par = &Adapter->registrypriv;
        	HAL_DATA_TYPE *pHalData = GET_HAL_DATA(Adapter);
        	u8 ret = _FALSE;

        	/* efuse not allow driver to switch usb mode */
        	if (pHalData->EEPROMUsbSwitch == _FALSE)
        		goto exit;

        	/* registry not allow driver to switch usb mode */
        	if (registry_par->switch_usb_mode == 0)
        		goto exit;

        	rtw_hal_set_hwreg(Adapter, HW_VAR_USB_MODE, &ret);

        exit:

        	return ret;
        }
```

kimocoder, 2018-03-01:
Have been reading it now. Feel free to do some edits and tests. I'll hurry and order an ammeter and a 8812au (allready got the 8814) to continue on it. As i said before, the range (power) issue is/should be a HIGH priority issue.

kimocoder, 2018-03-01:
what seems strange it that kernel v4.15 seems to give me higher output than the v4.14 kernel did..

[image]

Do some tests with ammeter and kernel v4.15 and check if there is any difference there.
Play with the code provided by RUSSIAN_BEAR maybe.. I'll go inside the AWUS1900 hardware tomorrow.

pixeldublu, 2018-03-03:
any news on this ?

AspenTheRealist, 2018-03-04:
@Jinovas have you found any version that runs with alfa awus036ach I can't get it to work for nothing I'm trying to using it in a virtual machine kvm... Any ideas?

AspenTheRealist, 2018-03-04:
@everyone

Jinovas, 2018-03-04:
I'll double check today on my 3 setups I have (VMware 14, pure kali laptop, pi 3), but as far as I know, the ACH does work, but even with txpower set to 33dBm, it's performance is not very good.  My alfa NH blows the ACH out of the water

Jinovas, 2018-03-04:
@AspenTheRealist btw, I haven't ran anything in kvm successfully in the past and ran into hella issues. Try VMware, virtualbox or a pure install.

aspentharealist, 2018-03-04:
ok thanks this is my first buy of a alfa or any antenna for this purpose and it really threw me into a mess lol!but it would probubly be a good idea to get a NH while this issue is being hashed out you think?@jinovas

aspentharealist, 2018-03-04:
thanks again for quick reply!

[100 more replies]

closed 2019-08-23 by kimocoder