aircrack-ng/rtl8812au #78: need help to find solution for rtl8814au in kali 2017.03

shell-sh0ckR, 2018-01-04:
Hi there i have build from source but there are already errors ....

Here's the output of dmesg | tail ...

[ 1970.027202] usb 1-2: Product: 802.11ac NIC
[ 1970.027205] usb 1-2: Manufacturer: Realtek
[ 1970.027207] usb 1-2: SerialNumber: 123456
[ 1970.056788] 8814au: disagrees about version of symbol module_layout
[ 2160.844891] perf: interrupt took too long (2542 > 2500), lowering kernel.perf_event_max_sample_rate to 78500
[ 2273.608517] 8814au: module_layout: kernel tainted.
[ 2273.611254] 8814au: Unknown symbol __vfs_read (err 0)
[ 3199.585208] perf: interrupt took too long (3235 > 3177), lowering kernel.perf_event_max_sample_rate to 61750
[ 3636.233407] 8814au: Unknown symbol __vfs_read (err 0)
[ 3664.285700] 8814au: Unknown symbol __vfs_read (err 0)

hope anyone can help me ...

thx so much!!

shell-sh0ckR, 2018-01-04:
..and here the output of dmesg after plug awus1900 in:

[ 1154.366443] usb 1-2: USB disconnect, device number 3
[ 1155.719565] usb 1-2: new high-speed USB device number 7 using xhci_hcd
[ 1155.867973] usb 1-2: New USB device found, idVendor=0bda, idProduct=8813
[ 1155.867978] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1155.867982] usb 1-2: Product: 802.11ac NIC
[ 1155.867984] usb 1-2: Manufacturer: Realtek
[ 1155.867987] usb 1-2: SerialNumber: 123456
[ 1155.888994] 8814au: disagrees about version of symbol module_layout

shell-sh0ckR, 2018-01-04:
uname -r :
4.14.0-kali1-amd64
lsusb:

Bus 002 Device 003: ID 05ac:8406 Apple, Inc. 
Bus 002 Device 004: ID 0781:5589 SanDisk Corp. 
Bus 002 Device 002: ID 2109:0812 VIA Labs, Inc. VL812 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 05ac:0273 Apple, Inc. 
Bus 001 Device 005: ID 05ac:8290 Apple, Inc. 
Bus 001 Device 007: ID 0bda:8813 Realtek Semiconductor Corp. 
Bus 001 Device 004: ID 1050:0407 Yubico.com 
Bus 001 Device 002: ID 2109:2812 VIA Labs, Inc. VL812 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lspci:

00:00.0 Host bridge: Intel Corporation Broadwell-U Host Bridge -OPI (rev 09)
00:02.0 VGA compatible controller: Intel Corporation Iris Graphics 6100 (rev 09)
00:03.0 Audio device: Intel Corporation Broadwell-U Audio Controller (rev 09)
00:14.0 USB controller: Intel Corporation Wildcat Point-LP USB xHCI Controller (rev 03)
00:15.0 DMA controller: Intel Corporation Wildcat Point-LP Serial IO DMA Controller (rev 03)
00:15.4 Serial bus controller [0c80]: Intel Corporation Wildcat Point-LP Serial IO GSPI Controller #1 (rev 03)
00:16.0 Communication controller: Intel Corporation Wildcat Point-LP MEI Controller #1 (rev 03)
00:1b.0 Audio device: Intel Corporation Wildcat Point-LP High Definition Audio Controller (rev 03)
00:1c.0 PCI bridge: Intel Corporation Wildcat Point-LP PCI Express Root Port #1 (rev e3)
00:1c.1 PCI bridge: Intel Corporation Wildcat Point-LP PCI Express Root Port #2 (rev e3)
00:1c.2 PCI bridge: Intel Corporation Wildcat Point-LP PCI Express Root Port #3 (rev e3)
00:1c.4 PCI bridge: Intel Corporation Wildcat Point-LP PCI Express Root Port #5 (rev e3)
00:1c.5 PCI bridge: Intel Corporation Wildcat Point-LP PCI Express Root Port #6 (rev e3)
00:1f.0 ISA bridge: Intel Corporation Wildcat Point-LP LPC Controller (rev 03)
00:1f.3 SMBus: Intel Corporation Wildcat Point-LP SMBus Controller (rev 03)
00:1f.6 Signal processing controller: Intel Corporation Wildcat Point-LP Thermal Management Controller (rev 03)
02:00.0 Multimedia controller: Broadcom Limited 720p FaceTime HD Camera
03:00.0 Network controller: Broadcom Limited BCM43602 802.11ac Wireless LAN SoC (rev 01)
04:00.0 SATA controller: Samsung Electronics Co Ltd Device a801 (rev 01)
05:00.0 PCI bridge: Intel Corporation DSL5520 Thunderbolt 2 Bridge [Falcon Ridge 4C 2013]
06:00.0 PCI bridge: Intel Corporation DSL5520 Thunderbolt 2 Bridge [Falcon Ridge 4C 2013]
06:03.0 PCI bridge: Intel Corporation DSL5520 Thunderbolt 2 Bridge [Falcon Ridge 4C 2013]
06:04.0 PCI bridge: Intel Corporation DSL5520 Thunderbolt 2 Bridge [Falcon Ridge 4C 2013]
06:05.0 PCI bridge: Intel Corporation DSL5520 Thunderbolt 2 Bridge [Falcon Ridge 4C 2013]
06:06.0 PCI bridge: Intel Corporation DSL5520 Thunderbolt 2 Bridge [Falcon Ridge 4C 2013]
07:00.0 System peripheral: Intel Corporation DSL5520 Thunderbolt 2 NHI [Falcon Ridge 4C 2013]

kimocoder, 2018-01-04:
This is weird, because i'm also using 8814au chipset on Kali 2017.3 @ kernel v4.14 and it works great over here. will anyway take a look at it soon. The __vfs_read error is related to kernel v4.14 support.

Have you tried using the "realtek-rtl88xxau-dkms" from apt-get instead?

shell-sh0ckR, 2018-01-04:
sudo modprobe 8814au:

modprobe: ERROR: could not insert '8814au': Exec format error

kimocoder, 2018-01-04:
Weird. I build from repo and it works just fine.
Will look at it when I got time & I have ny laptop and adapter available.

kimocoder, 2018-01-04:
Try "updatedb" and "locate 8814au.ko" remove them all from kernel space.
Then build from repo and install it again ..

CGarces, 2018-01-04:
@shell-sh0ckR

```
[ 3636.233407] 8814au: Unknown symbol __vfs_read (err 0)
[ 3664.285700] 8814au: Unknown symbol __vfs_read (err 0)
```
You are using kernel>= 4.14 with a module compiled on kernel < 4.14 or with outdated code.

The __vfs_read call was removed on 65a62134461330d66f73f87929b0bf1023b2cebc

kimocoder, 2018-01-06:
@CGarces provided information needed information. Remove (all) current installations of 8814au.ko and compile / build it on kernel v4.14.

__vfs_read error is related to kernel v4.14 support, which is supported by the driver.

closed 2018-01-06 by kimocoder

uZerkill3r, 2018-01-12:
thx so much for your help ...will test your advice on monday ...
will let u kn0w if its work ...have a nice weekend ..cheers

kimocoder, 2018-01-12:
It works.. tested many times. Just be sure to have dependencies installed, like "bc" and the CORRECT "kernel-headers"

shell-sh0ckR, 2018-01-18:
@kimocoder could you please so kind to add here a step by step instruction how you built it from repo please and how to install the correct kernel headers ? cause i followed your instructions above and it didn't work :-/ ...would be great to see what i was doing wrong ....thx mate