aircrack-ng/rtl8812au #82: Kernel 4.14

sheshes, 2018-01-09:
I can confirm that this driver doesn't compile on 4.14 kernel after kali linux dist-upgrade 2017.3

Also on kernel 4.13 it compiles and runs but with glitches (sometimes networks are not shown, packet injection is sluggish and after a while everything breaks and needs to be rebooted and sometimes reinstalled to get proper results.

The txpower goes upto 40dbm (which I haven't checked if that corresponds to the actual txpower)

If you need further testing on anything particular please let me know

kimocoder, 2018-01-09:
You may confirm? How about providing error messages or logs instead?
People are compiling it on kernel v4.14 without compilation problems, me included, just be sure you for correct kernel-headers installed to avoid those "kexec" errors.

Txpower is working, bit it's a power issue (see other issue reports!)

Simon5886, 2018-01-09:
i've managed to get it working on kernel 4.14 today, after spending hours tearing my hair out. I'm using the ALFA AWUS036AC, clean build of Kali 2017.3 64 bit. ran apt-get update, upgrade and dist-upgrade. installed bc and libelf-dev then the make && make install worked. using the standard master branch. 

didn't dkms as it kept giving me issues, the adapter is slow but at least i'm able to set it to monitor mode.

kimocoder, 2018-01-09:
master branch isn't really recommend, but if that works for you that's great :)

Simon5886, 2018-01-09:
is there better option? any advice would be greatly received!

kimocoder, 2018-01-09:
well, the best one these days by my opinion is the v5.1.5 branch which got several fixes (VHT, USB3 +++) it's the one we have done some work to, while master and v4.3.21 is more or less left behind. but beware that the drivers still got issues that needs to be sorted out :)

sheshes, 2018-01-10:
I was able to compile 5.1.5 drivers on 4.14 kernel following a clean install and dist-upgrade (using make) dkms is unable to install. BUT

Adapter goes to monitor mode and txpower reaches a maximum at 40, but it doesn't seem to work properly. I cannot see any networks available and they are a few including mine in the neighbourhood.

kimocoder, 2018-01-10:
DKMS got issues at the moment.

More information is found [here](https://bugs.kali.org/view.php?id=4464) & [here](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=886474)

ajwarlock, 2018-01-10:
Thanks a lot, was able to install v5.1.5 branch after following the instructions along with installing "libelf-dev" to my system. I am currently using 4.14.0-kali3-amd64 kernel and the previous driver stopped working after last night's system upgrade. Although this version works but I saw here people talking about some issues in this version of the driver. Would you please list all the issues existing in this v5.1.5 driver for rtl8812au?

Appreciate your work. Thanks again.

kimocoder, 2018-01-10:
* Power issue affecting range (see other issue report for extended information) also effects manually txpower adjustments.

* Injection is sluggish.
* Some PING buffer

These are the most important issues at the moment, getting them resolved would be great.
The power issue is well documented in another report.

These drivers are a mix of old/new code and I'm still wondering why Tenda/Realtek doesn't release a new one which works better than this mix does instead of us adding/fixing all the mess (VHT, USB3) besides many functions om in the driver were hardcoded. Stupid. But at least, we've done some work to it.

kimocoder, 2018-01-12:
Kernel v4.14 is supported. Be sure to have dependencies and correct kernel-headers installed.

closed 2018-01-12 by kimocoder