aircrack-ng/rtl8812au #83: Project License

antlarr, 2018-01-12:
What's the project license?
I found that https://github.com/diederikdehaas/rtl8812AU/ (currently abandoned) was using GPL-2.0 (https://github.com/diederikdehaas/rtl8812AU/blob/driver-4.3.20/LICENSE). Is this project based on that one? Could someone add a LICENSE file to the sources?

Thanks!

CGarces, 2018-01-12:
Well, all source code files has his own header with the license...

antlarr, 2018-01-12:
Good point. Would it be ok then if I submit a PR with a LICENSE file so the license of the project is clear in github?

CGarces, 2018-01-12:
A PR never hurt, please try, but I don't own or have write access to this repository.

kimocoder, 2018-01-12:
But I do.. anyway, look [here](https://github.com/lwfinger/rtl8812au/commit/87ccbd4c5f8b765f8b80f371d7b9472b5f98867b) what Larry Finger (lwfinger) did, he removed them all. I don't know whats the "right" thing to do, but removing them would also be an option

antlarr, 2018-01-12:
That commit removes just the FSF address because that's no longer part of the standard license text, but keeps the actual license on the files (the text just above those deleted sections). Please, it's important to keep clear what's the project license, so linux distributions can know if they can distribute this driver or not.

kimocoder, 2018-01-12:
Then make the PR and it will be pushed.

PR #84 (2018-01-12): Create LICENSE file
Fix #83
closed 2018-01-12 by kimocoder