aircrack-ng/rtl8812au #86: Unable to capture 4-stream traffic with Asus USB-AC68 dongle

jcard0na, 2018-01-15:
The Asus [USB-AC68](https://www.asus.com/us/Networking/USB-AC68/) works fine with this driver, but the phy is configured by default in `3Tx3Rx` mode and the driver will only capture up to 3-stream traffic.  I've tried module options (`rtw_rf_config=1`, `rtw_rf_config=7`) as well as a few tweaks to the driver but I have not been able to get this to capture 4 streams.

Has any one been successful at capturing 4-stream traffic with this card and driver?

[image]

ValZapod, 2018-02-26:
Spatial Streams can't be more than min (TX, RX), no?

kcdtv, 2018-07-26:
This is an AC 1900 device which means
 4 streams and 600 mbps in 2.4 Ghz (4x150mbps) + 3 streams and 1300 Mmps in 5Ghz (3 x 433mbps).
I am used to say stream (flujos) so not 100% about the "spatial stream" but i would say that is that.

ValZapod, 2019-02-15:
@kimocoder we can close this one, too)) Only 3 streams are possible... Even in 802.11n only 3 SS, but with not standart 150 mbit/SS, but 200 Mbit with 256-QAM, which is not even in 802.11n standart((((

kimocoder, 2019-02-15:
Yes. it's sad, that we manage to add functions to 8812/8821, but the 8814 HAL is a mess.. AS we use the same HAL (from 5.1.5 with some additions), the difference between the support in our driver is getting bigger and bigger 😣

closed 2019-02-15 by kimocoder