aircrack-ng/rtl8812au #87: v5.1.5 and kismet

FireFart, 2018-01-18:
I have a AWUS1900 connected to my raspberry pi (rtl8814 chipset).
Using the v4.3.21 branch the adapter picks up local wifis when running the latest kismet (the one with the webgui). After switching to v5.1.5 the wifi adapter works when spinnung up networks, but kismet does not show any found networks.
Is this a known limitation of the v5.1.5 branch?

FireFart, 2018-01-20:
I also tried the AWUS1200 (rtl8812 chipset) and kismet works fine. So this seems to be an issue with the 8814 driver

kimocoder, 2018-01-30:
we've left the v4.3.21 behind for some time now.. you could prone through some patches of v5.1.5 branch or simply change to it as it works out better than 4.3.21 branch does these days.

FireFart, 2018-01-30:
@kimocoder the 4.3.21 branch works with kismet but 5.1.5 not. Alltough I haven't tested the latest Radiotap FCS flag changes on the 5.1.5 branch.

kimocoder, 2018-01-30:
that fixed the WPA handshake capture and wireshark captures at least. may do a quick test right away.

FireFart, 2018-01-30:
Will test it tomorrow

kimocoder, 2018-01-30:
It seems to be working fine over here (AWUS1900), but try it tomorrow and report your findings back. Much appreciated. Thanks

FireFart, 2018-01-31:
Tested the latest version and it works now (but still only if you set the card manually into monitor mode) so closing this issue. Thanks!

closed 2018-01-31 by FireFart

kimocoder, 2018-01-31:
Yes. The driver lack support for virtual interface, it's there but it doesn't work on newer kernels. this also effects the usage of airmon-ng. thanks for reporting back.