aircrack-ng/rtl8812au #88: driver v5.2.20 released (for 8812au)

kimocoder, 2018-01-18:
Driver v5.2.20 is released. for 8812au at the moment.
More information found [here](https://github.com/gordboy/rtl8812au)

A complete changelog may be found [here](https://github.com/gordboy/rtl8812au/blob/master/Realtek_Changelog.txt)

The reason I'm posting this is that features may be ported to our repo if people find it interesting.
See some interesting things right away from the commits, one of them is the KRACK attack fix that we allready ported earlier. Another one is the "VHT extras" found [here](https://github.com/gordboy/rtl8812au/commit/754f76c6cf8a1563d07ac805f5c1cdf0955e3884)

Will take a futher dive into it when I got time again, but this is a heads up. Thanks.

RJVB, 2018-03-01:
One thing I noticed with the current driver in this repo as compared to the one from umdlife/rtl8812au_rtl8821au is that reported signal strengths are much lower, sometimes almost half. I now which value I find less surprising when I'm only getting about 75% signal strength when sitting next to the wifi modem with one driver while the other shows me a value close to 99%...

It'd be interesting to see if that new driver is any different in that department.

RJVB, 2018-06-17:
Any progress here? I'm about to file a ticket for another issue I'm having with the current driver (5Ghz networks no longer being detected after a few sleep/wake cycles) ...

kimocoder, 2018-06-19:
It's only for the 8812au. We use the rtl8821au version, which we added the 8814au HAL files too.
This was just for informational purposes.

RJVB, 2018-06-19:
8812au is exactly the chipset that interests me.

kimocoder, 2018-06-19:
A solution for is may be to break them up in two, and let 8814au stay om v5.1.5 and update 8812au to v5.2.20 and add injection to it but it would take some time and I'd don't know whats the best right now

kimocoder, 2018-06-22:
will order myself a 8812au adapter and start looking into it (adding injection to it for a starter) then maybe actually divide them up. seems like the smartest thing, if not they both stuck on v5.1.5 with all the issues it actually got (according to the v5.2.20 changelog).

RJVB, 2018-06-22:
Sounds like a plan.

Actually I'm beginning to think my adapter has an issue itself. Still have to try it on a 2nd Win10 machine but the old one I tried it on showed the same behaviour as I've been starting to see on my Linux notebook. The WiFi stack shows  a good, strong connection, but at some point nothing is actually being sent or received any longer. That happens with the 4.2 and 5.1 driver on Linux and the native one in Win10, so I think we can exclude any issues on the host and blame the adapter.

Fortunately these things cost almost nothing and this one might still be under warranty but getting it repaired might in fact turn out just as expensive as getting a new one. In which case I think I'll try to find a model with 2.4 and 5Ghz capabilities that doesn't require an external driver...

RJVB, 2018-06-22:
Well, with the tip of that new branch built and loaded I see this in my syslog when I insert the USB adapter

```
Jun 22 22:09:53 bola kernel: [84115.645161] usb 1-2: new high-speed USB device number 7 using xhci_hcd
Jun 22 22:09:53 bola kernel: [84115.816851] usb 1-2: New USB device found, idVendor=0bda, idProduct=a811
Jun 22 22:09:53 bola kernel: [84115.816857] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jun 22 22:09:53 bola kernel: [84115.816860] usb 1-2: Product: 802.11ac WLAN Adapter 
Jun 22 22:09:53 bola kernel: [84115.816865] usb 1-2: Manufacturer: Realtek 
Jun 22 22:09:53 bola kernel: [84115.816867] usb 1-2: SerialNumber: 00e04c000001
Jun 22 22:09:54 bola mtp-probe: checking bus 1, device 7: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2"
Jun 22 22:09:54 bola mtp-probe: bus: 1, device: 7 was not an MTP device
```

and lsusb prints what looks like the expected output (the same as with the 5.1.5 driver), but I'm not seeing a wlan device appear (and the led on the dongle remains off).

kimocoder, 2018-06-22:
give me your "lsusb" output please

kimocoder, 2018-06-22:
Added your VID/PID "0bda:a811". Do a "git pull" to update and build/install it again and try.
Which adapter is this btw? :)

RJVB, 2018-06-22:
I'll probably give you the verdict tomorrow.

The adapter:

https://www.amazon.fr/gp/product/B06XF118RS

kimocoder, 2018-06-22:
Great. Added it to the driver, so it should be supported now. Also did some other nifty changes.
The most important ones will have to wait until I get my own 8812 chipset, but it should work by now.

Will be back in a week for more. Thanks

RJVB, 2018-06-23:
Not so fast ...

That latest version is progress in the sense that the device is recognised, but now I get a sort of boot loop when I insert it, and each time I remove and put it back the wlan device counter increases by 1. Below is an extract. That loop may exit at some point and then the wlanN device shows up in `ifconfig` but uninitialised, and it's not visible in the network manager.

Reloading the 5.1.5 or 4.3.20 driver gives me back the wlan3 device I'm used to.

```
Jun 23 00:02:02 bola NetworkManager[5874]: <error> [1529704922.840607] [nm-system.c:768] nm_system_iface_get_flags(): (unknown): failed to get interface link object
Jun 23 00:02:02 bola wpa_supplicant[6682]: Could not read interface wlan7 flags: No such device
Jun 23 00:02:02 bola wpa_supplicant[6682]: Could not read interface wlan7 flags: No such device
Jun 23 00:02:02 bola wpa_supplicant[6682]: WEXT: Could not set interface 'wlan7' UP
Jun 23 00:02:02 bola wpa_supplicant[6682]: wlan7: Failed to initialize driver interface
Jun 23 00:02:02 bola NetworkManager[5874]: <error> [1529704922.870647] [nm-supplicant-interface.c:997] interface_add_cb(): (wlan7): error adding interface: wpa_supplicant couldn't grab this interface.
Jun 23 00:02:02 bola NetworkManager[5874]: dbus_g_proxy_cancel_call: assertion 'pending != NULL' failed
Jun 23 00:02:02 bola NetworkManager[5874]: <warn> Trying to remove a non-existant call id.
Jun 23 00:02:03 bola kernel: [90844.985239] usb 1-2: reset high-speed USB device number 14 using xhci_hcd
Jun 23 00:02:15 bola kernel: [90857.494759] usbcore: registered new interface driver 8814au
Jun 23 00:02:18 bola kernel: [90860.861931] usb 1-2: USB disconnect, device number 14
Jun 23 00:02:18 bola laptop-mode: Laptop mode 
Jun 23 00:02:18 bola laptop-mode: enabled, not active
Jun 23 00:02:19 bola laptop-mode: Laptop mode 
Jun 23 00:02:19 bola laptop-mode: enabled, not active
Jun 23 00:02:21 bola kernel: [90863.504981] usb 1-2: new high-speed USB device number 15 using xhci_hcd
Jun 23 00:02:21 bola kernel: [90863.680577] usb 1-2: New USB device found, idVendor=0bda, idProduct=a811
Jun 23 00:02:21 bola kernel: [90863.680587] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jun 23 00:02:21 bola kernel: [90863.680595] usb 1-2: Product: 802.11ac WLAN Adapter 
Jun 23 00:02:21 bola kernel: [90863.680602] usb 1-2: Manufacturer: Realtek 
Jun 23 00:02:21 bola kernel: [90863.680607] usb 1-2: SerialNumber: 00e04c000001
Jun 23 00:02:21 bola mtp-probe: checking bus 1, device 15: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2"
Jun 23 00:02:21 bola mtp-probe: bus: 1, device: 15 was not an MTP device
Jun 23 00:02:21 bola laptop-mode: Laptop mode 
Jun 23 00:02:21 bola laptop-mode: enabled, not active
Jun 23 00:02:21 bola kernel: [90863.841468] usbcore: registered new interface driver rtl8812au
Jun 23 00:02:21 bola NetworkManager[5874]: <info> rfkill15: found WiFi radio killswitch (at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/ieee80211/phy11/rfkill15) (driver rtl8812au)
Jun 23 00:02:21 bola laptop-mode: Laptop mode 
Jun 23 00:02:21 bola laptop-mode: enabled, not active
Jun 23 00:02:21 bola kernel: [90863.910732] rtl8812au 1-2:1.0 wlan8: renamed from wlan0
Jun 23 00:02:22 bola NetworkManager[5874]:    SCPlugin-Ifupdown: devices added (path: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/net/wlan8, iface: wlan8)
Jun 23 00:02:22 bola NetworkManager[5874]:    SCPlugin-Ifupdown: device added (path: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/net/wlan8, iface: wlan8): no ifupdown configuration found.
Jun 23 00:02:22 bola kernel: [90863.955581] systemd-udevd[31996]: renamed network interface wlan0 to wlan8
Jun 23 00:02:22 bola NetworkManager[5874]: <info> (wlan8): using nl80211 for WiFi device control
Jun 23 00:02:22 bola NetworkManager[5874]: <info> (wlan8): driver supports Access Point (AP) mode
Jun 23 00:02:22 bola NetworkManager[5874]: <info> (wlan8): new 802.11 WiFi device (driver: 'rtl8812au' ifindex: 14)
Jun 23 00:02:22 bola NetworkManager[5874]: <info> (wlan8): exported as /org/freedesktop/NetworkManager/Devices/16
Jun 23 00:02:22 bola NetworkManager[5874]: <info> (wlan8): device state change: unmanaged -> unavailable (reason 'managed') [10 20 2]
Jun 23 00:02:22 bola NetworkManager[5874]: <info> (wlan8): bringing up device.
UdevQt: unhandled device action "move"
Jun 23 00:02:22 bola NetworkManager[5874]: <info> (wlan8): deactivating device (reason 'managed') [2]
Jun 23 00:02:22 bola wpa_supplicant[6682]: Could not set interface wlan8 flags (UP): Operation not permitted
Jun 23 00:02:22 bola wpa_supplicant[6682]: nl80211: Could not set interface 'wlan8' UP
Jun 23 00:02:22 bola wpa_supplicant[6682]: Could not set interface wlan8 flags (UP): Operation not permitted
Jun 23 00:02:22 bola wpa_supplicant[6682]: WEXT: Could not set interface 'wlan8' UP
Jun 23 00:02:22 bola wpa_supplicant[6682]: wlan8: Failed to initialize driver interface
Jun 23 00:02:22 bola NetworkManager[5874]: <error> [1529704942.875948] [nm-supplicant-interface.c:997] interface_add_cb(): (wlan8): error adding interface: wpa_supplicant couldn't grab this interface.
Jun 23 00:02:22 bola NetworkManager[5874]: dbus_g_proxy_cancel_call: assertion 'pending != NULL' failed
Jun 23 00:02:22 bola NetworkManager[5874]: <info> (wlan8): supplicant interface state: starting -> down
Jun 23 00:02:22 bola NetworkManager[5874]: <warn> Trying to remove a non-existant call id.
[... cut ...]
Jun 23 00:02:30 bola NetworkManager[5874]: <error> [1529704950.132693] [nm-supplicant-interface.c:997] interface_add_cb(): (wlan8): error adding interface: wpa_supplicant couldn't grab this interface.
Jun 23 00:02:30 bola NetworkManager[5874]: dbus_g_proxy_cancel_call: assertion 'pending != NULL' failed
Jun 23 00:02:30 bola NetworkManager[5874]: <info> (wlan8): supplicant interface state: starting -> down
Jun 23 00:02:30 bola NetworkManager[5874]: <warn> Trying to remove a non-existant call id.
Jun 23 00:02:30 bola wpa_supplicant[6682]: Could not set interface wlan8 flags (UP): Operation not permitted
Jun 23 00:02:30 bola wpa_supplicant[6682]: nl80211: Could not set interface 'wlan8' UP
```

kimocoder, 2018-06-23:
It says "[90857.494759] usbcore: registered new interface driver 8814au" .. so this is a 8814au chipset, not the 8812au? Not doing more until mine has come, but this seems like the 8814au chipset?

kimocoder, 2018-06-23:
Yepp, this is the Realtek RTL8811 chipset you got there. It runs under RTL8821AU drivers, not the RTL8812AU unfortunately..

RJVB, 2018-06-23:
Indeed, Win10 shows it as 8811au, but googling led (and still leads) me to the 8812au driver. I can confirm the 8814au driver doesn't work, but the 8812au driver does (until it doesn't any longer for some weird timeout reason).

I'm pretty certain the vendor also claims it has a 8812 chipset in the comments on that Amazon page.

kimocoder, 2018-06-23:
It points to the 8812au, but the one marked 8812a_8821a (which almost is the same) and also is a part of the Makefile, but we don't have the HAL files for it to add it to the driver. This driver only supports the 8812au specific, while v5.1.5 supports 8812au, 8814au and 8821au .. it's a bit messy this one but this is how it is. Take a look St the screenshot from a "working" version from abperiasany

[image]

RJVB, 2018-06-23:
In short, I'm stuck on 5.1.5 or older unless we can somehow get those HAL files? (I thought HAL was obsolete and no longer used?)

kimocoder, 2018-06-23:
I bet si, could always take the HAL from v5.1.5, but that won't bring the changes from v5.2.20 to it anyway.
I bet the HAL files is needed, you may try to disable 8812 compilation (turn it to "n" in Makefile) and enabling the 8821A instead and try compiling and see if that works

kimocoder, 2018-06-24:
Actually, I can add the HAL files today and we'll how it goes, many og the changes made is NOT in the HAL itself so we may het a working one for your adapter. Would be nice if you can try it when it's added (in few hours) since you actually got a 8821a adapter

RJVB, 2018-06-24:
I should be able to test it quickly, yes. (From the moment I actually see the notification ;) ).

kimocoder, 2018-06-24:
It won't work. Also, tried to add the HAL for 8821a from v5.1.5 but that was not to easy as the though was, it needs quite some work

RJVB, 2018-06-24:
Well, I guess I'll continue to juggle between the 5.1.5 driver and my own 4.3.20 version (https://github.com/RJVB/rtl8812au_rtl8821au) then...

kcdtv, 2018-06-24:
Hi there! 
  I played a little around with this new branch and the chip rtl8812au (awuso36ach) in kali linux (`4.16.0-kali2-amd64`)  
 Compilation was clean with no warnings and mode monitor and packet injections seems to work correctly in both bands
 ex: DoS in 5Ghz to collect the handshake:
[image]
I checked also the handshake with wireshark and everything was fine, no "corrupted packet" 
I can't say if the mode monitor behavior as improved or not regarding the other branchs. At least it doesn't look worse at first sight. 
Unfortunately i couldn't check with the latest revisions of reaver and wash as i have a generic  error pointing to pcap
But it worked as intended with versions up to 1.6.4... We will check that

kimocoder, 2018-06-24:
Aaaah this is great! Will continue to work when I get my own 8812au chipset shortly.
The differences between the branches is found in the docs/ folder, the Realtek_changelog, there are quite some fixes since the v5.1.5 there. My additions is in the other changelog, there's also an TODO/upcoming notes. txpower setting (unleashing og power) hasn't been added yet too, but it's tba.

Great! I think the packet injection solution here is a future solution for Realtek drivers / other drivers from Realtek which for me is just awesome to know / learn.

Be back later, thanks for the feedback @kcdtv -- this looks promising

kcdtv, 2018-06-24:
I tried in Xubuntu 18.06 the branch 5.2.20 and i have an error compiling:
```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.15.0-23-generic/build M=/home/kcdtv/realtekac/aircrack-ng/rtl8812au  modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.15.0-23-generic »
Makefile:976: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"
  CC [M]  /home/kcdtv/realtekac/aircrack-ng/rtl8812au/core/rtw_cmd.o
In file included from /home/kcdtv/realtekac/aircrack-ng/rtl8812au/include/osdep_service.h:41:0,
                 from /home/kcdtv/realtekac/aircrack-ng/rtl8812au/include/drv_types.h:32,
                 from /home/kcdtv/realtekac/aircrack-ng/rtl8812au/core/rtw_cmd.c:22:
/home/kcdtv/realtekac/aircrack-ng/rtl8812au/include/osdep_service_linux.h: In function ‘_init_timer’:
/home/kcdtv/realtekac/aircrack-ng/rtl8812au/include/osdep_service_linux.h:280:8: error: ‘_timer {aka struct timer_list}’ has no member named ‘data’
  ptimer->data = (unsigned long)cntx;
        ^~
/home/kcdtv/realtekac/aircrack-ng/rtl8812au/include/osdep_service_linux.h:281:2: error: implicit declaration of function ‘init_timer’; did you mean ‘_init_timer’? [-Werror=implicit-function-declaration]
  init_timer(ptimer);
  ^~~~~~~~~~
  _init_timer
cc1: some warnings being treated as errors
scripts/Makefile.build:332: recipe for target '/home/kcdtv/realtekac/aircrack-ng/rtl8812au/core/rtw_cmd.o' failed
make[2]: *** [/home/kcdtv/realtekac/aircrack-ng/rtl8812au/core/rtw_cmd.o] Error 1
Makefile:1552: recipe for target '_module_/home/kcdtv/realtekac/aircrack-ng/rtl8812au' failed
make[1]: *** [_module_/home/kcdtv/realtekac/aircrack-ng/rtl8812au] Error 2
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.15.0-23-generic »
Makefile:1264: recipe for target 'modules' failed
make: *** [modules] Error 2
```
It first points out some missing dependencies which doesn't really make sense as i can compile the other branches. 
Anyway, i installed the missing packages "to be sure" and the same error happened (without any reference to the missing dependencies this time.) Gordboy  v5.2.20 version compiles correctly (not a surprise as he develops on Ubuntu)  Linux Mint users might face the same error as it is Ubuntu based. Cheers

kimocoder, 2018-06-24:
Yes, this is the _init_timer which was introduced with kernel v4.15 and above.
May fix it when I'm back to my laptop. Thanks for the heads up.

kimocoder, 2018-06-25:
@kcdtv it builds fine under Ubuntu Bionic using kernel 4.15.0-23 ? I installed the libelf-dev too and it stopped nagging about it straight away.. compiled without warnings at all.

[image]

kekkokk, 2018-06-25:
are these suitable for 8814au?

kimocoder, 2018-06-25:
As the README says, no. But will add the HAL files and support it om the future, if it fits to it

kimocoder, 2018-06-25:
* Fix injected frames drop
* Fixed compilation on some ARM platforms
* Added support for PPC platform
* Minor cleanups

Just merged! More tba in a week or so

kimocoder, 2018-06-27:
* Enable TX queue in the monitor mode
* Added support for namespaces (CONFIG_NET_NS)
* Added more Makefile debug
* Added more VID/PID's for future use (8814/8821)
* Updated to a more informational README

Just merged. More tba

kimocoder, 2018-07-01:
Ok.. used some hours today looking into 8814 and 8821 issue today, and what I've learned is that the v5.2.20 got some additions that makes is hard to add those chipset support to it, the only way is that we her the files needed from other releases.

But I've learned something else too, if we use the v5.2.9.3 we could add HAL support for both 8814 and 8821 to it. That would anyway bring some updates (as seen on the Realtek changelog) and that would anyway be better than the v5.1.5 I think.

As for Kali, the drivers should be divided into 2 (3) I think, when finished anyway.
The 8812 on v5.2.20 and 8814 (and 8821) on v5.2.9.3 possibly.

Will take some time to complete all of this, but I'm on the case (on the v5.2.20 currently)

RJVB, 2018-07-02:
Is there any way to get a statement from Realtek why they're excluding still very common (and for sale) chipsets from their latest drivers?

kimocoder, 2018-07-02:
Ok. I've uploaded the v5.2.9.3 to the "v5.2.9" branch today, with 8821 support, but here again.. I can't manage it to support (adding 8814 HAL) 8814 again. These driver make me go nuts.

Will get the 8821 up to v5.2.9.3 anyway, but adding all these patches to both branches will take some time. It's missing frame injection atm, but I'll add in in couple of days along with some monitor/channel patches.

kcdtv, 2018-07-02:
Just to confirm that the compiling worked in ubuntu this time with the updates you made. 
I personally do not notice any advantages in mode monitor with 5.2.9 and 5.2.20 over 5.1.5
I guess the real plus is more in managed mode with this vulnerabilities and bugs patched. And the code optimizations. A shame that adaptation for rtl8814au isn't more easy. 
But for me branch 5.1.5 delivers it too in mode managed so the situation is OK. 
Cheers and thanks for all this great job :smiley_cat:

kimocoder, 2018-07-02:
I'll start on the v5.2.20.2 asap, as the phydm has been improved reducing the 4 second glitch bug.
The 8814 sadly must stay om v5.1.5 until I get a working HAL for it, or if a update to it is released. Info tba

kimocoder, 2018-07-02:
@kcdtv i pushed the v5.2.20.2 driver in the repo, upon the v5.2.20. It needs some testing, just see how it works.. all the above patches have been added to it manually (by hand) but it still needs to be checked. Still haven't got my 8812 adapter, but I can't wait to continue :dancer:

kimocoder, 2018-07-02:
* Fix the CONFIG_NET_NS, support for network namespaces
* Count injected frames in interface stats
* Be quiet about power save
* Unmask all available channels 
* Enable the channel 14 and 80 MHz width channels in the regdom
* Fix error switching to 80MHz channels under certain conditions
* Stop exporting unnecessary symbols 
* Fix channel switch in monitor mode if both CONFIG_WIRELESS_EXT and CONFIG_CFG80211_WEXT are enabled

Just got merged. Enjoy!

kimocoder, 2018-07-03:
* Add CONFIG_USB2_EXTERNAL_POWER supported
* unlock all channels in REALTEK_DEFINE (default) channel plan
* enable DBG_HIGHEST_RATE_BMP_BW_CHANGE with CONFIG_DBG_TXPOWER
* fix txpwr-by-rate logging and value when txpower-override is enabled
* use BIT(n) macros instead of pre-calculated masks in bunch of places
* Add override module parameter for tx-power-index
* ioctl_cfg80211.init_vht_capab: print highest rates that are set there
* Enable rx_stbc bit for 5G in HT mode
* Updated README.md

Just merged. Enjoy!
I think this is where I have to stop until å get the 8812au chipset, cause adding txpower switching with "iwconfig <device> txpower" would needs testing. Also, my USB ammeter is on it's way from China, which would also be needed for the tests. It's on the TODO, so it's coming, one of the most important thing in this driver is that we fix the range (power / txpower) issue which you may find in another issue.

What's next?
* txpower control needs to be added
* USB2/3 switching/function needs to be tested
* iw infomation needs to get more output

RJVB, 2018-07-03:
I'm lost a bit, what branch was that committed to?

Anyway, I fetched the 5.2.9.3 branch and built it. Seems to work with my adapter.

FWIW, I tried a few build tricks to see if those had any incidence on the disconnects and connection timeouts I've been seeing.

For now, it looks like building with "-O3 -ftracer -march=native" gives me the most stable connections (using GCC 7.3 on an Intel N3150).

kimocoder, 2018-07-03:
To the v5.2.20 which now actually is v5.2.20.2.
I've just added the v5.2.9.3 to the v5.2.9 branch, but haven't added frame injection to it, but at least you should have a bit more working driver then the v5.1.5

kimocoder, 2018-07-03:
Thanks to another repo, a fork of Realtek which are based on another "version" of this driver, I've managed to pull some ideas on howto add the 8821 chipset.. therefor, we've added the 8821au support!
It needs to be tested still, but it looks good!

* Added support for 8821au chipset
  Just flip the switches in Makefile, both "CONFIG_RTL8812A" and "CONFIG_RTL8821A" should be turned to "y" to build the 8821au driver.

@RJVB this is for you :)

RJVB, 2018-07-04:
>@RJVB this is for you :)

Why thanks ... with this version the interface no longer keeps increasing each time I reconnect the device, but I'm not seeing any networks through it ;)

With yesterday's 5.2.9.3 rebuild I could use a 5Ghz network for the day (hadn't happened in a long time), but this morning I was again asked to enter the WPA key after waking the machine from sleep. That's a sure sign of "don't bother, won't connect anyway" with this device.

Why oh why is there no IWL-based USB WiFi dongle ...

etem, 2018-07-04:
Hi guys, I've been following this and I can also confirm that 5Ghz networks aren't working with v5.2.20.2 and 8811au chipset. Haven't tried with 5.2.9.3 though.

kimocoder, 2018-07-04:
Ok. Could you try [this](https://github.com/fariouche/rtl8812AU/tree/driver-5.2.20) version and see if that works? It's the v5.2.20 with 8821au support, see if the problems exists over there? Then I may have a pinpoint on in the issue, as the support patch is the same on that repo

RJVB, 2018-07-04:
No, this one seems to work. I had to enter the 5ghz WPA key (of course, even though it was registed in the corresponding network manager profile). The first connection attempt got stuck on "configuring interface" so I disconnected, removed the device and stuck it back in - then it connected just fine.

We'll see how it holds up.

kimocoder, 2018-07-04:
Great. I appreciate getting reports back on how it works! Thanks

kimocoder, 2018-07-04:
Made some more additions today, preparing to support the 8814au chipset and a 8821au firmware download issue is solved + some minor additions.

* Added 8814au HAL files + fixed locations [support not finished yet]
* Fix 8821au firmware download issue
* Add easy build switch for 8821au
* Update README.md

8821au users should "git pull" and rebuild/install again, to fix the firmware download issue.
Enjoy!

RJVB, 2018-07-04:
Did you forget to pull? I'not seeing anything new.

kimocoder, 2018-07-04:
I did actually forget to merge it from my private :p but some additions to it..

* Fix ISO C90 warning on mixed declarations and/or code 
* Added LICENSE file
* Fixed README.md a bit

Merged.

RJVB, 2018-07-04:
And now I can't get a connection anymore at all

```
Jul  4 23:18:02 bola NetworkManager[2718]: <info> (wlan3): supplicant interface state: inactive -> associating
Jul  4 23:18:02 bola wpa_supplicant[3298]: wlan3: Associated with 30:7c:b2:35:21:18
Jul  4 23:18:02 bola NetworkManager[2718]: <info> (wlan3): supplicant interface state: associating -> associated
Jul  4 23:18:06 bola wpa_supplicant[3298]: wlan3: CTRL-EVENT-DISCONNECTED bssid=30:7c:b2:35:21:18 reason=2
Jul  4 23:18:06 bola NetworkManager[2718]: <warn> Connection disconnected (reason 2)
Jul  4 23:18:06 bola NetworkManager[2718]: <info> (wlan3): supplicant interface state: associated -> disconnected
```

and so on, and idem for 2.4Ghz networks.

kimocoder, 2018-07-04:
Solution 1
---------------------------------
Not sure if the build switch was used correct, can't check it right now, but instead of building with "make RTL8821=1", flip both switches in Makefile instead, both "CONFIG_8812=y" and "CONFIG_8821=y" then build/install it again. Remember to re-plug adapter to reload the driver used

Solution 2
---------------------------------
Try this, look [here](https://github.com/kimocoder/rtl8812au/commit/38fa577990440a840221487de57a18eb853674e4) you see the changes on the bottom there, in file "hal/hal_com_phycfg.c" .. try changing it back, remove the "struct txpwr_idx_comp tic;" below the "int    i = 0" and on the other line marked, remove the uncomment, the "//" in front. Then compile the driver again with "make RTL8821=1 && make install" and then "modprobe -r 8812au" to unload and "modprobe 8812au" to load the updated compiled version.

kimocoder, 2018-07-05:
Alfa Networks contacted me today, they are shipping me the 8811 chipset for testing/debug purposes, then I have them all 3 to work with. Anyway, from here I think I'll pause the project for 2-3 weeks, as any commit from here would require functions to be properly tested and checked before commited, so I don't introduce more bugs than Realtek did om the first place :p

* AWUS036ACH (8812au chipset) is on it's way
* AWUS036ACS (8811au chipset) is on it's way
* RD UM34C USB ammeter is on it's way

etem, 2018-07-05:
@kimocoder that's great. Ping me if you need any help with testing the 8811.

kimocoder, 2018-07-05:
DKMS solution only supports 8812au atm.. you have to build manually for 8821au.
I get the 8811 chipset soon anyway, then we'll figure it out

RJVB, 2018-07-05:
>DKMS solution only supports 8812au atm.. 

AFAICT it supports whatever the code can support and the Makefile is configured for. Note that I don't use the dkms-install script but a symlink in /usr/src, with a proper version suffix.

>you have to build manually for 8821au.

I sort of do just that: I call dkms manually :)

>I get the 8811 chipset soon anyway, then we'll figure it out

Exactly what I was thinking...

kimocoder, 2018-07-10:
Just tested the rtl8821au driver (8811 chipset) and it works on first test. Continue the work when I got time :)

etem, 2018-07-14:
@kimocoder how did you build it?. I can't get it working here :( I've built the driver with both `CONFIG_RTL8812A = y` and `CONFIG_RTL8821A = y`. The module is loaded and 5/2.4 Ghz networks are detected but can't connect, even with Open authentication. No error error messages on dmesg. I'm using Kali with kernel 4.14.0-kali1-amd64.

kimocoder, 2018-07-14:
You have to enable them both actually. Both CONFIG_RTL8812A and CONFIG_RRL8821A has to be turned to "y", I can do some more tests today, as I only tested the usages of monitor mode to it, not using Networks Agder (",)

[53 more replies]

closed 2018-09-04 by kimocoder