aircrack-ng/rtl8812au #91: iwconfig wlan0 mode monitor gives error

oguzhantopcu, 2018-01-30:
root@javier-f22:/temp# ifconfig wlx00c0ca960c26 down
root@javier-f22:/temp# iwconfig wlx00c0ca960c26 mode monitor
Error for wireless request "Set Mode" (8B06) :
SET failed on device wlx00c0ca960c26 ; Invalid argument.

The things I did is simply installing driver from this repository.

What could be wrong?

kimocoder, 2018-01-30:
Did you enable monitor support in Makefile before compiling?

oguzhantopcu, 2018-01-30:
No, I could not see any information about this in readme.md. Uninstalling and trying this.

oguzhantopcu, 2018-01-30:
Well, what can I do to enable monitor mode in Makefile? I searched "monitor" and found nothing.

oguzhantopcu, 2018-01-30:
I think I have found it: CONFIG_WIFI_MONITOR = y

But this is already enabled as default. And it is enabled in mine, too.

kimocoder, 2018-01-30:
Oops I thought i was commenting in another repo. It is enabled and should be working just fine.

Jinovas, 2018-02-19:
I've run into this issue before with a couple different driver forks as mentioned in #77. Been trying to isolate the exact issue but I know a few factors are the wireless-regdb and crda databases which need to be recompiled and signed.

FightingForFun, 2018-03-05:
Today download v5.1.5, check makefile CONFIG_WIFI_MONITOR = y 
install using dkms... Try set monitor mode... and have... 

```
>ip link set wlan0 down
>iw dev wlan0 set type monitor
command failed: operation not permitted (-1)
>iwconfig wlan0 mode monitor
Error for wireless request "Set mode" (8B06): set failed on device wlan0; Operation not permitted.
```

kimocoder, 2018-03-05:
Both adapters is supported and should be working. Provide more logs/information/debug.

AspenTheRealist, 2018-03-05:
I couldn't dkms I tryed multiple times I had to make install on kl if that helps idk

kimocoder, 2018-03-05:
Build it manually, do not use DKMS.

kimocoder, 2018-07-24:
No debug/logs or information was given, btw, try jumping over to the v5.2.20 repo which is newer.
Closing for now

closed 2018-07-24 by kimocoder

Chadster766, 2018-12-30:
I've seen this behavior when a two different module versions are loaded for this device.