aircrack-ng/rtl8812au #93: Virtual interface

usama7628674, 2018-02-07:
Does this driver support virtual interface as most of the wifi tools need this feature to work properly?

kimocoder, 2018-02-07:
no i does not support virtual interface, an you are correct. this impact usage of airmon-ng.
the code is there, but isn't fixed or maintained. it should be a priority so i leave this report open for others to see.

kimocoder, 2018-02-16:
Here is provided a commit that almost works, it just needs to be finished up.

https://github.com/astsam/rtl8812au/pull/26

usama7628674, 2018-02-17:
@kimocoder Yeah it's there but it doesn't work yet properly.

vanrzuf, 2018-02-19:
@kimocoder and how is work going on this? any chances to make it work soon?

kimocoder, 2018-02-19:
will try some fixes soon, it's not top priority at this point, as it works fine with setting it manually.

vanrzuf, 2018-02-19:
@kimocoder how it can be done manually? It would be nice if it will work with fluxion :D

usuarionuevor, 2018-02-20:
@vanrzuf fluxion (which is just a forked and renamed [linset](https://github.com/vk496/linset)) doesn't even support 5ghz... use [airgeddon](https://github.com/v1s1t0r1sh3r3/airgeddon) it support 5ghz already and works with rtl8812au drivers.

vanrzuf, 2018-02-20:
@usuarionuevor which drivers willa worki without airgeddon? IT cannot switch to monitor mode when chosen on script, need to set IT manually and IT fails in evil twin attack. I'm using dkms drivers

kimocoder, 2018-03-08:
Found some even more interesting stuff for adding "virtual interface" support.
These 2 commits were made to a v4.3.21 astsam fork. Don't have the time to test it (or port it for testing) at this time, but here they are.

https://github.com/cristi28/rtl8812au/commit/b4f97ee5bd323006b80a0a93a33052e175341c32

https://github.com/cristi28/rtl8812au/commit/7200f38cb27f83ee9a2b19985427244108387799

jarilaos, 2018-08-18:
@vanrzuf [airgeddon](https://github.com/v1s1t0r1sh3r3/airgeddon) has a new release. Now it works with  airmon  non compatible cards, even with the evil twin attack.

kimocoder, 2018-08-18:
Next release of the aircrack-ng (with airmon-ng) will have support for this driver.

v1s1t0r1sh3r3, 2018-08-18:
Hi, I'm one of the authors of airgeddon. Yeah @jarilaos is right. From v8.11 now airgeddon can handle any card compatible with monitor mode even if it is not compatible to airmon. Happy hacking! 😸

kimocoder, 2018-08-18:
It has been added anyway, the commit to aircrack-ng may be seen at https://github.com/aircrack-ng/aircrack-ng/commit/3cc9de5e7cbe6565c7e1fe12ebbcf2c6f7413c8d

v1s1t0r1sh3r3, 2018-08-18:
Great, any improvement an new compatibility on aircrack is welcome! thanks for your work.

ValZapod, 2019-02-15:
@kimocoder Have you read this https://wireless.wiki.kernel.org/en/users/documentation/iw#adding_interfaces_with_iw
I suppose this command is not working `iw phy phy0 interface add moni0 type monitor`.

kimocoder, 2019-04-06:
ooooh no that won't work.. that's just to easy.
this is the output from the command:

ethtool failed...
Only mac80211 devices on kernel 2.6.33 or higher are officially supported by airmon-ng.

ValZapod, 2019-04-06:
@kimocoder I suppose because it is not implemented in driver, as well as monitor flags, that is why this issue #240.

ValZapod, 2019-04-06:
@kimocoder I find this in google https://www.wireshark.org/lists/wireshark-dev/201601/msg00066.html
"So it [the otherbss and control flag] will be set by default in most cases. However I can see that some other drivers do their own management in this area. So it is probably a good idea to set the flags anyway. Having a UI to manage the flags would be even better."

kimocoder, 2019-06-17:
This is only going to happen if we move from nl80211/cfg80211 to mac80211. closing the issue.

closed 2019-06-17 by kimocoder

Medullitus, 2022-02-02:
Help please. I have the same chipset please help!!!

Medullitus, 2022-02-02:
> This is only going to happen if we move from nl80211/cfg80211 to mac80211. closing the issue.

Helpppppp

usama7628674, 2022-02-03:
@Medullitus No help can be provided unless you submit your plea to Realtek and they might consider your request and move from cfg80211 to mac80211.

Medullitus, 2022-02-04:
> @Medullitus No help can be provided unless you submit your plea to Realtek and they might consider your request and move from cfg80211 to mac80211.

Why don't you do that? It seems you love talking too much!

usama7628674, 2022-02-05:
@Medullitus They don't consider requests of outsiders. Maybe your desperation may cause them to change their stance.