aircrack-ng/rtl8812au #96: Aircrack

l4wy3r, 2018-02-21:
Hy!

I am not able to capture handshake, failed the pocket injection test, and the monitoring is really slow. 
I tried it on Parros OS (4.14 kernel) and Ubuntu 17.10. I installed the driver from dkms-install to both os.

Best Regards,
    lawyer

kimocoder, 2018-02-21:
I also use 8814au chipset (on AWUS1900) and handshake capturing is working.
airodump-ng also shows up quick, using only milliseconds.. what are you trying to capture handshake with?

l4wy3r, 2018-02-21:
[image]

**airodump-ng --band abg wlan1:**

[image]

**airodump-ng --bssid XX:XX:XX:XX:XX:XX --channel X --band abg wlan1**:
I never meet the "fixed channel wlan1: 1" before.

[image]

It takes a minute to show up the channel. 
I am continually reconnect to the AP with my phone, but takes no effect.

kimocoder, 2018-02-21:
Works fine on mye Kali, so I dont know where you got it wrong. will post tomorrow when im next to my laptop again.

l4wy3r, 2018-02-21:
First of all, Thanks!! :)

just simply install the package:
`apt install realtek-rtl88xxau-dkms`
then plug in the adapter, after that use the aircrack-ng...

**Some output, if help:** 

`dmesg`: 

>[17548.106096] usb 1-5: new high-speed USB device number 31 using xhci_hcd
[17548.246495] usb 1-5: New USB device found, idVendor=0bda, idProduct=8813
[17548.246499] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[17548.246502] usb 1-5: Product: 802.11ac NIC
[17548.246505] usb 1-5: Manufacturer: Realtek
[17548.246507] usb 1-5: SerialNumber: 123456
[17548.963046] usb 1-5: reset high-speed USB device number 31 using xhci_hcd
[17549.102940] usb 1-5: device firmware changed
[17549.103216] usb 1-5: USB disconnect, device number 31
[17549.226016] usb 1-5: new high-speed USB device number 32 using xhci_hcd
[17549.366670] usb 1-5: New USB device found, idVendor=0bda, idProduct=8813
[17549.366675] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[17549.366678] usb 1-5: Product: 802.11ac NIC
[17549.366681] usb 1-5: Manufacturer: Realtek
[17549.366683] usb 1-5: SerialNumber: 123456
[17549.511038] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[17549.916075] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[17549.916096] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[17549.973019] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[17556.962744] pcieport 0000:00:1c.0: AER: Corrected error received: id=00e0
[17556.962754] pcieport 0000:00:1c.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, id=00e0(Transmitter ID)
[17556.962764] pcieport 0000:00:1c.0:   device [8086:a112] error status/mask=00001000/00002000
[17556.962770] pcieport 0000:00:1c.0:    [12] Replay Timer Timeout 

`uname -a`:

>Linux parrot 4.14.0-parrot13-amd64 #1 SMP Parrot 4.14.13-1parrot13 (2018-01-21) x86_64 GNU/Linux 

`lsmod | grep 8814au`: 
>814au               1556480  0
cfg80211              716800  5 wl,mac80211,8814au,ath,ath10k_core
usbcore               286720  7 uvcvideo,usbhid,8814au,xhci_pci,rtsx_usb,btusb,xhci_hcd

kimocoder, 2018-02-21:
using the apt package and this repo should anyway be working both of them. you got it working then?

l4wy3r, 2018-02-21:
Nope :(

kcdtv, 2018-02-22:
You need to focus airodump-ng using --channel and --bssid filters and **do not use** the tag _--band abg_.
First because you don't need it, it is redundant with the channel. Secondly because if you do so you have to use the correct value... **a** is for **5Ghz** band and **b** is for **2.4Ghz.**.

l4wy3r, 2018-02-24:
Not the `--band` was the problem.
The main problem was with the `check kill`. I used tl-wn722n before, so sometimes i forget to kill processes. I dont think this driver is that much sensitive to the network-manager.
It seems to work fine now, thaks for your help :)

closed 2018-02-24 by l4wy3r