aircrack-ng/rtl8812au #97: kernel v4.15 problems

kimocoder, 2018-02-24:
Kali released kernel v4.15 (4.15.4) in their repos, I updated and tested the drivers and it looks like scanning doesn't work at the moment.

Will have to spend some time finding the problem. Feel free to join me.

Thanks.

Jinovas, 2018-02-24:
Happy to help any way I can! I'll see if I can make another kali box to run the 4.15 kernel and poke around

kimocoder, 2018-02-25:
The commit which added (at least compilation support) for kernel v4.15 is found here https://github.com/aircrack-ng/rtl8812au/commit/f221a169f281dab9756a176ec2abd91e0eba7d19

other source which has successfully added kernel v4.15 support, the commit from lwfinger is found here
https://github.com/lwfinger/rtl8812au/commit/3b4e11e9ecddd3f2fb399f3c53ea5bf782ff2cfc

RJVB, 2018-02-27:
I'm  not even getting this far:

```
> make KVER=4.15.6-ck1-mainline-core2-rjvb V=1 VERBOSE=1
make -j 4 ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.15.6-ck1-mainline-core2-rjvb/build M=/path/to/RTL881x/rtl8812au-git  modules
make[1]: Entering directory `/usr/src/linux-headers-4.15.6-ck1-mainline-core2-rjvb'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (                \
        echo >&2;                                                       \
        echo >&2 "  ERROR: Kernel configuration is invalid.";           \
        echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
        echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";      \
        echo >&2 ;                                                      \
        /bin/false)
mkdir -p /path/to/RTL881x/rtl8812au-git/.tmp_versions ; rm -f /path/to/RTL881x/rtl8812au-git/.tmp_versions/*
make -f ./scripts/Makefile.build obj=/path/to/RTL881x/rtl8812au-git
make[2]: *** No rule to make target `/path/to/RTL881x/rtl8812au-git/core/rtw_cmd.o', needed by `/path/to/RTL881x/rtl8812au-git/8812au.o'. Stop.
make[1]: *** [_module_/path/to/RTL881x/rtl8812au-git] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-4.15.6-ck1-mainline-core2-rjvb'
make: *** [modules] Error 2
```

That's after installing the kernel but without having rebooted - all other add-on modules installed via dkms built just fine. What is wrong here?

kimocoder, 2018-02-27:
Try building manually, not with DKMS, but it was working for v4.15rc2, but not on 4.15.4 (stable) so.. annoying. But try manually and give me the feedback/output

RJVB, 2018-02-27:
That was from a manual build, with the command shown on the 1st line (`make KVER=4.15.6-ck1-mainline-core2-rjvb V=1 VERBOSE=1`)!

kimocoder, 2018-02-27:
As the error saus, this is a kernel error itself, not the driver. The driver should compile on v4.15 but scanning doesn't work, there you got a kernel error that's not related to the driver it seems.

RJVB, 2018-02-27:
Scanning of what?

To me it looks more like something has to be adapted for 4.15 in the Makefile. If this were a regression introduced in the kernel after the last build that worked for you, it should affect other contributed modules as well, no? That doesn't appear to be the case ...

kimocoder, 2018-02-27:
scanning for networks (iwlist, airodump-ng etc).

There are changes in timer setup and the associated handler routines
is kernel 4.15. 

Other drivers (included in kernel) are upstream and maintained for releases, this ones Are fairly new and not upstream, therefor support has to be checked/added for each kernel release to ensure it's supported (both compiling & working).

RJVB, 2018-02-27:
Are we talking about the same things?

The build process fails because somehow it doesn't know how to build rtw_cmd.o . That's way before any scanning for networks can take place.

The other kernel modules I was talking about are NOT part of the kernel (evidently...) but 3rd party modules just like rtl8812au.

kimocoder, 2018-02-27:
For your kernel, I can't really tell whats wrong. Haven't tested that version or been looking at other 3rd party modules. All i know is that this drivers needs to for the new handler routines and timer setup to build with the new kernel, how er going to achieve support I really don't know yet either, cause I'm quite far from my laptop as this issue appeared again (after testing it with rc2, which was working). All i know is that on Kali's kernel provided by the repos, it compile/builds but scanning (for networks) does not work. For other distros/custom kernels .. can't really tell whats wrong (atm at least).

I'm focused at getting it working with Kali, for a starter.

kimocoder, 2018-03-01:
Updated to latest "aircrack-ng" from git and airodump-ng (scanning) was working again.

closed 2018-03-01 by kimocoder

RJVB, 2018-03-01:
>For your kernel, I can't really tell whats wrong.

I'm beginning to wonder if my make-kpkg/kernel-package package isn't too outdated. I installed the latest 4.14 kernel (which should give less issues AFAIU) and the build got to work until I replaced the /lib/modules/4.14.23*/{source,build} symlinks, pointing them to /usr/src/linux-headers-4.14.23* instead of to my build directory. Then that error message I copied above came back.

R.