aircrack-ng/rtl8812au #98: Help with WARNING! Diff between built and installed module!)

vsmsh96, 2018-02-25:
Greetings, 

Per the above, I have ran into some issues.

Today after i set up persistance on USB, I tried installing the driver.

What happened:
1. Downloaded by using : git clone -b v5.1.5 https://github.com/aircrack-ng/rtl8812au.git
cd rtl*
2. sudo apt install dkms
3. Installed driver : sudo ./dkms-install.sh

Then realised that the AWUS036ACH was not working, so I tried restarting but the adapter was still not detected. 
Performed uninstall with : sudo ./dkms-remove.sh
Tried reinstalling and uninstalling multiple times but did not work
Deleted the downloaded rtl8812au folder (after uninstall).

Try install again, this time by using git clone https://github.com/aircrack-ng/rtl8812au.git
Somehow this version worked yesterday when i installed but was lost as I did not set up persistance on my usb yet.
Downloaded and installed but still not working, tried uninstall and reinstall multiple times but adapter still was not working.

Ran dkms status and got this : rtl8812au, 5.1.5, 4.14.0-kali3-686-pae, i686: installed (WARNING! Diff between built and installed module!)
Then tried uninstalling, v5.1.5 was able to be removed but when i try installing again, it says : Good news! Module version v5.1.5_19247.20160830 for 8812au.ko
exactly matches what is already found in kernel 4.14.0-kali3-686-pae.
DKMS will not replace this module.
You may override by specifying --force.
I am not sure what to do now to remove the modules entirely...

Any help is appreciated.

Thank you and warm regards.

kimocoder, 2018-02-25:
"updatedb" and do "locate 8812au.ko" delete them..

If you use Kali, simply use "apt-get install realtek-rtl88xxau-dkms", or build it manually with:

apt-get install bc
git clone https://github.com/aircrack-ng/rtl8812au -b v5.1.5
cd rtl8812au
make
make install

But remove ALL existing 8812au modules first!

vsmsh96, 2018-02-25:
Hi,

Thank you for the reply,

I removed the modules and installed by using : apt-get install realtek-rtl88xxau-dkms but the adapter is still not detected.

What should I do now?

vsmsh96, 2018-02-25:
1. lsusb

Bus 007 Device 002: ID 0781:5581 SanDisk Corp. Ultra
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 002: ID 1bcf:0053 Sunplus Innovation Technology Inc. 
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 002: ID 04d9:1603 Holtek Semiconductor, Inc. Keyboard
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 006: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

2. iwconfig

lo        no wireless extensions.

eth0      no wireless extensions.

kimocoder, 2018-02-25:
"modinfo 8812au" also

vsmsh96, 2018-02-25:
modinfo 8812au

modinfo: ERROR: Module 8812au not found.

kimocoder, 2018-02-25:
so.. when you use "updatedb" and "locate 8812au.ko" where is it located? :P

vsmsh96, 2018-02-25:
Nowhere, it shows me nothing, just root@kali

vsmsh96, 2018-02-26:
Not sure how to do that. 

May be I will try compiling manually per your instructions above and see how it goes.

By the way, do you know if this driver can work with XiaoPan Os and if yes, what commands do I use since Kali commands do not work the same.

kimocoder, 2018-02-26:
Build it manually. Download with "git clone https://github.com/aircrack-ng/rtl8812au -b v5.15" "cd rtl8812au && make && make install" understood?

vsmsh96, 2018-02-28:
Tried manually installing on Xiaopan OS (and failed). 

Used command:
 
git clone https://github.com/aircrack-ng/rtl8812au -b v5.15
cd rtl8812au && make && make install

tc@xiaopan:~$ git clone https://github.com/aircrack-ng/rtl8812au -b v5.15
Cloning into 'rtl8812au'...
fatal: Remote branch v5.15 not found in upstream origin
Unexpected end of command stream
tc@xiaopan:~$ 

tc@xiaopan:~$ git clone https://github.com/aircrack-ng/rtl8812au
fatal: destination path 'rtl8812au' already exists and is not an empty directory.
tc@xiaopan:~$ cd rtl8812au
tc@xiaopan:~/rtl8812au$ make && make install
/bin/sh: bc: not found
make -j 2 ARCH=i386 CROSS_COMPILE= -C /lib/modules/3.16.6-tinycore/build M=/home/tc/rtl8812au  modules
make[1]: *** /lib/modules/3.16.6-tinycore/build: No such file or directory.  Stop.
Makefile:1871: recipe for target 'modules' failed
make: *** [modules] Error 2
tc@xiaopan:~/rtl8812au$

kimocoder, 2018-02-28:
Kernel v3.16.6 ? Too old..

kimocoder, 2018-02-28:
You are missing the kernel-headers package... and "bc" read the instructions om HOWTO install the drivers before spamming the issue report.

gushmazuko, 2018-02-28:
I had the same problem. Solution for fix it:
Remove/purge all other drivers and do next
```bash
apt update && apt dist-upgrade

apt install build-essential bc
apt install linux-headers-`uname -r`

git clone https://github.com/kimocoder/rtl8812au.git
cd rtl8812au

make
make install

modprobe 8812au
```

closed 2018-07-24 by kimocoder