aircrack-ng/rtl8812au #99: The received signal strength in monitor mode is always 0dbm or -128dbm

imnobodyet, 2018-02-28:
Today, I compiled the latest v5.1.5 driver under lede-17.01.2 SDK for ar71xx target(mips_24kc), (need to add a header "<net/cfg80211.h>" in "core/rtw_xmit.c" to make it right). The kernel module was loaded successfully as well as the monitor mode on a mips board. But the signal strength of received packets printed by tcpdump was always 0dbm or -128dbm.

tcpdump outputed like this:
15:37:29.475301 0dBm signal 27657 sq [bit 22] RA:98:9e:63:33:01:d6 BA

i've tried to use wireshark to open saved pcap file, and the signal strength was the same(0dbm).

kimocoder, 2018-02-28:
Thanks for the "lede" support info there. Yes we're aware of the power/txpower issue. See the other issue report thats related to the issue. Thanks

imnobodyet, 2018-02-28:
@kimocoder 
i've searched the other issue reports but they are all about txpower. I was using it in monitor mode and the problem is it cloud not get the right signal strength of every RX packet (0dbm or -128dbm).

BTW, the usb id is "0bda:a811" , linux kernel 4.4.71
Thanks

kimocoder, 2018-07-24:
Try upgrading to the v5.2.20 branch. If the problem persists, just open the issue and report back again. Thanks!

closed 2018-07-24 by kimocoder