aircrack-ng/rtl8812au #100: duplicate code in ioctl_cfg80211.c

RJVB, 2018-03-01:
Consider this, from `rtw_cfg80211_add_monitor_if()`:

```
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 12 ,0))
	mon_ndev->needs_free_netdev = true;
#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 11, 9))
	mon_ndev->priv_destructor = rtw_ndev_destructor;
#else
	mon_ndev->destructor = rtw_ndev_destructor;
#endif
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 11, 9))
	mon_ndev->needs_free_netdev = false;
	mon_ndev->priv_destructor = rtw_ndev_destructor;
#else
	mon_ndev->destructor = rtw_ndev_destructor;
#endif
```

Is there a difference in what `needs_free_netdev` should be set to between 4.11.9 and 4.12.0??

kimocoder, 2018-08-11:
@RJVB which branch was this?

RJVB, 2018-08-11:
>@RJVB which branch was this?

the 5.2.20 branch

kimocoder, 2018-08-11:
Ok. I'll have a look, cause the v4.11.9 and v4.12.0 kernels should be "the same", so it's possible it's a duplicate here. Thanks, I'll look into it.

kimocoder, 2018-08-11:
Well, it doesn't seem to be duplicated in v5.2.20, but in v5.1.5 it seems to be a duplicate.

**In v5.2.20 we got**

`#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 12 ,0))
	mon_ndev->needs_free_netdev = true;
#else
	mon_ndev->destructor = rtw_ndev_destructor;
#endif
#if (LINUX_VERSION_CODE > KERNEL_VERSION(4, 11, 8))
	mon_ndev->priv_destructor = rtw_ndev_destructor;
#else
	mon_ndev->destructor = rtw_ndev_destructor;
#endif
`

**while in v5.1.5 we see**

`#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 12 ,0))
	mon_ndev->needs_free_netdev = true;
#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 11, 9))
	mon_ndev->priv_destructor = rtw_ndev_destructor;
#else
	mon_ndev->destructor = rtw_ndev_destructor;
#endif
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 11, 9))
	mon_ndev->needs_free_netdev = false;
	mon_ndev->priv_destructor = rtw_ndev_destructor;
#else
	mon_ndev->destructor = rtw_ndev_destructor;
#endif`

RJVB, 2018-08-11:
Fwiw I’m currently running a 4.14 kernel.

Sent from a pocket terminal

kimocoder, 2018-08-11:
I'll do some tests on those kernels and see what happens.

RJVB, 2018-08-11:
I wouldn't expect this to make any difference, but I have a stock kernel with just the Con Kolivas patches.

kimocoder, 2018-08-13:
Everything seems to working correctly here, as for the v5.1.5.. I'll leave that one behind.

closed 2018-08-13 by kimocoder