aircrack-ng/rtl8812au #101: [Monitor mode]airmon-ng fails with this driver

sagypt, 2018-03-03:


aircrack-ng, 2018-03-03:
Check out the README.md

closed 2018-03-03 by aircrack-ng

sagypt, 2018-03-03:
I'm sorry but your reply is too short, I don't understand.
I have read README file, but the steps provided don't help me.

I used:
```
ifconfig wlan1 down
iwconfig wlan1 mode monitor
ifconfig wlan1 up
```
... a these work.

But, then, indirectly, I need to use

`airmon-ng start wlan1`

and this fails.

PS: yes, before I used

```
airmon-check
airmon-check kill
```

kimocoder, 2018-03-03:
airmon-ng check kill
ifconfig wlan1 down
iwconfig wlan1 mode monitor
iwconfig wlan1 txpower 30

simple as that.

Jinovas, 2018-03-04:
For quick/easy monitor mode, use my script.  The device will go into monitor mode without name change btw. 
https://github.com/Jinovas/iwmon