aircrack-ng/rtl8812au #108: Remove linux kernel regulatory limits

rodizio1, 2018-03-29:
Thanks a lot for removing all the regdomain stuff from the Realtek driver (https://github.com/aircrack-ng/rtl8812au/commit/ec103e0fa66aaac4d7349bffbfd73e4f3256452e)

I figured you guys could have a use for this patch I made: https://github.com/bortek/EZ-WifiBroadcast/blob/master/kernel/linux-4.9.28-crda.patch

It removes all the CRDA limitations from the kernel and allows injection on all frequencies from 2312-2732Mhz and from 4920Mhz to 6100MHz.

Quick howto:
- Remove all userspace CRDA crap
- Fetch kernel sources
- Apply above patch
- Use internal regdb and disable crda support in kernel config:
```
CONFIG_CFG80211_INTERNAL_REGDB=y
# CONFIG_CFG80211_CRDA_SUPPORT is not set
```
- Build kernel
- Inject frames on every frequency you like

kimocoder, 2018-03-29:
This should be a patch for OS distros (kernel devs) itself, but I'll forward it to Kali developers to look at. Thanks

rodizio1, 2018-03-29:
Indeed, makes more sense. I've posted the patch (and others) here now: https://github.com/offensive-security/wireless-patches/issues/1

kimocoder, 2018-03-29:
Yes, thats where I was thinking it should be. Great, thanks.