aircrack-ng/rtl8812au #109: Add missing device ID for rtl8814au

francoism90, 2018-03-29:
Tried this driver, but it doesn't detect my device. Seems ASUS has a new revision:

```
$ lsusb
Bus 001 Device 004: ID 0b05:1853 ASUSTek Computer, Inc.
```

Thanks!

kimocoder, 2018-03-29:
could you show me the output of "airmon-ng" and I will add it

rodizio1, 2018-03-29:
Looks like there is "0x1852" and "0x1853" for the Asus USB-AC68:

https://github.com/tpircher/rtl8814AU/commit/d2ca79c65545cb002d132e233ac32350002b41bf
https://github.com/tpircher/rtl8814AU/commit/a0c44795f06d846b656ebac3310ba8f80f41333d

kimocoder, 2018-03-29:
Make a PR and I'll add it. on a little easter trip for a week

francoism90, 2018-03-29:
@rodizio1 That was indeed my PR, but I don't know where this should be added for this one. :)

kimocoder, 2018-03-29:
The VID/PID has been added. Use "git pull" or clone & compile/build the driver again for support. Thanks