aircrack-ng/rtl8812au #111: ASUS USB-AC68 Unable to connect

francoism90, 2018-03-30:
I'm able to see networks and when trying to connect to a known network (+ given the WPA-key), it simple hangs in a loop, see `journal`:
```
mrt 30 10:23:36 desktop NetworkManager[8968]: <info>  [1522398216.4909] device (wlp0s20f0u3): supplicant interface state: associating -> disconnected
mrt 30 10:23:46 desktop NetworkManager[8968]: <info>  [1522398226.5053] device (wlp0s20f0u3): supplicant interface state: disconnected -> scanning
mrt 30 10:23:50 desktop wpa_supplicant[2734]: wlp0s20f0u3: CTRL-EVENT-SSID-REENABLED id=0 ssid="WirelessNetwork"
mrt 30 10:23:50 desktop wpa_supplicant[2734]: wlp0s20f0u3: Trying to associate with c4:a... (SSID='WirelessNetwork' freq=5240 MHz)
mrt 30 10:23:50 desktop NetworkManager[8968]: <info>  [1522398230.3718] device (wlp0s20f0u3): supplicant interface state: scanning -> associating
mrt 30 10:23:50 desktop wpa_supplicant[2734]: wlp0s20f0u3: CTRL-EVENT-ASSOC-REJECT status_code=1
mrt 30 10:23:50 desktop wpa_supplicant[2734]: wlp0s20f0u3: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="WirelessNetwork" auth_failures=2 duration=20 reason=CONN_FAILED
mrt 30 10:23:50 desktop wpa_supplicant[2734]: wlp0s20f0u3: CTRL-EVENT-ASSOC-REJECT status_code=1
mrt 30 10:23:50 desktop NetworkManager[8968]: <info>  [1522398230.4213] device (wlp0s20f0u3): supplicant interface state: associating -> disconnected
mrt 30 10:23:58 desktop NetworkManager[8968]: <warn>  [1522398238.0034] device (wlp0s20f0u3): Activation: (wifi) association took too long
mrt 30 10:23:58 desktop NetworkManager[8968]: <info>  [1522398238.0035] device (wlp0s20f0u3): state change: config -> failed (reason 'no-secrets', sys-iface-state: 
'managed')
mrt 30 10:23:58 desktop NetworkManager[8968]: <info>  [1522398238.0038] manager: NetworkManager state is now DISCONNECTED
mrt 30 10:23:58 desktop NetworkManager[8968]: <warn>  [1522398238.0051] device (wlp0s20f0u3): Activation: failed for connection 'WirelessNetwork'
mrt 30 10:23:58 desktop kernel: IPv6: ADDRCONF(NETDEV_UP): wlp0s20f0u3: link is not ready
mrt 30 10:23:58 desktop NetworkManager[8968]: <info>  [1522398238.0064] device (wlp0s20f0u3): state change: failed -> disconnected (reason 'none', sys-iface-state: 
'managed')
mrt 30 10:24:02 desktop NetworkManager[8968]: <info>  [1522398242.4298] device (wlp0s20f0u3): supplicant interface state: disconnected -> inactive
```

kimocoder, 2018-03-30:
For dd-wrt with tweaks i don't know. Tested & working on my router collection at least.

francoism90, 2018-03-31:
I'm using the following settings:
```
Regulatory Domain: Singapore
Wireless Channel: 48 - 5.240 GHz
Channel Width: VHT80 (80 MHz)
Extension Channel: upper upper
```

Seems there are still some valid checks in the driver. The 'older' v4.3.21 driver works fine.

francoism90, 2018-05-12:
@kimocoder I'm now using a different AP (Asus RT-AC87U) and now everything works fine. :)
Don't know why it didn't work with the dd-wrt one, maybe some options aren't 100% compatible with the v5.1.5 regcode.

closed 2018-05-12 by francoism90