aircrack-ng/rtl8812au #112: Support for Linux-headers-4.15.0-kali2

pbr0ck3r, 2018-03-30:
When trying to install on kali. When I run `uname -r` I receive `4.14.0-kali3-amd64` but when I try to install I get a dependency missing of `linux-header-4.14.0-kali3-amd64`. When I try to apt-get install that dependency it says that it does not exist, but `linux-header-4.15.0-kali2-amd64` exists.

Is there something I should change in code to support these other linux-headers?

kimocoder, 2018-03-30:
"apt-cache search linux-headers" will reveal headers package. install the correct one

pbr0ck3r, 2018-03-30:
Just shows `linux-headers-4.15.0-kali2-amd64`, `-common`, `-cloud-amd64`, `-all`. Then a bunch of `linnux-libc-dev` drivers.

kimocoder, 2018-03-30:
well, I'm on easter vecation and can't check it out right away, but I'll leave it open until monday when I'm back.

pbr0ck3r, 2018-03-30:
No worries! Enjoy vacation!

pbr0ck3r, 2018-03-30:
I am using the latest kali version as well... Kali-2018.1

kimocoder, 2018-03-30:
Take a look [here](https://http.kali.org/kali/pool/main/l/linux/) it's listed there and should also be found in apt repo. But installing ut manually should work just fine

pbr0ck3r, 2018-03-30:
Found it. But after trying to install it. It is missing a bunch of other dependency and not all exist in `kali/pool/main/l/linux`. One dependency being `linux-headers-4.14.0-kali3-common`. 4.14.0-kali3-common_4.14.17 exists` I tried installing that but the `linux-headers-4.14.0-kali3-amd64` deb install did not like that. Said it depended on specifically 4.14.0

kimocoder, 2018-03-30:
well, whats the output of "cat /etc/apt/sources.list" ?

pbr0ck3r, 2018-03-30:
`deb http://http.kali.org/kali kali-rolling main non-free contrib`

kimocoder, 2018-04-02:
You will need to add "deb http://http.kali.org/kali kali-dev main non-free contrib" to your "/etc/apt/sources.list" then do "apt-get update && apt-get install linux-headers-4.14.0-kali3-amd64".

`
root@kali:~# apt-cache policy linux-headers-4.14.0-kali3-amd64
linux-headers-4.14.0-kali3-amd64:
  Installed: 4.14.17-1kali1
  Candidate: 4.14.17-1kali1
  Version table:
 *** 4.14.17-1kali1 500
        500 http://http.kali.org/kali kali-dev/main amd64 Packages
        100 http://http.kali.org/kali kali-experimental/main amd64 Packages
        500 http://http.kali.org/kali kali-dev-only/main amd64 Packages
        100 /var/lib/dpkg/status
root@kali:~# 
`

pbr0ck3r, 2018-04-02:
`linux-headers-4.14.0-kali3-amd64` do not exist in that source. Just the `common` and `common-rt` for `4.14.0`

pbr0ck3r, 2018-04-02:
I was able to successfully install `rtl8812au` after install manually the following from [https://http.kali.org/kali/pool/main/l/linux/](https://http.kali.org/kali/pool/main/l/linux/)
`linux-headers-4.14.0-kali3-common_4.14.13-1kali1_all.deb`
`linux-headers-4.14.0-kali3-common-rt_4.14.13-1kali1_all.deb`
`linux-headers-4.14.0-kali3-amd64_4.14.13-1kali1_amd64.deb`

kimocoder, 2018-04-02:
It's listed inn the sources above. Just checked & confirmed. This "issue" is rather a Kali question (https://bugs.kali.org) but great you got it working.

closed 2018-04-02 by kimocoder

pbr0ck3r, 2018-04-02:
Reported the issue to kali.

kimocoder, 2018-04-02:
.. that is the way to put it into monitor mode.. check README.md - the airmon-ng isn't supported due to lack of virtual Interface support in driver (atm)