aircrack-ng/rtl8812au #113: Unknown event 47

Karlodun, 2018-04-03:
Starting hostapd /etc/hostapd.conf -d shows unknown event 47 (rtl8814au):

nl80211: Drv Event 34 (NL80211_CMD_NEW_SCAN_RESULTS) received for wlp0s20u1
wlp0s20u1: nl80211: New scan results available
nl80211: Scan probed for SSID ''
nl80211: Scan included frequencies: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 5170 5180 5190 5200 5210 5220 5230 5240 5260 5280 5290 5300 5310 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5745 5765 5785 5805 5825 5845
wlp0s20u1: Event SCAN_RESULTS (3) received
RTM_NEWLINK: ifi_index=2 ifname=wlp0s20u1 wext ifi_family=0 ifi_flags=0x11043 ([UP][RUNNING][LOWER_UP])
VLAN: RTM_NEWLINK: ifi_index=2 ifname=wlp0s20u1 ifi_family=0 ifi_flags=0x11043 ([UP][RUNNING][LOWER_UP])
VLAN: vlan_newlink(wlp0s20u1)
nl80211: Drv Event 33 (NL80211_CMD_TRIGGER_SCAN) received for wlp0s20u1
wlp0s20u1: nl80211: Scan trigger
wlp0s20u1: Event SCAN_STARTED (47) received
Unknown event 47

kimocoder, 2018-04-03:
Yes, a similar one may be seen [here](https://github.com/astsam/rtl8812au/issues/58) also it's explained in other reports in the astsam repo. will have a look at it soon. thanks for reporting.

kimocoder, 2018-04-08:
this issue is related to the driver not supporting (running on) the newer mac80211. closing for now, because this will (maybe) be fixed in the future. the whole driver needs to be overhauled.

closed 2018-04-08 by kimocoder