aircrack-ng/rtl8812au #114: Frame Injection not working on AWUS1900 with RTL8814AU

LegitBurst, 2018-04-06:
Hey,

I got the RTL8814AU working with the newest Kali Linux build on VMWare running on the latest MacBook Pro. Monitor Mode is working fine using the 5.1.5 branch.

Running

> aireplay-ng -9 wlan0

gives me:

> root@Flow:~# aireplay-ng -9 wlan0
> 16:42:59  Trying broadcast probe requests...
> 16:43:01  No Answer...
> 16:43:01  Found 0 APs
> root@Flow:~# 

And yes I ran 

> airmon-ng check kill

before.

Does anyone know how to fix the clearly not working frame injection even though this driver states that it should work.

kimocoder, 2018-04-06:
I use AWUS1900 myself (on Kali) and it's working great. Can't really tell whats wrong without more information/debug logs

TooPrecise, 2018-04-06:
What exactly do you need? @kimocoder

kimocoder, 2018-04-08:
I use Kali as direct install. and it's working great. btw, there are some issues related to at least some VMware setups, just tried another issue report (related to Debian 9) and there the driver has some issues which I don't know if the issue is related to the VMware itself or the 4.9 kernel. anyway, pure install (direct install) works fine.

root@kali:~/Downloads# aireplay-ng -9 wlan1
18:07:20  Trying broadcast probe requests...
18:07:20  Injection is working!
18:07:22  Found 14 APs

TooPrecise, 2018-04-09:
OK, so I just installed the latest Kali build on Virtualbox instead of VMWare and here it seems to work:

> 17:23:59  Trying broadcast probe requests...
> 17:23:59  Injection is working!
> 17:24:01  Found 1 AP 
> 
> 17:24:01  Trying directed probe requests...
> 17:24:01  AA:AA:AA:AA:AA:AA - channel: 1 - 'WLANTEST'
> 17:24:06  Ping (min/avg/max): 4.684ms/10.213ms/14.727ms Power: -65.80
> 17:24:06   5/30:  16%

##CHANGED MAC-ADDRESS AND SSID FOR OBVIOUS REASONS

Is it possible to do a proper direct install as DualBoot in Mac OS High Sierra? Can't find the the right information on this. And on which Device did you perform the Direct Install?

Cheers!

kimocoder, 2018-04-10:
You would have to use another bootloader for Mac, not really recommended. Besides, I'm not OSX friendly either. Closing.

closed 2018-04-10 by kimocoder