aircrack-ng/rtl8812au #117: Cannot set MTU higher 1500

alexeypertcev, 2018-04-19:
Hi,
I'm trying run airbase-ng on each driver versions (8812au and 8814au), and i receive same errors in both: 
```
  Trying to set MTU on wlan0 to 1800
  error setting MTU on wlan0
  MTU on wlan0 remains at 1500
```
also, i can't change MTU to 1800 with ifconfig and ip link set.

Is this issue hardware or driver limitation? If problem in the driver, can you give me the cue how correct this?

francoism90, 2018-04-24:
Any logging in `journal` or `dmesg`?

alexeypertcev, 2018-04-25:
Both, journal and dmesg contain same record:
`wlan0: Invalid MTU 1800 requested, hw max 1500`

kimocoder, 2018-07-24:
You are correct. Issue acknowledged, don't know what to do about it yet but there's no option to set MTU at all in the driver.

ValZapod, 2019-01-10:
And is it possible to set VLAN? I suppose if it's possible, than changing MTU must also be possible, as VLAN  Header changes the MTU.

ValZapod, 2019-02-15:
@kimocoder Also I suppose there is no vlan support as well.

kimocoder, 2019-02-15:
Aaah no, it's there but it's not working though.. one of the things that annoys me as I've been looking at it for 2 years. In the v5.3.9 base it's actually more complete.

We went the other way and added support into airmon-ng until we magically find a fix to it

kimocoder, 2019-02-15:
@ValZapod looks binded to CONFIG_CONCURRENT_MODE. Haven't tested CONCURRENT and P2P yet, but I will soon just to see what actually happen.

For the VIF (virtual interfaces code), see "usb_intfs.c" and "os_intfs.c" and "ioctl_cfg80211.c"
it's laying around there.

------------------------------------------------------------------------------------------
os_ndevs_deinit:
        if (status != _SUCCESS)
                rtw_os_ndevs_deinit(dvobj);
#endif
free_if_vir:
        if (status != _SUCCESS) {
                #ifdef CONFIG_CONCURRENT_MODE
                rtw_drv_stop_vir_ifaces(dvobj);
                rtw_drv_free_vir_ifaces(dvobj);
                #endif
        }

free_if_prim:
        if (status != _SUCCESS && padapter)
                rtw_usb_primary_adapter_deinit(padapter);

free_dvobj:
        if (status != _SUCCESS)
                usb_dvobj_deinit(pusb_intf);

ValZapod, 2019-02-15:
@kimocoder No, I am not talking about virtual interfaces, *wlan0mon* and so on,  it is "VLAN", like https://en.wikipedia.org/wiki/Virtual_LAN
When vlan tag is added to Ethernet header it changes the MTU...

kimocoder, 2019-02-15:
Know about it, but.. got no clue to it :-1: 
VHT and antenna code also.. kiils me totally

reg1reg1, 2019-06-02:
Does this mean that we still have the original issue of fragmentation if this is displayed while starting up airbase-ng? So the only solution is different wireless hardware?
[https://github.com/aircrack-ng/aircrack-ng/issues/469](https://github.com/aircrack-ng/aircrack-ng/issues/469)
I still face the problem described in the issue #469

kimocoder, 2019-06-02:
The solution would be having drivers with less issues, but since february  we got 2 new releases, the v5.3.4 and 5.6.3 (not finished up, missing frame injection + 8821au and 8814au support). And the v5.6.4 would be a step up it looks by the code.

Even better is that Realtek has started on rtw88, a mac80211 stack base for 8822b and others, would have been nice if they've add support for our chipsets into that, so we could move away from old nl80211/cfg80211 mix we're running on. We'll see, I'll be in Taipei, Taiwan in August, I'll be happy to share that toughts with Realtek :+1:

kimocoder, 2019-06-02:
Btw, [this](https://patchwork.kernel.org/patch/10928417/) fix has been added to v5.6.4 and seems to fix range issues, besides that the phydm is updated. as for the MTU, I'm really not that familiar to it and it's usage/function so I'll have to read/learn about it before I may dig into it :+1:

reg1reg1, 2019-06-02:
I did install the drivers but I don't think my card supports that driver. It was running the rtl88xxau drivers earlier. On uninstalling the drivers and reinstalling the drivers you mentioned but the card does not pick it up. Thanks for the prompt response though! I will try to switch to Atheros in the meanwhile.

kimocoder, 2019-10-22:
@alexeypertcev 

This patch below has been added to branch v5.2.20.
may you confirm the issue is resolved?

closed 2019-10-24 by kimocoder